ExpressLRS/ExpressLRS #508: RX (R9mm) Powers off after rxloss

madmiked182, 2021-04-22:
Hey folks! First of all want to say thanks for all your effort and this outstanding project!

Coming from Germany and getting really pissed about the F---y policies I did flash my R9 Hardware over to ExpressLRS and had quite a hard time to get the R9mm running on the Omnibus F4 (Emax) since the RX on the uart1 one seems to be reserved for SBUS only.

Finally got that to work, altogether with a stable link and telemetry but discovered the following issue. When turning of the transmitter (D9x plus) receiver sends signal loss to the Omnibus F4 and after 1 maybe 2 seconds the RX is shutting down completely with no LEDs flashing an no possibility to gain back connection when turning the radio back on.

I need to to pull the battery and reconnect it to get the R9mm back to work which could probably a hassle in the field.

Note: Measured power on + and - lines directly on the Rx and I still got power from the board after the r9mm did shut down so I consider this an software issue from receivers side.

Any workaround or similar problems discovered with that?
By the way I did flash the actual 1.0.0-RC1 on both RX and TX of course
Many thanks in advance!
Regards,
Mike

madmiked182, 2021-04-22:
Hi.
Verified this issue on another board with another r9mm. Pretty much the same.
Switched back to 0.1.3-RC2 with no issues anymore.
Receiver flashes red and gets connection back after turning radio back on. As it should be.

So I guess this might be an evolution bug in 1.0.0-RC1.

Kind regards,
Mike

schugabe, 2021-04-22:
I cannot reproduce the issue - the rx gets the connection back after I turn the tx back on.

What options did you enable in the configurator?

Sebastos34, 2021-04-22:
@madmiked182, I have noticed the same problem with my R9M 2019 and R9mm.

nangel0, 2021-04-23:
Can you try to reproduce that using vscode flash and monitor to see if the RX really shuts off or goes to rf modes scrolling by looking at that 'matrix code' it outputs.

dentex, 2021-04-23:
Same problem here, with the same setup. I turned off the radio while the quad was armed, then again ON. It won't arm, emitting a quick series of tones from the buzzer upon switching the arm channel (Emuflight FW).
The buzzer code seems to be "15", 5 quick tones + 2 long.
[image]

What happens is that the RX stays in failsafe for about 15 seconds, in fact this is what the Emuflight configurator is reporting:
[image]

Afterwards, the R9MM stops to blink, gets again the solid green color and arming is possible.

CapnBry, 2021-04-23:
@madmiked182 Can you reply with what defines you have set, what packet rate, and telemetry ratio you are using? I also can not reproduce.

dentex, 2021-04-23:
> @madmiked182 or @dentex
> Can you reply with what defines you have set, what packet rate, and telemetry ratio you are using? I also can not reproduce.

From the configurator logs:
`build flags: ['-D PLATFORM_STM32=1', '-O2', '-DTARGET_RX=1 -Wall -Iinclude', '-D TARGET_R9M_RX', '-D TARGET_100mW_MODULE', '-D TARGET_USE_EEPROM=1', '-D TARGET_EEPROM_ADDR=0x50', '-D HSE_VALUE=24000000U', '-DVECT_TAB_OFFSET=0x08008000U', '-DRegulatory_Domain_EU_868', '-DMY_UID=97,124,64,141,190,14', '-DHYBRID_SWITCHES_8', '-DENABLE_TELEMETRY', '-DFAST_SYNC', '-DLATEST_COMMIT=0x9,0x2,0xd,0xf,0x8,0x0']`

- packet rate: 200Hz
- TLM ratio 1:32

dentex, 2021-04-23:
What I didn't mention, is that the telemetry restarts to come in as soon as I turn the radio on again, with the same short delay as it does every time.

CapnBry, 2021-04-23:
Thanks for the information dentex. I do not have any 900MHz hardware but your details can hopefully be used by one of the other developers. Wez also reported something similar on discord so I am going to add his information as well. 

> 1. Happymodel 900mhz RX, gets a link fine, but if you turn off the handset (drop link) and turn it back on again, the link never recovers. No LEDs on the RX. Power cycling the RX recovers link.

It seems to be related to 900MHz because Team2.4 can reboot the TX over and over. Tried 6 times and one time rebooted so fast, the RX never even reported a link loss, the other 5x it lost the link and went back to red blinkie.

Sebastos34, 2021-04-23:
I have finally found what was wrong! I had to untick the "LOCK_ON_FIRST_CONNECTION" box in the R9mm Firmware, now all works fine, and the link come back after a failsafe! I can change the rate without link loss too. :)
My gear : R9M 2019, and R9mm

CapnBry, 2021-04-23:
I do see something thanks to sebastos34's comment. When I enable LOCK_ON_FIRST_CONNECTION, the LED doesn't blink when it is disconnected, it just goes off and stays off. The FM30 mini connects right up again though, so the LED thing is purely cosmetic but prevents being able to indicate that it isn't dead. We can probably fix at least that if the other devs let me, but unfortunately it doesn't help with reproducing this issue for me.

gabrielgradinaru, 2021-04-25:
I have this issue on both R9MM and R9 Slim+ OTA so it is not an issue only with R9MM.

dentex, 2021-04-25:
> I have this issue on both R9MM and R9 Slim+ OTA so it is not an issue only with R9MM.

Do you have the delay in re-connection or a power loss?

gabrielgradinaru, 2021-04-26:
In my case the LED's stop so it looks like a power loss.

CapnBry, 2021-04-26:
@gabrielgradinaru Are you running with LOCK_ON_FIRST_CONNECTION? If so, all LEDs turn off on disconnect and stays off, which can look like a power loss even though it is still powered. This will be fixed in RC2.

gabrielgradinaru, 2021-04-26:
@CapnBry Yes, I am running the default LOCK_ON_FIRST_CONNECTION and thank you for the update.
But besides the LEDs, the problem still remains that the link remains down when turning the TX back on.

CapnBry, 2021-04-26:
Ah thanks for the additional information, there's a separate change intended to try to address the link not coming back up after disconnect coming in RC2 so when that's available that might be a fix for you.

CapnBry, 2021-04-28:
RC2 is out with fixes for the non-blinking LED and changes to prevent the RX from not going back to receiving after losing connection, as well as more stable syncing. Please test and see if these fixes resolve the problems you folks are having.

gabrielgradinaru, 2021-04-28:
I have tested RC2 and confirm it fixes this issue.

I've flashed using LOCK_ON_FIRST_CONNECTION on R9MM and R9 Slim+ OTA and observed 2 things:
- While TX is powered off, the RX is blinking the red LED
- When powering on the TX, the link is re-established as expected

Great work on this, keep it up.

dentex, 2021-04-28:
Today I flew a few packs on RC2, all good. 
But the ~20 sec delay upon switching the radio back on persisted.
I did not use LOCK_ON_FIRST_CONNECTION.

Sebastos34, 2021-04-29:
> Today I flew a few packs on RC2, all good.
> But the ~20 sec delay upon switching the radio back on persisted.
> I did not use LOCK_ON_FIRST_CONNECTION.

With the RC2 you have to tick "LOCK_ON_FIRST_CONNECTION", the bug have been corrected. For me, there is near no delay for the link to come back now. Try to remove the module instead of switch the radio off to see if the link come back quickly.

CapnBry, 2021-04-29:
> Today I flew a few packs on RC2, all good.
> But the ~20 sec delay upon switching the radio back on persisted.
> I did not use LOCK_ON_FIRST_CONNECTION.

This is the case if you're on 900MHz and don't use LOCK_ON_FIRST_CONNECTION. The issue is the RX needs to sit at each rate for one complete hop cycle to make sure that rate isn't the one being used. For 900MHz, that's 1.62s (200Hz), 3.24s (100Hz), 6.48s (50Hz) and 12.96s (25Hz) = 24.3s per cycle like you're seeing. On RX initial boot, if your TX is on, it will sync in the first period and starts at 200Hz, so it will appear very fast. When you turn off your TX without LOCK_ON, the RX starts cycling looking for the TX's rate and it seems to take forever to sync. I believe the use case for someone leaving the RX on but power cycling the TX is pretty rare so users will practically never see this. It can be a problem if you failsafe because it will take a long time to re-sync if it doesn't happen pretty quickly (10x the above timeout), but that's what LOCK_ON_FIRST_CONNECTION is for, so just turn that on if that's a concern.

dentex, 2021-04-29:
> This is the case if you're on 900MHz and don't use LOCK_ON_FIRST_CONNECTION. The issue is the RX needs to sit at each rate for one complete hop cycle to make sure that rate isn't the one being used. [...]

Thank you very much for the explanation. That's clear. 
I'll have a chance to try this out later today (just to confirm, if it's even necessary) and I'll report here.

dentex, 2021-04-29:
I made a few tests, always same R9 TX module, and same RX type, the R9MM, but on 3 builds:

1. FC: Aribot F7 - FW: Emuflight => same behavior between RC1, RC2 and wether LOCK_ON_FIRST_CONNECTION is enabled or not: about **20 sec delay** in reconnection powering up the TX after an RX-LOSS;
2. FC: MambaF722-I2C - FW: Emuflight => same behavior between RC1 without LOCK_ON_FIRST_CONNECTION and RC2 with LOCK_ON_FIRST_CONNECTION: **no delay** in reconnection;
3. FC: iFLight F7 45A AIO - FW: Betaflight => same as above, **no delay**.

I don't know how much is this useful, because it makes no sense; apart from blaming the FC for some communication issues over its UART or that particular R9MM receiver.

After all it's not a big issue, because I don't plan to fly that far with the build n.1, so I do not expect any failsafe.

nangel0, 2021-04-29:
> I made a few tests, always same R9 TX module, and same RX type, the R9MM, but on a total of 4 builds, with mixed parameters (due to different time of testing and such):
> 
> 1. FC: Aribot F7 - FW: Emuflight => same behavior between RC1, RC2 and wether LOCK_ON_FIRST_CONNECTION is enabled or not: about **20 sec delay** in reconnection powering up the TX after an RX-LOSS;
> 2. FC: MambaF722 - FW: Emuflight => I only tried RC2 with LOCK_ON_FIRST_CONNECTION enabled: about **20 sec delay** in reconnection;
> 3. FC: MambaF722-I2C - FW: Emuflight => same behavior between RC1 without LOCK_ON_FIRST_CONNECTION and RC2 with LOCK_ON_FIRST_CONNECTION: **no delay** in reconnection;
> 4. FC: iFLight F7 45A AIO - FW: Betaflight => same as above, **no delay**.
> 
> I don't know how much is this useful, because it makes no sense; apart from blaming the FCs for some communication issues over their UARTs.
> 
> After all it's not a big issue, because I don't plan to fly that far with the build n.1 and 2, so I do not expect any failsafe.
> 
> EDIT: added the last build I have with R9M and ELRS.

How do you wire them? Make sure to use high uart ports, stay away from ports that may be used with or without inversion, they can mess up comunication between fc and rx...

CapnBry, 2021-04-29:
> How do you wire them? Make sure to use high uart ports, stay away from ports that may be used with or without inversion, they can mess up comunication between fc and rx...

The FC makes no difference when it comes to ELRS syncing, we don't even send any data to the FC until after a connection is established. I would say the difference in timing is purely coincidental, unless the FC is doing something insanely weird like spamming us with so much data we can't do any RF work (almost impossible).

If you want to see what is happening @dentex, you can turn on serial passthrough in the betaflight configurator by going to the CLI and typing `serialpassthrough X 420000` where X is the index of your RX serial uart when you type `serial`. Then you should see the RX saying something like:
```
5000
10000
20000
40000
5000
10000
20000
40000
```
Each of those is starting a listen at that rate. Also, if you see those AFTER the initial connection, you don't have LOCK_ON_FIRST_CONNECTION on. But if you could try that and see if there's some other message after the "conn lost" message that occurs when you turn off TX that would be helpful.

dentex, 2021-04-30:
@CapnBry thanks Bryan, I'll try that and report back.

dentex, 2021-04-30:
Turning the radio off, I get the message `lost conn`.
Turning it back on, i get after one or two seconds `tentative conn` and after a fraction of a second `got conn`.

The problem in CLI and serialpassthrough is that I can't arm anyway. Also the addressable LEDs' pattern is different (due to the arming prevention). 

I think that the `got conn` message is basically when I get telemetry back on the radio, after the reconnection.
The problem is that there's this additional delay, during which the addressable LEDs flash with a colored pattern and there's the arming prevention n. 15 as reported here: https://github.com/ExpressLRS/ExpressLRS/issues/508#issuecomment-825788314

CapnBry, 2021-04-30:
That's excellent information. So it sounds like the receiver is fully up and running in a couple of seconds, it is just the FC for some reason is preventing arming after the connection comes back up. I don't have any of that hardware or run Emuflight on anything but I'll see if I can reproduce something similar since I don't think Emuflight's been messing with the RX code at all since they forked.

kusnendar, 2021-04-30:
> Turning the radio off, I get the message `lost conn`.
> Turning it back on, i get after one or two seconds `tentative conn` and after a fraction of a second `got conn`.
> 
> The problem in CLI and _serialpassthrough_ is that I can't arm anyway. Also the addressable LEDs' and the buzzer stay off.
> 
> I think that the `got conn` message is basically when I get telemetry back on the radio, after the reconnection.
> The problem is that there's this additional delay, during which the addressable LEDs flash with a colored pattern and there's the arming prevention n. 15 as reported here: [#508 (comment)](https://github.com/ExpressLRS/ExpressLRS/issues/508#issuecomment-825788314)
> Then the buzzer emits a short beep, the LEDs stop flashing and arming is possible (none of this happens during _serialpassthrough_).

If u ise 50hz & 100hz so you get delay. Try use 200hz

dentex, 2021-04-30:
> If u ise 50hz & 100hz so you get delay. Try use 200hz

Where do you deduce I've used 50 or 100Hz. In fact I've always been on 200Hz.

CapnBry, 2021-05-05:
I've tried on my FM30 and today again with a new EP2 and the RX connection comes back so fast in Betaflight that I can see the LQI rise from at most 37 to 100%. I'm not sure if what you're seeing is an 900MHz thing, an R9 thing, or a Emuflight thing (except for the one that doesn't do it)?

BrandonRC has some 900MHz gear so he's going to test it out as well and see how long it takes for the connection to come back up after restarting the TX.

dentex, 2021-05-05:
>[...] I'm not sure if what you're seeing is an 900MHz thing, an R9 thing, or a Emuflight thing (except for the one that doesn't do it)?

Yes...
> BrandonRC has some 900MHz gear so he's going to test it out as well and see how long it takes for the connection to come back up after restarting the TX.

Great! Looking forward to it.

schugabe, 2021-05-05:
I did some tests with my R9 TX + r9mm RX and betaflight 4.2 running on a matek F722-SE (telemetry enabled, hybrid switches, LOCK_ON_FIRST_CONNECTION enabled).

Test sequence: 
1. Arm quad
2. power off tx wait for FS 
3. switch TX back on 
4. wait unti opentx shows the link
5. rearm quad. 

I did try 200Hz, 50Hz, 25Hz with different tlm ratios. The link is established within 1-2 seconds after TX module plays the startup sounds.

I repeated the tests without LOCK_ON_FIRST_CONNECTION the link is established again but it takes longer (~5 seconds).

brandonrc, 2021-05-05:
@madmiked182 what firmware are you running on that FC? BF? EMU?

brandonrc, 2021-05-06:
@CapnBry 
I cannot replicate this issue. 

Tested on:
Radio: Horus X10s opentx nightly (5/3/2021)
TX: R9 2018
RX: Jumper r900 mini

Domain: EU (@madmiked182 said he was from germany)
Binding Phrase: password123
Lock_on_first_connection
Hybrid switches
telemetry

Setting on lua... 
Tried all power range 10mw to 100mw
Tried all telemetry ratios 1:2 to 1:128

When I power off the tx the rx goes red blinky and when I turn it on.. back to green in less than a second.

CapnBry, 2021-05-06:
I've even taken a flight controller, soldered a receiver to it, flashed it with Emuflight and configured it to test. Cycled the TX ten times and every time the connection came back up and came out of "RX_FAILSAFE" in 1-2 seconds. Sorry @dentex, nobody can reproduce the behavior you're seeing so I'm going to close this ticket. I do not believe there is any issue like you describe systemic to ELRS 1.0. Perhaps you can investigate on your own and discover your own issue. It still sounds to me like the RX doesn't have LOCK_ON_FIRST_CONNECTION enabled and is rate cycling. You can tell if it is doing this using the serialpassthrough command I described above.

closed 2021-05-06 by CapnBry

dentex, 2021-05-06:
@CapnBry Hey no problem, I appreciate your time and efforts spent on this issue.