ExpressLRS/ExpressLRS #509: 1.0.0-RC1 Telemtry only shows 10 elements instead of 14 on previews releases.

alexvazquez, 2021-04-22:
Today I flashed the latest release 1.0.0-RC1 to my TX and RX. On both, I set the ENABLE_TELEMETRY option and also HYBRID_SWITCHES_8. 

Additionally in the TX I enable TLM_REPORT_INTERVAL_MS and set the default of 320LU because my radio uses OpenTX.

Plugged the FC to the battery and turned on my radio and then I Delete all sensors from the telemetry page and started discoverting new ones. My surprise is that now instead of 14 from previews builds I only got 10. For example RxBt is gone.

Am I missing something?

schugabe, 2021-04-22:
The rx sends all telemetry packages it does receive and I did just verifiy that RxBt still is displayed on my test setup (also after deleting + discovering the sensors).

What FC firmware are you using?

alexvazquez, 2021-04-22:
> The rx sends all telemetry packages it does receive and I did just verifiy that RxBt still is displayed on my test setup (also after deleting + discovering the sensors).
> 
> What FC firmware are you using?

Hi @schugabe thanks for your reply. Im using a Mamba F405 Mini MK2 that users FURYF4OSD betaflight firmware 4.2.8 Feb 15 2021. Using this same FC but with 0.1.3-RC2 I can see RxBt (total 14 sensors). If I do the ExpressLRS firmware update to my receiver and TX to 1.0.0-RC1 then after I discover new sensors I can only see 10 and RxBt is not there anymore. So I don't think this could be related to the FC firmware. 

Anything I could try?

schugabe, 2021-04-22:
Can you try to open the expresslrs lua script and set the tlm ratio to 1:16 and then try to discover the sensors?

alexvazquez, 2021-04-22:
> Can you try to open the expresslrs lua script and set the tlm ratio to 1:16 and then try to discover the sensors?

@schugabe smart!! you found the issue. Once I set to 1:16 I got all of them. Beautiful.

Thanks a lot

alexvazquez, 2021-04-23:
> Can you try to open the expresslrs lua script and set the tlm ratio to 1:16 and then try to discover the sensors?

@schugabe I could validate that it worked perfect with Betaflight, but now I use the same receiver but in a Matek F411 WSE FC having INAV firmware and when I do discover sensors I only get 10 sensors and there is no one about battery voltage. Is this an issue or expected?

Thanks

0crap, 2021-04-23:
Yeah, run into the same issue. I had telem stil on 1:64 and lost a few sensors.
After setting TLM to 1:16 they came back, including GPS and a few.

I did notice not all sensors have now the same refresh rate, is that normal?
(The * sign behind the sensors indicate different refresh rates.)

schugabe, 2021-04-23:
> Yeah, run into the same issue. I had telem stil on 1:64 and lost four sensors.
> After setting TLM to 1:16 they came back, including GPS and a few.
> 
> I did notice not all sensors have now the same refresh rate, is that normal?
> (The * sign behind the sensors indicate different refresh rates.)

That is normal: Link status is updated independent: https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry I did record some gifs to show the expected behaviour.

@alexvazquez inav should work as well but I will try it on my inav quad. Can you please change the title of the issue to "Telemetry sensors missing with tlm ratio 1:64" because that seems to be the configuration that does not work

alexvazquez, 2021-04-23:
> That is normal: Link status is updated independent: https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry I did record some gifs to show the expected behaviour. [...]

Title changes @schugabe , let’s see what you get on your INAV quad. Thanks a lot.

CapnBry, 2021-04-23:
I can also confirm that 1:64 does not seem to work on 500Hz, 250Hz, 150Hz rate (all I tested). The RX thinks it is sending it, getting Telemetry confirmations too, so I think it is on the TX side. I'm looking into it and will hopefully have a fix shortly.

alexvazquez, 2021-04-23:
> I can also confirm that 1:64 does not seem to work on 500Hz, 250Hz, 150Hz rate (all I tested). The RX thinks it is sending it, getting Telemetry confirmations too, so I think it is on the TX side. I'm looking into it and will hopefully have a fix shortly.
> 
> EDIT: 1:128 also appears to not be working either

Awesome, appreciate it so much.

schugabe, 2021-04-23:
> @schugabe I could validate that it worked perfect with Betaflight, but now I use the same receiver but in a Matek F411 WSE FC having **INAV** firmware and when I do discover sensors I only get 10 [...]

I did try my inav quad and it shows all tlm values (including gps). So I'm not sure why it's not working for you: try to check every part of the configuration to verify that all options are set accordingly.

alexvazquez, 2021-04-23:
> I did try my inav quad and it shows all tlm values (including gps). So I'm not sure why it's not working for you: try to check every part of the configuration to verify that all options are set [...]

Configuration for the Flight controller? I have Telemetry enabled in the Configuration tab in INAV dont know if I have to set something else. Any clue?

alexvazquez, 2021-04-23:
> Configuration for the Flight controller? I have Telemetry enabled in the Configuration tab in INAV dont know if I have to set something else. Any clue?

I erased the FC and flashed again with the firmware (INAV) and now is working. Very strange. But it worked. Thanks a lot

alexvazquez, 2021-04-24:
With Express Lúa script 1:16 I get i my Taranis every 5 seconds “Lost telemetry” then “Telemetry recovered” and this all time. Very annoying 😫

schugabe, 2021-04-24:
> With Express Lúa script 1:16 I get i my Taranis every 5 seconds “Lost telemetry” then “Telemetry recovered” and this all time. Very annoying 😫

Do you loose control if it reports tlm lost?

Please compare your telemetry page on the taranis with the gif here: https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry the link senors should update in the same interval.

sfjuocekr, 2021-04-24:
If you get "telemetry lost" you failsafe, if you get "sensor lost" you keep control.

closed 2021-04-24 by AlessandroAU

alexvazquez, 2021-04-26:
> If you get "telemetry lost" you failsafe, if you get "sensor lost" you keep control.

I dont feel Im getting failsafe when Telemetry lost message happens on my Taranis. I can keep flying but is annoying to have that message looping every 10 secs.

alexvazquez, 2021-04-26:
> Do you loose control if it reports tlm lost?
>
> Please compare your telemetry page on the taranis with the gif here: https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry [...]

I don't loose control of the quadcopter.

alexvazquez, 2021-04-28:
> I can also confirm that 1:64 does not seem to work on 500Hz, 250Hz, 150Hz rate (all I tested). The RX thinks it is sending it, getting Telemetry confirmations too, so I think it is on the TX side. [...]

@CapnBry Hi, good morning. Do you have any estimation on the fix? Appreciate so much

AlessandroAU, 2021-04-28:
Should be fixed in rc2

alexvazquez, 2021-04-28:
> Should be fixed in rc2

Appreciate all the hard work. Is there any release date for RC2 ? Thanks 🙏

AlessandroAU, 2021-04-28:
Released a few hours ago :)

alexvazquez, 2021-04-28:
> Released a few hours ago :)

Wooow that was fast jaajajajaj. will flash my receivers and TX and test. Again thanks a lot for the great effort. If I can help on something please dont hesitate to let me know.

alexvazquez, 2021-04-29:
Is there a way to get the Cell Voltage in the telemetry?

schugabe, 2021-04-29:
> Is there a way to get the Cell Voltage in the telemetry?

If the flight controller sends cell voltage in the telemetry stream it's received as cell voltage (betaflight has a setting for that). Please use the github issues only for bug tracking and all further questions are better discussed on discord

PR #515 (2021-04-24): Dynamically adjust StubbonSender timeout based on rate
The timeout was set too low by mistake when merging MSP (changed it from 1000 to 100), so 1:64 and 1:128 rates were timing out before the ack came. Fixes #509 (theoretically, but tested)

Now the timeout scales from 40 packets at 1:2 to 5120 packets at 1:128. Note that 25Hz 1:128, StubbornSender will try for over 7 hours and 16 minutes to send the same telemetry item, but what do you expect over a **four bits per second** link?

I also removed the volatile flag from the member variables of StubbornSender. Sure, it is called from ISRs, but  [... cut]
commit f75c3d5e (2021-04-24): Dynamically adjust StubbonSender timeout based on rate (#515)

* Dynamically adjust StubbonSender timeout based on rate
The timeout was set too low by mistake when merging MSP,
so 1:64 and 1:128 rates were timing out before the ack came.
Fixes #509

* Update tests for new failure threshold

* StubbornSender state is the only volatile needed
Since the state guards against access and the other functions
are ISR only or can withstand being missed for one loop

* Remove extra debugging message