ExpressLRS/ExpressLRS #513: [1.0.0-RC1] Ghost Atto LED burns a hole in your retina :-)

0crap, 2021-04-23:
I love my glowing Atto LED, but with 1.0.0-RC1 it just stays bright white.
Not a big deal, but hope it can be fixed in RC2.

Thx!

[image]

AlessandroAU, 2021-04-24:
Confirmed! timing code might need tweaking. I will look into it

0crap, 2021-04-25:
Awesome! Looking forward to it.
FYI, I just flashed "master" branch and that fixed the LED issue.

Only a few compiler warnings still show, but that was also the case before 1.0.0-RC1
Maybe you can have a look at it.

Compiling .pio\build\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\lib9e6\CRC\crc.cpp.o
.pio\libdeps\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp: In member function 'byte extEEPROM::read(long unsigned int, byte*, unsigned int)':
.pio\libdeps\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp:185:51: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
  185 |         communication->requestFrom(ctrlByte, nRead);
      |                                                   ^
In file included from .pio\libdeps\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\extEEPROM/extEEPROM.h:62,
                 from .pio\libdeps\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\extEEPROM\extEEPROM.cpp:58:
C:\Users\Bas\.platformio\packages\framework-arduinoststm32\libraries\Wire\src/Wire.h:100:13: note: candidate 1: 'uint8_t TwoWire::requestFrom(int, int)'
  100 |     uint8_t requestFrom(int, int);
      |             ^~~~~~~~~~~
C:\Users\Bas\.platformio\packages\framework-arduinoststm32\libraries\Wire\src/Wire.h:97:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'
   97 |     uint8_t requestFrom(uint8_t, uint8_t);
      |             ^~~~~~~~~~~
Compiling .pio\build\GHOST_ATTO_2400_RX_via_BetaflightPassthrough\lib99c\MSP\msp.cpp.o

brandonrc, 2021-04-26:
@AlessandroAU I thought I fixed this issue.  Did I not submit a PR. This happens to my ATTO as well on master right now.

brandonrc, 2021-04-26:
https://github.com/ExpressLRS/ExpressLRS/pull/525

Fixed, if you want to merge and test it, we can close this issue out.

brandonrc, 2021-04-26:
@0crap can you pull changes and flash master now.

0crap, 2021-04-26:
Fix confirmed!
Thx for the work gents, I'll close this one.

closed 2021-04-26 by 0crap