ExpressLRS/ExpressLRS #519: R9MX - unable to establish connectivity

FishballFPV, 2021-04-24:
I decided to have this posted since I used both R9MX and JumperMin RX (915hz) set up on the same quad. Issue is below

Issue - "On the R9MX rx, right after unplugging the Lipo and plugging the lipo back the TX and the RX is unable to establish a solid link again, the TX prompts that the "telemetry is critical, recovered, lost"(the warning cycles through it and never connects solid)

Firmware: 1.0.0-RC1
[image]

ELRS Lua Settings: 50mw, telemetry: 1:16 

On the same quad i hook up the Jumper R900Mini and flash using the same parameters and im not experiencing this issue. It seems like i have to leave the Lipo unplugged for more than 2 mins before plugging it back in to get a solid RC link.

apeelme, 2021-04-25:
Today I also tested my fresh build and the same thing happened to me. I was also surprised because I didn’t understand what happened.
TX: R9M
RX: R9Mini
0.1.3RC2
Lua: 200Hz, 1:16, 10mW
[image]

djoe78, 2021-04-26:
Same issue I have r9m 2019 and r9mm 1.0.0 RC1. When unplugged the lipo and plug it back, the rx and tx are unable to establish connection again.

mdj36, 2021-04-26:
The same issue: 
Same issue I have r9m 2019 and r9mm 1.0.0 RC1. When turn off the radio or lost connection for poor signal, the rx and tx are unable to establish connection again. For establish connection again I power reboot the drone.
Then "LOCK_ON_FIRST_CONNECTION" is selected.

FishballFPV, 2021-04-26:
So i tried to play around with the set up and i did attempt to just power cycle the TX(off and On) still the same thing. it just cycles through RF signal Critical, connected and Lost. 

I really had to wait for more than 2 minutes before plugging the lipo to establish a solid RC Link between the RX and TX for some reason.

rodrigokava, 2021-04-26:
I am testing ELRS with TX R9LITE and RX R9MX 1.0.0 RC1
When plug Lippo the communication take a long time, like 10/20 seconds. LQ number stay jumping, maybe is searching or trying to sync something. After 20 seconds connects solid. In older version of ELRS was super fast.

FishballFPV, 2021-04-27:
So i managed to reach out to our Discord Help and Support chatroom and below is one suggestion. No idea what 3v3 i was looking for but let me know if you guys have any idea about what ballistrc is referring to. i was at work and had a hard time keeping up with the chat.

[image]

apeelme, 2021-04-27:
> So i managed to reach out to our Discord Help and Support chatroom and below is one suggestion. No idea what 3v3 i was looking for but let me know if you guys have any idea about what ballistrc is referring to. i was at work and had a hard time keeping up with the chat.
> 
> [image]

He is thinking of the CPU power supply (3.3V). The FC has a 3.3V pad (Spectrum receivers are commonly used). Shorting this with  GND.

FishballFPV, 2021-04-27:
Thanks for pointing that out..all this time i was looking at the RX, he meant the FC.. ill try it out later. Thanks again..i just hope if it works - i dont have to keep on shorting the 3v and Gnd every time  >.<

closed 2021-04-28 by FishballFPV

apeelme, 2021-04-28:
> Today I also tested my fresh build and the same thing happened to me. I was also surprised because I didn’t understand what happened.
> TX: R9M
> RX: R9Mini
> 0.1.3RC2
> Lua: 200Hz, 1:16, 10mW
> [image]

working fine with RC2! thanks guys!

P2FPV, 2021-08-14:
Hi im trying to flash r9mx but its not working its giving me error 2 via stlink do i need to flash bootloader for stlink method please help