ExpressLRS/ExpressLRS #521: ELRS Telemetry missing

mrk55319, 2021-04-25:
I have a Matek 722-STD FC running BF 4.2.8 and an R9M 2018 and R9MM flashed to ELRS 1.0.0.0-RC1. For ELRS configuration I have hybrid switches and telemetry enabled. My transmitter is an older Taranis X9D running the special version (2.3.10) of OpenTX. My problem is that I am not seeing all the telemetry. I've tried various combinations of packet and telemetry rates and am not seeing any additional sensors being discovered. This was also a problem on 0.1.3-RC2. 

Also note that RxBt appears to not be 
scaled properly. The FC was reporting 15V.

[image]

CapnBry, 2021-04-25:
We do not modify the telemetry values, we only pass through exactly the data that comes from the flight controller. Make sure you don't have any scaling on the opentx telemetry item maybe?

As far as missing items, again, we send everything we get. Make sure your telemetry ratio is 1:16 or faster (up to 1:2) due to a current bug in RC1, but if they don't show up, we're not getting them from the FC.

mrk55319, 2021-04-25:
Thanks for getting back to me. Interesting...when I use an Frsky XSR and Smartport I get all telemetry. Maybe I’m not setting up the FC correctly for ELRS, but there doesn’t appear to be too many knobs for me to adjust.

CapnBry, 2021-04-26:
Yeah you're right if you're using latest betaflight, opentx nightly, and ELRS 1.0.0-RC1 there's really not any knobs to adjust. You seem to be missing **a lot** of sensors though, which seems impossible to me, and a value of 3dB for 1RSS is not valid. There should be **nine** link telemetry sensors, three roll/pitch/yaw sensors, four battery sensors, and one flight mode. All nine link telemetry sensors are generated by ELRS itself, and all are sent to opentx in a single CRC-protected transmission. It simply is not possible that you'd only get one, and that the value would be wrong.

Have you tried creating a blank new model in opentx, set it to External CRSF, then go discover sensors? I just can't think of anything we could be doing on our side to create the behavior you're seeing.

mrk55319, 2021-04-26:
Hey Bryan, Thanks for your help. I updated with the nightly and created a new model and now see all the sensors. Best, Mike

From: Bryan Mayland
Sent: Monday, April 26, 2021 7:34 AM
To: ExpressLRS/ExpressLRS
Cc: mrk55319; Author
Subject: Re: [ExpressLRS/ExpressLRS] ELRS Telemetry missing (#521)

Yeah you're right if you're using latest betaflight, opentx nightly, and ELRS 1.0.0-RC1 there's really not any knobs to adjust. You seem to be missing a lot of sensors though, which seems impossible to me, and a value of 3dB for 1RSS is not valid. There should be nine link telemetry sensors, three roll/pitch/yaw sensors, four battery sensors, and one flight mode. All nine link telemetry sensors are generated by ELRS itself, and all are sent to opentx in a single CRC-protected transmission. It simply is not possible that you'd only get one, and that the value would be wrong.
Have you tried creating a blank new model in opentx, set it to External CRSF, then go discover sensors? I just can't think of anything we could be doing on our side to create the behavior you're seeing.

CapnBry, 2021-04-26:
Awesome news, Mike! I'm really not sure how to describe the behavior you were seeing but it is good to see on the latest everything works out. I am going to close this issue and you can reopen it if there are any other problems related.

closed 2021-04-26 by CapnBry