ExpressLRS/ExpressLRS #522: [1.0.0-RC1] Lua script briefly throws a warning on first start

0crap, 2021-04-25:
When I first start the ExpressLRS Lua script after I boot my radio (Taranis QX7S), it always briefly throws a warning before it continues. That is a bit annoying.
After that is functions as normal.
Pre 1.0.0-RC1 Lua script did not have this issue.

To my understanding it's some kind of version checker code that has to do with this.
For example in Betaflight you know you need a new Lua script that goes with every new major version, as the documentation states.
So IMHO totally fine to remove this code from the ExpressLRS Lua if that fixes the issue.
Makes it less complicated at the same time.

pitts-mo, 2021-04-29:
Hello @0crap,
I see the example below briefly on launch of ELRS lua from the tools menu of TX16S OpenTX 2.3.11 release. 

Does this seem consistent with what you are reporting?
-p

# begin example
ExpressLRS ****** 0:0

!!! VERISON MISMATCH !!!
Update TX module
LUA v0.3, TX?
[force use]
# end example

0crap, 2021-04-30:
Yes, that's the warning that flashes shortly.

I read a comment from @CapnBry that the Lua script is getting a rework done.
Maybe he can tell if that will be ready before the final ExpressLRS v1.0.0 release?

CapnBry, 2021-04-30:
No, the new configuration protocol and LUA script for it is a planned v1.1 feature. We aren't adding any more features to 1.0, only bug fixes.

0crap, 2021-04-30:
Fair enough.
I do hope you will fix the short warning on Lua start for v1.0
Thx!

0crap, 2021-05-01:
Fix is merged merged and I already tested it.
Works great! So I'll close this issue. 
Thx! 👍

closed 2021-05-01 by 0crap