ExpressLRS/ExpressLRS #532: 1.0.0-RC1 - Include battery voltage to link telemetry

phobos-, 2021-04-26:
In v0.1.3-RC2 basic telemetry sends voltage together with link information.
In v1.0.0-RC1 basic telemetry sends only link data, to enable battery voltage sensor I have to compile with -DENABLE_TELEMETRY.

I really liked the original behavior for racing. Now I have to use -DENABLE_TELEMETRY to have voltage information, but even then battery voltage is not updated as frequently as before (every telemetry packet).

pitts-mo, 2021-04-29:
I was thinking along some similar lines...

Maybe add receiver specific build options to enable each specific telemetry option that is desired... Or perhaps a pre-defined -DENABLE_MINIMAL_TELEMETRY build option if all us low latency enthusiast can agree on what should be included.

From my perspective it seems ideal to build all telemetry elements into RX by default and use a LUA to toggle and save the desired telemetry elements per receiver.
-p

0crap, 2021-04-30:
> From my perspective it seems ideal to build all telemetry elements into RX by default and use a LUA to toggle and save the desired telemetry elements per receiver.
> -p

You can already flash some receivers with and some without Telemetry and use it on the same TX.
https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry
But the Lua toggle sure would be nice.

schugabe, 2021-04-30:
> In v0.1.3-RC2 basic telemetry sends voltage together with link information.
> In v1.0.0-RC1 basic telemetry sends only link data, to enable battery voltage sensor I have to compile with [...]

What update rate for a voltage sensor would you consider as "good enough"? 

I would think that a update in the range of seconds is fast enough for such a sensor and I can't really think of a usecase that requires real fast updates for the sensor.

brandonrc, 2021-05-09:
@schugabe 

in 0.1.3-RC2 you still need the telem ratio in order to get that battery voltage correct? 

Is there any performance difference between 0.1.3-RC2 and 1.0.0-RC3 with -DENABLE_TELEMETRY as long as both 
0.1.3-RC2 and 1.0.0-RC3 have the same ratio (1:128) for example.

schugabe, 2021-05-10:
> @schugabe
> 
> in 0.1.3-RC2 you still need the telem ratio in order to get that battery voltage correct?
> 
> Is there any performance difference between 0.1.3-RC2 and 1.0.0-RC3 with -DENABLE_TELEMETRY as long as both
> 0.1.3-RC2 and 1.0.0-RC3 have the same ratio (1:128) for example.

Yes you need tlm ratio in 0.1.3-RC2 since the battery voltage is embedded in the link package.

The radio link performs the same since the number of tlm messages does not change with -DENABLE_TELEMETRY. The only difference is: you also need to enable hybrid switches and the update rate of the telemetry sensors depends on the tlm ratio/frequency settings and the number of sensors the fc is sending (GPS is only sent if a module is connected).

A "battery only" telemetry mode would allow ~3Hz update rate of the full set of battery sensors at 500Hz 1:64 tlm ratio. This would be easy to implement and should be fast enough for most use cases.

CapnBry, 2021-05-16:
I thought I had already posted to this Issue and I can see a marginal benefit to this, but I'm not sure why it would be needed.

As schugabe pointed out, you need to have a telemetry ratio set for any telemetry to be sent. At 500Hz / 128, that's ~4x packets per second, split as 2x LINK type and 2x TELEM type giving you two packets per second of each type. Betaflight can be set to only send Battery telemetry so that guarantees both packets are of type battery. Battery telemetry is 2 byte header + 8 byte payload + CRC, so 11 bytes divided by our 5 bytes per telem packet, that's 3x packets needed, or 1.5s per update of the battery info. You can't turn the telemetry ratio any lower than this so worst case you're getting battery voltage every 1.5s instead of 4x a second meaning this change would get you 1200ms of faster battery level updates. Is that worth it?

At 200Hz 1:128, you still need 3x TELEM packets so 6x total telemetry packets, or 3.75s between updates instead of 0.64s or 3.11s faster per update.

I feel like adding this is of marginal benefit because you don't need sub-second update rates for battery level at the TX, right?

brandonrc, 2021-12-09:
Telemetry functionality has changed since this original issue. Based on https://github.com/ExpressLRS/ExpressLRS/pull/769

I am closing this issue. 

If we want to change the way our telemetry works please open a new enhancement request.

closed 2021-12-09 by brandonrc