ExpressLRS/ExpressLRS #544: add RX Receiver Number / Model ID support

pitts-mo, 2021-04-29:
Please evaluate for the possibility to add RX Receiver Number / Model ID support.
Thank you,
-p

Reason:
I have long found RX Number to be a helpful safety feature to prevent accidental operation of a model while an incorrect model profile is selected within the transmitter. 

A modern usage example:
Utilize one model ID for binding fixed wing flight controllers and a different model ID for binding multi rotor flight controllers.

Note: 
OpenTX only added working RX Number handling for the CRSF protocol in the 2.3.10 release.
**https://github.com/opentx/opentx/pull/7881** 

Related thought:
I expect RX Number would primarily be utilized during traditional binding of an RX but it does seem like it could be helpful to allow setting of an RX Number at time of build along with the binding phrase.

CapnBry, 2021-04-30:
I'm just adding my notes here to this so I can remember for later. I think this won't be too difficult to implement.

```
#define UART_SYNC                      0xC8
#define MODULE_ADDRESS                 0xEE
#define COMMAND_ID                     0x32
#define SUBCOMMAND_CRSF                0x10
#define COMMAND_MODEL_SELECT_ID        0x05

// CRSF command sent to select a model
{ UART_SYNC, 8 (len),
  COMMAND_ID, MODULE_ADDRESS, RADIO_ADDRESS, SUBCOMMAND_CRSF, COMMAND_MODEL_SELECT_ID,
  model_id, crc8_BA(2, 6), crc8(2, 7) }
```
We can then just XOR the model_id with the binding UID. This could also be baked in to a receiver using a new define like `MY_BINDING_MODELID`. I am not sure if that gets too confusing that you don't want to flash a TX with a modelid otherwise it can only talk to that one model and changing the modelid in OpenTX would break all communication. Maybe just have MY_BINDING_MODELID be an RX-only usage.

pitts-mo, 2021-04-30:
This sounds good @CapnBry.
-p

In case this helps... On recalling the behavior of my TBS gear when this first became available in OpenTX I suspect that:
1- TX should handle model ID as pass-through and ignore(null) or only insert a master Model ID when it has not recieved a Model ID from OpenTX. 
2- RX only needs to sync when model ID of incoming signal matches the one at binding OR the Model ID at binding was the null/master Model ID. 

more background detail:
After OpenTX 2.3.10 update my receivers initially kept right on working no matter what I changed the Model ID to. Then after I re-bound each of my receivers I found they would only work on the Model ID set in OpenTX at time of binding... Hence why I believe TX handles this as pass-through.

oryfoxer, 2021-06-30:
I run a single model for ELRS - Please do not model match. This would create overhead for no reason. Crossfire doesn't even model match.

If you do implement model match, please have an option to disable.

CapnBry, 2021-06-30:
> I run a single model for ELRS - Please do not model match. This would create overhead for no reason. Crossfire doesn't even model match.

Crossfire actually does have model match (model num or model id or whatever). You won't have to use it so long as you don't change the model number in OpenTX. This would only be implemented for RX that use traditional binding in my mind, since changing the modelid otherwise would require a recompile and upload to the RX which is extremely inconvenient (at least until we have over-the-air RX configuration somewhere down the line).

Dropsleutel, 2021-08-07:
I'd like the way deviation Tx handles the modelmatch.

Basically they take unique ID Nr (from the transmitter CPU) + a user defined bind ID a Nr ( or in case of ELRS a bind Phrase), this bind ID part/stored in each model file.    The combination (Tx ID + bind ID)  is then bound to the Receiver.  

Benefits;
No accidental, flying wrong models.(modelmatch)
Or keep all models same binding phrase.  Setting the same bind ID NR/ Phrase in each Model, it will just operate all models.
Sharing modelfiles and bind phrases to other pilots, does not result in field collission, as also the Unique Tx ID is used.

In deviation if you own multiple transmitters, you can override the unique Tx ID, and set a fixed Tx ID in Tx.ini file in that case you could also operate from multiple transmitters same model.

closed 2021-08-20 by AlessandroAU