ExpressLRS/ExpressLRS #553: Close Range Failsafe with r9m+r9mini v1.0 RC2

RD5500, 2021-05-01:
i'm having this strange problem where the quad would just failsafe for no reason. as you can see in the video, it failsafed even though its only 5 meters away. earlier today i flew my first pack with the new v1.0 RC2 firmware at my usual flying field and the same thing happened, but this time it happened at 50 meters away. so i dont think the failsafe is related to distance, perhaps its bug? i also notice th lq keeps randomly dropping from 99 to 88 and then lq warning starts to appear. i'll stick with v0.13 RC2 for now.

oh, i tried using nightly build opentx as suggested in the wiki, but this doesnt seem to fix the problem anyway so i'll install the older custom fw v2.3.10 for now

im sorry for the sh*tty vid, i didnt record any dvr too because i thought the problem was solved. the video is pretty much what i can see in my goggles when i was flying lol

settings used for the flight in the video:
100mw, 100hz, 1/64 tlm, 115k baudrate locked, adc filter off
settings used earlier today:
100mw , 200hz, 1/64 or 1/32 tlm (i forgot), rest is the same
radio: Jumper T12 Pro with R9M 2018 (ort moxon v1) and R9mini rx (immortal t)
https://user-images.githubusercontent.com/83501592/116783269-0a649380-aab8-11eb-9871-aaf6db01bc58.mp4

rashmandash, 2021-05-03:
Hi,

I put the 1.0.0-RC2 Release on my R9m[TX] and R9mm[RX].  
i did not flew jet, only in my appartment but i get RssI warnings and telemetry lost already.

I just want to know if other people have the same problem. 

I testet it with two different RX's and both are the same. 
I followed more ore less the Joshua Bardwell video.

Thx 
Cheers Rashmandash

DroneQuad, 2021-05-04:
> Hi,
>
> I put the 1.0.0-RC2 Release on my R9m[TX] and R9mm[RX].
> i did not flew jet, only in my appartment but i get RssI warnings and telemetry lost already.
>
> I just want to know if other people have [...]

same like me, try reflash RX and TX, for TX use the "TLM_REPORT_INTERVAL" try default 320LU and "ENABLE_TELEMETRY"
if the issue same like i've flown my quad and no failsafe even that telemetry warning said often

mg-minchev, 2021-05-07:
Hello. I've got same problem since RC1. Even on 100mW at visual distance less than 100 meters LQ moving like crazy from 5:100 to 5:70 and OSD LQ text start blinking. Now I'm going back to release 92df0d from master, because I had no issues with it no matter it is quite "old".

RD5500, 2021-05-08:
> Hello. I've got same problem since RC1. Even on 100mW at visual distance less than 100 meters LQ moving like crazy from 5:100 to 5:70 and OSD LQ text start blinking. Now I'm going back to release 92df0d from master, because I had no issues with it no matter it is quite "old".
> 
> Edit: the release before 92df0d which is the tagged release.

yup this is basically the same problem im having. lets just see what the next release would be like :)

CapnBry, 2021-05-09:
Just to clarify, this issue is not about OpenTX saying "telemetry lost" "telemetry recovered", that we'll consider separate as they can be unrelated.

We have had several reports now of 868/915MHz users just dropping off entirely for no apparent reason and switching back to 0.13 seems to resolve it. It is difficult to say where this issue lies because of the number of changes since 0.13 and the fact that we can't reproduce the issue (although Jye may have experienced it once). We are investigating a number of possible fixes, but it does seem to only be an issue on SX1276 hardware (858/915) at this time.

Reports
* Rogero: T16, R9M 2018 with Resistor Mod and Fan Mod, always at 420000 with LUA showing 0:200, everything was working perfect before no problems ever, that night i updated both OpenTX to nightly build 2.3.12N488 and ELRS to 1.0.0 RC2, before that i had OpenTX 2.3.10 ELRS build with ELRS 0.1.3 RC2. 915 FCC and using Hybrid switches, Arm on Aux1 for the record. (LQI 99, 0, no recovery) [Discord](https://discord.com/channels/596350022191415318/797109686285107241/838417152486408232) x [Youtube Video](https://youtu.be/APA-sXkZToo)
* Kengaboonga: X9D+ with R9M + resistor mod 4000000 baud 100mW, Super 8 antenna but "standing like a tree" so not bumping antenna. (LQ 98, 91, 44, 1, 0, recovers)  [Discord](https://discord.com/channels/596350022191415318/596350022191415320/840669350390530148) x [Streamable Video](https://streamable.com/v3f521)
* JrMinLinFPV R9M 200Hz 25mW (LQ 100, 92, 32, 12, 0) [Youtube Video](https://www.youtube.com/watch?v=d0mnCpSJmcA)

coryking, 2021-05-10:
For what it's worth, I too have an issue with the link just dropping out. 1.0.0-RC3. Like, I'll be literally LOS less than 50m away and the darn thing just drops right out of the sky. It will not re-establish the link until I power-cycle the quad. This is brand new hardware too, the old hardware did the same thing. This is totally random too--I can get several packs in with zero problems and then boom... the quad drops from the sky like it was shot with a gun.

R9M (2019) + R9MM.  Stock antennas on  both ends.

Output power doesn't seem to make a difference, though I've never ventured beyond 250mw. Interference can't really be it as I've experienced this in several locations.

Running whatever the patched version of OpenTX you guys offer. Installed on a QX7.

Not really sure how to troubleshoot this because I'm completely unfamiliar with this sort of thing. I have no clue where to even begin diagnosing what causes a link failure.

Rogero-, 2021-05-10:
> For what it's worth, I too have an issue with the link just dropping out. 1.0.0-RC3. Like, I'll be literally LOS less than 50m away and the darn thing just drops right out of the sky. It will not [...]

> For what it's worth, I too have an issue with the link just dropping out. 1.0.0-RC3. Like, I'll be literally LOS less than 50m away and the darn thing just drops right out of the sky. It will not [...]

just roll back to release version 0.1.3 RC2 and the link should work fine, apparently a bug in latest releases that is being investigated...

AlessandroAU, 2021-05-11:
Identified a possible reason and fix, it's in this branch 
[image]

CapnBry, 2021-05-11:
Alessandro and I have been working on this almost exclusively over the past few days, and hopefully the changes just pushed to that branch (more changes since his post above) pass our testing. If it does then we'll post a new RC for general testing.

CapnBry, 2021-05-11:
> i've tested the patch earlier today, armed 3 times and not a single random failsafe :) just gotta wait for others to also confirm if the patch fixed the bug. Keep up the good work guys!
> lua setup i used today: 200hz 100mw 1/32 tlm

If you're going to be doing any other testing, would you mind updating to the latest again? Alessandro identified another possible issue that would prevent reconnecting if the connection is dropped and there was also a small issue in the packet phase lock addressed as well (which was added only in this branch by earlier changes).

CapnBry, 2021-05-11:
> i notice that when i turn down the packet rate to 25hz the lua say 0:40 instead of 0:25 on any tlm setting, possibly a bug? not that i would fly with that low packet rate anyway haha
> 
> edit: wait i just realized, i just do my usual range check by walking with 10mw output on the tx. if there's nothing wrong, when i lose signal i should be able to reconnect with the rx, right?

To answer the first question, it is because opentx can't go as low as 25Hz. So it sends packets at 40Hz

And yes you should be able to regain connection after it is lost, is it not reconnecting? Alessandro thought he had that fixed.

Rogero-, 2021-05-11:
just a remark to add here about LQ, even on 0.1.3 RC2 the LQ was always changing quickly while testing it from 2 days with many packs, it dropped to 80 in about 100m away from me on 250mw / 200Hz but 0.1.3 Rc2 never had a failsafe so far for me, i tend to ignore the LQ lately and just enjoy my flight, maybe in latest patches LQ will be more consistent, thx for looking into the problem guys, your efforts are always appreciated.

coryking, 2021-05-11:
I might be able to test this out over the weekend. Hopefully it will work!

RD5500, 2021-05-12:
> To answer the first question, it is because opentx can't go as low as 25Hz. So it sends packets at 40Hz
>
> And yes you should be able to regain connection after it is lost, is it not reconnecting? [...]

good news! it regains connection easily. also is it just me or did you guys improved the r9m link? because i got close to 100m without losing connection, and its also not in line of sight (block by several house). my r9m has never done this before, with the frsky firmware it has to be at least 500mw to reach the same spot. even with the v0.13 firmware i had to set it to like 100mw. great job!🙌

rashmandash, 2021-05-13:
Hi, 
first of all thx to the developers for the work you all put into this project!!! I appreciate it very much!

I flashed my RX with this settings:
[image]

My TX with:
[image]

and 200Hz; 1/32; 100mW.

I still get the telemetry lost, telemetry recoverd and low RssI massage.

Or should i do an update with the OpenTX-NB aswell?

Cheers Rash

deadbytefpv, 2021-05-15:
To those still having this issue even with 1.0.0-RC5, could you guys include into your report the OpenTX version you have in your radio, and what radio it is?

Just so we can verify it is indeed on ExpressLRS... 
Thanks!

jamoos, 2021-05-17:
> To those still having this issue even with 1.0.0-RC5, could you guys include into your report the OpenTX version you have in your radio, and what radio it is?
> 
> Just so we can verify it is indeed on ExpressLRS...
> Thanks!

Just upgraded to rc5 from rc1 that was working quite good. Getting terrible range. Leas then 100m in a open field that before i could go quite beyond.
Its a jumper t16 with access mod running elrs open tx build. 2.3.10
R9m 2019 tx and r9mm rx.

AlessandroAU, 2021-05-19:
Can yo go back to RC1 and confirm it's fixed on that release?

GeoffChudley, 2021-05-19:
I am now in a position to be able to do some testing on this. So what setups and results would be of benefit? TX16S with R9MX and R9M 2019

CapnBry, 2021-06-05:
Anyone experiencing RX lockups on R9MM / R9 Mini should try RC8 where I've identified one major issue and can't find anything else wrong now. This issue will be closed soon as resolved by RC8 unless I hear more reports of RX lockups that require a complete power cycle of the RX to resolve.

jurgelenas, 2021-10-04:
Is this still relevant or we can close the issue?

rashmandash, 2021-10-04:
Can be close.

017662371129

closed 2021-10-04 by AlessandroAU

GeoffChudley, 2021-10-04:
As far as my issue it can now be closed thanks to all
Geoff

AngelisFpv, 2022-04-25:
> Hi,
>
> I put the 1.0.0-RC2 Release on my R9m[TX] and R9mm[RX]. i did not flew jet, only in my appartment but i get RssI warnings and telemetry lost already.
>
> I just want to know if other people have [...]

Yes I'm having the same issue and don't know how to fix it