ExpressLRS/ExpressLRS #554: [1.0.0-RC2] Ghost Atto LED, link color

0crap, 2021-05-01:
When the Ghost Atto RX is powered, without the TX turned on, it flashes Red, Blue, Green, which is perfect.
When the TX is turned on and a link is established the LED turns into a solid Red.

Red is usually the color to tell something is wrong, which is not the case.
Can the Ghost Atto Led be a solid green when the link is established?

When this is considered being to anal, please feel free to close this issue. ;-) 👍 

[image]

CapnBry, 2021-05-01:
Try a different packet rate 😉

0crap, 2021-05-01:
> Try a different packet rate 😉

Brilliant! Didn't see that one coming. 👍  
Case closed!

closed 2021-05-01 by 0crap