ExpressLRS/ExpressLRS #557: Flight Mode - !ERR*

nightflasher, 2021-05-01:
[image]

As you can see, I have this message on my second quad. I also had problems with link quality in osd, which disappeared after flashing new betaflight (4.2.9 over 4.2.6). This happens on my OMNIBUS_F4_PRO (OMNIBUSF4SD target), diy 20x20 868MHz RX on UART6.

Another receiver (same type) is working with flight modes.

CapnBry, 2021-05-01:
We don't make that value, that's created by the flight controller and we pass it through unmodified. The flight controller is telling you "ERROR".

nightflasher, 2021-05-02:
Okay, I've checked everything and suddenly the FM changed to "WAIT", and after a while outside (bc of gps) I got the flight modes right.
Don't know why someone did this, but now I know what is happening.
Thank you!

closed 2021-05-02 by nightflasher