ExpressLRS/ExpressLRS #564: Telemetry Warning

DroneQuad, 2021-05-03:
Hi, does anyone have telemetry issues?
after flashing to R9m and R9slim+ ota, i've got telemetry warning every 3-15sec, even when fly, still fly around 50m max no failsafe still.
My set
200Hz
1:2
10-100mw same issue

schugabe, 2021-05-03:
What TX do you have?

Can you compare the first gif here with the telemetry page in opentx? https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#telemetry

The * next to 1RSS and the following sensors should blink in a regular interval.

DroneQuad, 2021-05-04:
[image]
I used x9d+ thats telemetry value i've got. Any suggestions?

DroneQuad, 2021-05-04:
When i checking again, telemetry warning came from RSNR, got 8 max and lost warning offten

deadbytefpv, 2021-05-04:
Have you tried using "TLM_REPORT_INTERVAL_MS" set to 100LU on the ExpressLRS Configurator?

DroneQuad, 2021-05-04:
i reflash again  R9 RX and TX with 
TX "TLM_REPORT_INTERVAL_MS (320LU)" and "ENABLE_TELEMETRY"
RX "ENABLE_TELEMETRY and "USE_DIVERSITY
still got telemetry warning but not often as before
Any downside when i set "TLM_REPORT_INTERVAL" to 100LU?

deadbytefpv, 2021-05-04:
> i reflash again R9 RX and TX with
> TX "TLM_REPORT_INTERVAL_MS (320LU)" and "ENABLE_TELEMETRY"
> RX "ENABLE_TELEMETRY and "USE_DIVERSITY
> still got telemetry warning but not often as before
> Any downside when i set "TLM_REPORT_INTERVAL" to 100LU?

Nothing really.. It just sends the Telemetry data at that interval (100Ms) to OpenTX instead of the usual 320Ms (the telemetry ratio doesn't really matter here).. some handsets/radio just happen to sometimes require a more frequent telemetry updates.

The x9d series of handset is kinda wonky at the moment (not sure if it has to do with OpenTX Nightly build).. but don't take my word for it.. just kinda  guessing here..

DroneQuad, 2021-05-04:
i'll try 100LU, and test list like before, will be report soon after getting the result...

schugabe, 2021-05-04:
> i'll try 100LU, and test list like before, will be report soon after getting the result...

Can you try to disable tlm in the lua script and check if the warning is still issued? And what happens if you flash without the telemetry option? Both with the default setting of TLM_REPORT_INTERVAL_MS of 320LU

Nordle420, 2021-05-05:
I own almost the same HW, just old accst stuff. I had similar problems until 1.0.0-Rc2, you're on that?

DroneQuad, 2021-05-05:
After testing the problem was less i guess, and this is a good progress about ELRS on FRsky hardware
i use R9m 2019 TX, R9slim+ ota 
flash it TX with "TLM_REPORT_INTERVAL_MS" 100LU and "ENABLE_TELEMETRY"
RX "ENABLE_TELEMETRY and "USE_DIVERSITY
Lua set to 10mw, 50hz, tlmratio 1:16
i've got 15 source telemetry on my radio after delete and discover telemetry.

DroneQuad, 2021-05-07:
Problem fix when update to 1.0.0 RC3

rashmandash, 2021-05-07:
Hi,

i have the same Problem. I'm on the new 1.0.0 Rc3 version. 

HW:
x9d+
R9 accst module

rashmandash, 2021-05-07:
> Problem fix when update to 1.0.0 RC3

Ok, what are your settings? :)

alexvazquez, 2021-05-07:
> Problem fix when update to 1.0.0 RC3

I had same warning issues with RC2, will update right now to RC3 to check if thats fixed. Thanks alot

rashmandash, 2021-05-13:
Hi, 
finally i found the time to test and write the results. :)

I compiled the OpenTX-NB-Companion-file with this Version [[companion-windows-2.3.12N497.exe - 13-May-2021 02:05 - 17332163](https://downloads.open-tx.org/2.3/nightlies/companion/windows/)] and flashed my TX with the normal/official OpenTX-Companion. 

Also i used the latest SD-Card version [[sdcard-212x64-2.3V0035.zip - 08-Jan-2021 14:27 - 118300379](https://downloads.open-tx.org/2.3/release/sdcard/opentx-x9d%2B/)] of the normal/official OpenTX-Companion.

Run the [r9m_elrs_bl.frk](https://github.com/ExpressLRS/ExpressLRS/wiki/Flashing-R9M-TX) file on my TX. 

Flashed my RX with this settings:

[image]

My TX with:

[image]

and 200Hz; 1/32; 100mW and tried all the others. 

I still get the telemetry lost, telemetry recoverd and low RssI massage.

Cheers Rash

alexvazquez, 2021-05-13:
> Hi,
> finally i found the time to test and write the results. :)
> 
> I compiled the OpenTX-NB-Companion-file with this Version [[companion-windows-2.3.12N497.exe - 13-May-2021 02:05 - 17332163](https://downloads.open-tx.org/2.3/nightlies/companion/windows/)] and flashed my TX with the normal/official OpenTX-Companion.
> 
> I also used the latest SD-Card version [[sdcard-212x64-2.3V0035.zip - 08-Jan-2021 14:27 - 118300379](https://downloads.open-tx.org/2.3/release/sdcard/opentx-x9d%2B/)] of the normal/official OpenTX-Companion.
> 
> Run the [r9m_elrs_bl.frk](https://github.com/ExpressLRS/ExpressLRS/wiki/Flashing-R9M-TX) file on my TX.
> 
> Flashed my RX with this settings:
> 
> [image]
> 
> My TX with:
> 
> [image]
> 
> and 200Hz; 1/32; 100mW and tried all the others.
> 
> I still get the telemetry lost, telemetry recoverd and low RssI massage.
> 
> Cheers Rash

I can confirm that exactly the telemetry lost warnings still happens. Hope someone can fix this.