ExpressLRS/ExpressLRS #568: Multiple transmitting module on one board

annoyinganime, 2021-05-05:
This is a feature request: is it possible to make one module with two transmitters onboard (915 and 2.4) and switch between them, i.e. with the script? It will be really awesome to be able to fly at 2.4ghz on something like whoops and 3" toothpicks, and then switch to 915mhz to fly tens of kilometers away on 7" or a plane. I know, that ExpressLRS on 2.4 can already achieve those distances, however interference in cities from wifi hotspots is huge even for lora, and can cause problems, and 915mhz has much lower risk of loosing signal

brandonrc, 2021-05-09:
@annoyinganime I really like the idea of a dual band system but with our 2.4 you can go 33km+ on 250hz 100mw LOS. 

If you reduce that to 50hz with 250mW I am sure you can get more range that would help penetration. 

Not sure this enhancment is really needed unless someone wants to fly 100km+

annoyinganime, 2021-05-09:
The critical thing here is "LOS". Sometimes there is no option to fly LOS, and you need greatest penetration possible, so you fly at 900, and sometimes you're flying close enough so you don't care about LOS

brandonrc, 2021-05-09:
So LOS is just a way to measure the max range. When it comes to penetration of walls you can think of a wall as a rssi value. You will lose X rssi when going through a wall. 

The fact that we can go 33km on 2.4ghz at 100mw with 250hz shows that we can probably fly a bando without working about penetration at all. 

Also, if not LOS then what video feed are you using. ELRS 2.4 LoRa will out run 5.8 analog, 2.4 analog, and 5.8/2.4 DJI digital. 

Do you plane to use 1.3ghz video feed? 

Again, I really like the idea of a dual band system but I am struggeling to find a use case of 900mhz system. ELRS is not like TBS. ELRS is not like FrSky. You do not need 900mhz for 99.9 percent of flying, unless you want 100+km or go through 10 buildings. With that last part... I have no idea what video feed would keep up with our 2.4ghz link.

pitts-mo, 2021-05-10:
Some of my initial thoughts about multiple ELRS bands on a single module:

-Antenna management seems like it could be a headache.
-Does CRSF have an existing framework to support radio selection that could be added under CRSF in OpenTX models?
-Maybe a new ELRS protocol to select from available radios in OpenTX models?
-Can DIY-Multiprotocol-TX-Module somehow allow CRSF pass through to 2.4ghz / 900mhz sub modules?

Summary thoughts:
I could imagine this may still be desirable for some users even if we needed to select the desired radio in a LUA script... However, finding safe and effective ways for radio and module manufactures to deal with the antenna management required for long range seems like the biggest hurdle to me.

-p

annoyinganime, 2021-05-10:
@brandonrc, I'm mostly concerned about 2.4 wifi interference: you can easily outrun your 2.4 RC link with 5.8 video if there are dozens of wifi hotspots near you

annoyinganime, 2021-05-10:
@pitts-mo about antenna management: you don't need antenna-managing circuit if you just add two antenna ports. But I thought about better solution, that can also cover frequency selection: on minimal power modules won't fry themselves, even without antenna, so you can easily power them both on, search wich receiver is active (because you won't be flying both systems at once) and turn the other module completely off. Selection through lua script is also an option, but that would require both antennas to be installed on module at some point.
I think the best idea though is to have internal 2.4 antenna and port for external 900, and to switch between them by searching active receiver

pitts-mo, 2021-05-11:
@annoyinganime I was more concerned with quality of the antenna placements, allowable orientations, and their proximity to each other.

It does seem like making a radio with 2.4ghz ELRS as the internal module might be desirable or maybe see if a framework can support ERLS as a low latency sub module off of a DIY-Multiprotocol-TX-Module.

-p

brandonrc, 2021-12-12:
@JBKingdon Have you been working on this?

JBKingdon, 2021-12-12:
Kind of. I'm working towards a dual 915/2G4 link where both would be used at the same time, but it should be fairly simple to have that be able to operate in 'legacy' mode with either of the modems individually. Lots of caveats though, not least of which is I'm expecting it to be 2 to 3 years before it's ready. On the other hand, I think @cruwaller already has a working switchable 900/2G4 setup in his handset.