ExpressLRS/ExpressLRS #569: 900 MHz TX pcb

nightflasher, 2021-05-05:
As I've read in the fpv-racing-forum, someone is going to build a really small tx module to place it inside the jumper t-lite, using the stock external module socket.
But this module will be 2.4 GHz, but I could find any diy 900 MHz TX. Did I overread something?

CapnBry, 2021-05-05:
The issue system is not a Q&A Help Forum. Please ask questions on Discord or the Facebook Group:
 * [[image] Community Discord](https://discord.gg/dS6ReFY)
 * [[image] Facebook Group](https://www.facebook.com/groups/636441730280366)

I don't believe there is though.

closed 2021-05-05 by CapnBry

3518813067, 2021-05-11:
https://oshwhub.com/ELRS/915TX