ExpressLRS/ExpressLRS #574: HM EP1 RX Wifi update from 1.0.0-RC2 and RC3 not possible.

DeusAbsconditus, 2021-05-07:
Hello,

just tried to update my HappyModel EP1 RX's via Wifi to 1.0.0-RC3. Whatever I tried it wasn't possible. After 20 sec the LED started to blink very fast and the "ExpressLRS RX" network appeared. As soon as I tried to connect to the network the RX started to blink slowly again and the wifi network vanished. After some time the fast blinking started again and so did the network. 
Also an update via BF BetaflightPassthrough did not work. 
The only way to update was the FTDI method. After flashing 1.0.0-RC3 I tried to enter the Wifi and 10.0.0.1 page thing again to see if the problem also exists with RC3 and it does. 

However updating the 2.4 TX via Wifi to RC3 was not a problem.

closed 2021-05-07 by DeusAbsconditus

FireNWater, 2021-05-14:
Just had the same exact issue while updating to RC5.  Fast blink, WiFi shows up, then slow blink. . . back to fast. . .no web page shows up at 10.0.0.1 in browser.  Had to use BetaFlight passthru to update Nano Rx.

JupaCreations, 2021-05-15:
In the end I could update through FTDI UART from RC3 to RC5.
I updated the wiki for that...........
I have not tried to update RC5 through Wifi yet but see a new bug report about RC5 with same beviour as mentioned above.
Passtrough is ought to be be working in RC5 however which was not in RC3.
We will get there in the end with some patience.

Paco

dannyact, 2021-07-27:
Did the ""CLEAR PLATFORMIO DEPENDENCIES" method but unfortunately it still doesn't work.
Updating the TX wasn't a problem. But can't connect RX Wifi.
Internet keep saying  No connection "secured"
Regards,
Danny

AlessandroAU, 2021-07-27:
'No connection secured' doesn't mean it's not connected. It means the PC can't detect an internet connection, which is normal.

dannyact, 2021-07-27:
Allesandro, the receiver is connected, but as You say, won´t connect with the internet.