ExpressLRS/ExpressLRS #575: TTGO LoRa as a receiver? With PWM outputs?

Tygrys-1, 2021-05-08:
LoRa TTGO may work as a TX module, but this board can also be a receiver. As there are a lot of PWM enabled pins, it (maybe) might work as a simple receiver for flying wings and other planes. Size does not matter much there. 
Is it possible to make a TTGO LoRa RX target?

CapnBry, 2021-05-09:
I've changed this ticket to be strictly about TTGO LoRa as an RX target, and created a new ticket for PWM output support for any target #581.

Tygrys-1, 2021-05-09:
Will it be possible to use analog input of ESP as a battery sensor?

Tygrys-1, 2021-05-17:
Also #606

brandonrc, 2021-12-09:
@CapnBry is this something we still want to mess with? 

Based on https://github.com/ExpressLRS/ExpressLRS/issues/606 and the fact Matek has released those crsf-to-pwm boards.. I think we should close this issue and avoid RX's with ppm support directly for now.

What do you think? Open this back up if you think we should revisit it.

closed 2021-12-09 by brandonrc

Tygrys-1, 2021-12-13:
Not to reopen the issue, as the PWM has been done in a different way, but I still have unused TTGO on the shelf, so - how hard is to make it the RX target? As it is ESP and Lora, is it more to do then reasigning pins form EP1?

CapnBry, 2021-12-16:
We don't have any ESP32 RX targets currently, so that would be a whole new thing to deal with first. We can reopen this issue when we start having ESP32 RX at some point, but it isn't a priority now to get that done.

Tygrys-1, 2021-12-16:
Ah, have not noticed this detail ;-) I thought everyone is using ESP32 these days :)

Tygrys-1, 2022-07-02:
Quote from 3.0 Release: Unified ESP targets - Our internals are now set up to produce just one firmware image for all devices of the same type (all ESP32-based TXes, all ESP8285 RXes and now ESP32 based RXes)
Does that mean that TTGO as RX is possible? Does this require only copying the DIY 900 TTGO json to the RX folder, or more magic is required?

CapnBry, 2022-07-02:
I don't know if anyone tried it, so you're on your own, but if you flash a Unified ESP32 RX target you can then drop the json file into the webui. The json needs to be edited to accommodate the differences between any labels different for RX and TX though and those changes are (trails off)...

Tygrys-1, 2022-07-02:
It looks compatible. The only problem is that for the transmitter the serial RX and TX are the same pin, but for the receiver should be different I suppose. I do not know, if extra parameters as screen_rst and similar will do any harm to the configuration, or it simply does not matter?

Tygrys-1, 2022-07-02:
@CapnBry I do not see the "Unified ESP32 RX" Target in the configurator, 1.4.1 and there is no mention of Unified_ESP32_900_RX* in sources. Can you put more light on this please?

pkendall64, 2022-07-02:
Currently to use the unified targets directly you need to build the code from platofrmIO.

Tygrys-1, 2022-07-02:
Is there a 900 MHz ESP32 RX choice?

pkendall64, 2022-07-02:
Hmm, seems there isn't but it wouldn't be difficult to add one.

Tygrys-1, 2022-07-03:
Thanks, will give it a try.

Tygrys-1, 2022-07-03:
I am unable to build some targets, either under VS or ELRS configurator. The R9MM builds ok, ESP argets not. There must be something weird with my enviroment. Where to look for clues?
[Uploading ExpressLRSBuildLog_20220703113034559.txt…]()

pkendall64, 2022-07-03:
That looks like something stuffed up in the PIO install. It's not even getting to any of our build code there.
I have seen something like this before and just doing the build multiple times makes it work. It seems to be something wrong in the PIO code because it's mentioning `framework-stm32cube` during an espressif build?!

Tygrys-1, 2022-07-06:
@pkendall64 You were right. I deleted the pio directory and things sorted itself.