ExpressLRS/ExpressLRS #580: Happymodel ES24TX wrong esp-01f firmware, purpose..

gibranZawahra, 2021-05-09:
Hello everyone !
i'd like to thank you all for the hard work you've put in to get the project to where it is!

i just got my ES24TX and tried to update it over the wifi network that's created on startup by the esp-01f, problem is it has the r9m back pack firmware on it so that made it impossible, i'll be getting my transmitter soon so i'll try the wifi update method from the lua script, it should be working, if not i'll try flashing using the diy target instead.

so to sum it up is it normal that the esp-01f has the r9 backpack firmware on it ? and what is the use for the esp-01f if the esp32 is already capable of handling the update procedure ... ? thanks !

brandonrc, 2021-05-09:
@gibranZawahra that esp-01f is not doing anything right now. It is for later features. 

Maybe a backup incase we can't get wifi+elrs running at the same time.

gibranZawahra, 2021-05-09:
good to know thanks !
sooo are the only pins used on the esp-01f the rx tx and gpio02 for the led ? is there a schematic for this version of the board ? if so can i modify the pcb to add "accessories" like a neopixel or an i2c screen ? does the esp32 report any info to the esp-01f ? thanks !

CapnBry, 2021-05-13:
The ESP01 has no purpose at all currently, it is for future stuff like Brandon said. It shouldn't have even been flashed with anything from the factory because the wifi AP is always on. You can wipe it with a blink sketch or something to disable the wifi by setting the jumpers to USB-ESP01 positions (middle two I think?), holding down the GPIO0 button before plugging in, and then flashing whatever you want over USB.

None of its pins are brought out or connected to anything except serial and the LED.

gibranZawahra, 2021-05-13:
thank you very much for the info ! I totally forgot that the Wi-Fi will stay on i will definitely fix that ! do you think there's a way to send a serial message whenever the link is established or lost to the esp01 to maybe use the esp01 to control an RGB led and maybe even an i2c screen ? but only send a message in these cases :
- link lost/established
- bootup settings or settings changed
- other one time cases not thought of yet XD
This way i wouldn't add latency or more CPU load on the esp32 but would be a useful use for a currently useless esp01 ? thanks !

CapnBry, 2021-05-13:
Yeah the serial is connected to the ESP32 so you'll get all the logging messages that are sent. Isn't there a log section on the ESP01's website that shows the log? That's just a copy of what the TX's main chip is sending, so you could key off something there. This will be expanded to have some sort of protocol between the two chips at some point, to do telemetry or something more fun, but is is just there doing nothing for now so have at it!

I think you may have a hard time interfacing the ESP8266 with anything else though without the pins brought out to a header or pads or anything. There are TX modules with RGB leds, I think the Ghost TX and the NamimnoRC TX I think will too. Keep us in the loop though on Discord if you make something out of it. People will be excited to see what you make I'm sure, and maybe inspire some creativity in others!

gibranZawahra, 2021-05-13:
Thanks again ! you're on a roll today XD, you mean there are already debugging messages sent over uart ? and all i have to do is interpret them into something useful ? dude that's like a dream! yes now that you mention it even though it was supposed to be for an r9m backpack the log was still visible on the page ! i don't know how i forgot that !

i don't need pads or breakouts at least for the first time connecting everything and seeing if it'll work it'll be a prototype sort of endeavor, i can solder i2c to esp01f directly and I'll find power for the screen and led somewhere on the board !

I'll definitely share it on the discord if i manage to get it working ! thanks a lot !

gibranZawahra, 2021-05-14:
Hello again ! i have managed to solder an i2c SSD1306 OLED and a WS2812B led to the esp01 processor directly and managed to get it working so now i have only have to find a way to add strain relief for the tiny wires and write the program to display the values and change the led color accordingly !

brandonrc, 2021-12-12:
Closing issue.

closed 2021-12-12 by brandonrc