ExpressLRS/ExpressLRS #581: PWM Servo Output Support

CapnBry, 2021-05-09:
Instead of CRSF output, generate PWM signals to drive servos.

It would be great to have ExpressLRS drive servos directly for fixed wings that wouldn't have a flight controller. The developers have already been discussing this feature and agree it would be welcome as long as it didn't reduce the performance of the existing ELRS code. Jye has already tried a proof of concept and found there to be far too much jitter in the servo output using a single timer interrupt and software pin toggling (using default Servo library). Raphael already has expressed desire for this feature as well and seems to have some ideas about the implementation already.

Challenges
* 4 full resolution channel limit in ELRS 1.0. There is space in lower packet rates to increase the number of channels sent, but not to 16. We need to determine how many full-res channels we should support and what encoding methods would be used for the others. A possibly hybrid switches approach could be used for some channels.
* Lack of pins on hardware. ESP8266 targets don't have enough pins to drive many servos. Extending this with 1 or more shift registers could be a possibility.

This issue is for discussion if needed, please do not post "I want this" spam to this ticket.

Tygrys-1, 2021-05-09:
IMHO,
Full ranga channels: Not that much more. Typical plane needs 4 servos (2 Ail, Ele, Rud) and a throttle. Sometimes flaps. Then there is a pan-and-tilt. So 8 analog will be good. Gear, lights and bombs might use switch channels. Typical big radio as FrSky or Radiomaster have 8 analog controls (sticks, pots and sliders) and the rest are switches. So I vote for 8 Analog + 8 Digital, at the cost of 1/2 packet rate. BTW, 50 Hz for the plane should be enough, as most servos use that timing.

Thought about the lack of PWM outputs: Most receivers like R9 series, or Ghost can have some PWM outputs. I suggest maybe output some channels as PWM and sbus (or PPM) signal with all channels on one line. Then users may use SBus to PWM decoder to drive the additional servos. It also may help building the larger planes, where it will be convinient not to connect all servos directly to the receiver.

brandonrc, 2021-05-09:
> IMHO,
> Full ranga channels: Not that much more. Typical plane needs 4 servos (2 Ail, Ele, Rud) and a throttle. Sometimes flaps. Then there is a pan-and-tilt. So 8 analog will be good. Gear, lights [...]

For race wings we might only need two pwm output's. One for throttle and one for elevon's.  That way you can have a better packet rate. Not sure if you need 500hz for Racing FPV wings.. but might be better than 50hz?

Tygrys-1, 2021-05-09:
> For race wings we might only need two pwm output's. One for throttle and one for elevon's. That way you can have a better packet rate. Not sure if you need 500hz for Racing FPV wings.. but might be better than 50hz?

To be honest - you need 3, there should be the motor somewhere :-)
Typical analog servo refresh rate is 50 Hz, so there will be no benefit of faster RC rate. There are faster serwos, and some radios allow faster rates. The different story is the ESC, which could go up to 400 Hz. But on the plane it is not that important to have it sensitive, as the motor afects only the speed, not the stabilization nor attitude.

The main difference between the plane and a multirotor is tha they do not need the flight controller. FC benefits from the fater rate to feed PID controller and do computations, on the plane we work with phycical surfaces with big inertia of the system, and the rate is not that critical. Of course we can still use flight controler for the plane - and the only change this might require is at least extra 2 full range channels. This is because all the mixing will be done on the transmitter and the receiver should be able to drive all surfaces on the plane directly.

CapnBry, 2021-05-09:
I think Brandon might have been saying some hardware could work with just 2 channels using their existing pins, but that would only work for a 2ch glider (no motor like you say). I don't think anyone was suggesting we support **fewer** than our existing number of channels in the protocol though since our existing protocol already supports that at all packet rates. The challenge listed is for determining the extended number of channels we could support and at what rates.

Dropsleutel, 2021-05-09:
Maybe if multiple servo channels at some point are being considered (at costs of refresh rate),  maybe PCA9685 I2C / PWM chip  could be considered to be supported.  It can adress 16servos and boards cost nothing.

Tygrys-1, 2021-05-09:
I vote for 8 full channels total, where the 6 is absolute minimum. Thinking about 8 full + 8 switch we will support most controls on the OpenTx radio and the packet rate should not go less than 2 times slower. It will be extra 5 bytes more on the single packet.

Tygrys-1, 2021-05-09:
> Maybe if multiple servo channels at some point are being considered (at costs of refresh rate), maybe PCA9685 I2C / PWM chip could be considered to be supported. It can adress 16servos and boards cost nothing.

This was implemented in Inav some time ago. The problem I see is that most receivers does not expose I2C, so it will be hard to implement everywhere. The better solution is to output SBus, as there always is a serial port and inverter if needed. AFAIR the servo refresh rate was low on I2C solution.

CapnBry, 2021-05-09:
I also agree that 6 is the bare minimum but would prefer 8+8. I do not want to split the full resolution channels across multiple packets due to the possibility that only one of the two halves will arrive so we'd have to hold the first half and then throw it out if we don't get the second half, and that can get into a situation where we never update the servos again if we lose every other packet. I believe they all need to update together in case the user has an implied connection across the channels (e.g. CH3 high and CH6 low). It might be a limitation we'll have to accept though due to bandwidth, that CH1-4 are linked and CH5-8 are linked so arrange the channels so any interdependency you need is contained in one channel grouping.

brandonrc, 2021-05-10:
> 
> 
> > For race wings we might only need two pwm output's. One for throttle and one for elevon's. That way you can have a better packet rate. Not sure if you need 500hz for Racing FPV wings.. but might be better than 50hz?
> 
> To be honest - you need 3, there should be the motor somewhere :-)
> Typical analog servo refresh rate is 50 Hz, so there will be no benefit of faster RC rate. There are faster serwos, and some radios allow faster rates. The different story is the ESC, which could go up to 400 Hz. But on the plane it is not that important to have it sensitive, as the motor afects only the speed, not the stabilization nor attitude.
> 
> The main difference between the plane and a multirotor is that they do not need the flight controller. FC benefits from the faster rate to feed PID controller and do computations, on the plane we work with phycical surfaces with big inertia of the system, and the rate is not that critical. Of course we can still use flight controler for the plane - and the only change this might require is at least extra 2 full range channels. This is because all the mixing will be done on the transmitter (if no FC) and the receiver should be able to drive all surfaces on the plane directly. Or indirectly through the FC, but still with the full resolution.

You are right.. you would need at least 
Channel 1 = Left ELevon
CHannel 2 = Right Elevon
Channel 3 = Throttle

I was thinking you could do one channel for both elevons and have a y-split but that would not work because of pitch. up and down.  They need to be mixed to seperate channels.

P-I-Engineer, 2021-05-10:
Sbus? would allow use of sbus to pwm converters.

Tygrys-1, 2021-05-10:
> You are right.. you would need at least
> Channel 1 = Left ELevon
> CHannel 2 = Right Elevon
> Channel 3 = Throttle
> 
> I was thinking you could do one channel for both elevons and have a y-split but that would not work because of pitch. up and down. They need to be mixed to seperate channels.

Technicaly you may have y-spliter on ailrons and a separate elevator, but this will work for unpowered gliders only, as @CapnBry mentioned.

raphaelcoeffic, 2021-05-10:
> Sbus? would allow use of sbus to pwm converters.

SBUs support is trivial, I have it currently on top of my CRSF cleanup branch. I think this could be merged after the cleanup is done ;-)

raphaelcoeffic, 2021-05-10:
> Technicaly you may have y-spliter on ailrons and a separate elevator, but this will work for unpowered gliders only, as @CapnBry mentioned.

This is so 90s! It is a real pain as you need to setup the servos to be mechanically the same, rather than correcting linkage differences in the channel settings, as most pilots do nowadays.

that being said, “normal gliders”, even powered, need at most 8 channels. Jets, huge gliders (4-5m wingspan) use multiple servos for many things. These are always much easier to setup in the radio rather than mechanically.

For now, 8 full-res channels should be the first goal, IMHO.

raphaelcoeffic, 2021-05-10:
> Typical analog servo refresh rate is 50 Hz, so there will be no benefit of faster RC rate.

most digital servos support 100Hz. Beyond is pointless. Some analog servos do support 100Hz as well, but I’ve seen some funny behavior with some that look like they support it, but behave strangely at once.

Tygrys-1, 2021-05-10:
> SBUs support is trivial, I have it currently on top of my CRSF cleanup branch. I think this could be merged after the cleanup is done ;-)

And SBus will allow to directly use the SBus servos. :-)

raphaelcoeffic, 2021-05-10:
> > SBUs support is trivial, I have it currently on top of my CRSF cleanup branch. I think this could be merged after the cleanup is done ;-)
> 
> And SBus will allow to directly use the SBus servos. :-)

If you find some ;-) apart from Futaba, I could not find any.

Tygrys-1, 2021-05-10:
> > > SBUs support is trivial, I have it currently on top of my CRSF cleanup branch. I think this could be merged after the cleanup is done ;-)
> > 
> > 
> > And SBus will allow to directly use the SBus servos. :-)
> 
> If you find some ;-) apart from Futaba, I could not find any.

I think FrSky took the idea and has some Xact models. But they are FPort 2.0 compatible, not SBus.

pitts-mo, 2021-05-10:
sbus (inverted or not) seems to be a relatively easy option to make available for most ELRS RX. This could allow remote receiver/antenna placement with a centrally located pwm converter. A quick web search found several 4-10 channel sbus-pwm converters options for less than $15 (I do already have one).

Support for four servos like R9MM/Mini would be nice but it seems there are plenty of other improvements that are more widely desired / higher priority.

It also seems like a big mult-servo airplane would benefit from more fully proportional channels that are updated with more consistent frequency/priority.... So this makes me point back to an sbus/CRSF flight controller or a more traditional receiver with PWM outputs.

Just my $0.02
Thank you all,
-p

somewhat related thought:
Can RX be compiled without telemetry support other than what is required for binding? ... got to thinking about sbus input and making the receiver automatically pass another sbus signal out on fail-safe.

MarkusPfaff, 2021-05-11:
For the more traditional fixed wing airplanes 8 channels full would suffice. But of course there are planes that need more than 8 channels. Especially sail planes (which often have 6 servos in the wings) need more channels. Planes using multiple engines, e.g. 4 IC engines with throttle and additional glow or ignition control and flaps and gear and special functions need 16 channels easily.

Traditional setups don't make much use of switch channels by the way, because even on/off functionality like a tow hook needs servo end point settings which are normally set up on the transmitter. 

A good alternative to slowing down the complete protocol too much would be something like 8 full speed full channels and 8 low update rate full channels. Also combinations like 7+9 or 6+10 if that would fit bandwidth better would be viable options. Or may be 4 full rate, 4 half rate, 8 slow rate channels.

agetommy, 2021-05-15:
On a f3b or f3f modell 6 chanel in full res are enough.  8 chanel can be useful on a 3 flap wing. A sbus output was ok for me. Direct pwm out to direct connect servos isn't needed for all servo chanels.  Because I install a sbus decoder in most cases direct in the wing and the fuselage servos direct to the receiver pwm output.

raphaelcoeffic, 2021-05-15:
I think I’ll start to do it this way as well. Save quite some wire in these tight fuselages.

djoe78, 2021-05-15:
Hi, I use a r9mm on a quadcopter with expressLRS and an openTX radio with R9M 2019. I also have a boat in which a r9mm is mounted with still the original frsky firmware and I use 2 pwm channels to control it.  Now since the Tx R9M module has ExpressLRS how do I fetch two pwm outputs to control the boat?

Tygrys-1, 2021-05-17:
> Hi, I use a r9mm on a quadcopter with expressLRS and an openTX radio with R9M 2019. I also have a boat in which a r9mm is mounted with still the original frsky firmware and I use 2 pwm channels to [...]

Unless devs will do PWM outputs, you can buy any cheep flight controller and use it's PWM output. It is an overkill, but will work.
As a bonus you will be able to add GPS and make missions with iNav, or at least program RTH/failsafe :-) Can save you some swimming/rowing for a model :-)

raphaelcoeffic, 2021-05-17:
> Unless devs will do PWM outputs, you can buy any cheep flight controller and use it's PWM output. It is an overkill, but will work.

SBUS converter are quite cheap as well and do not require any config ;-) (less than 10$)

djoe78, 2021-05-17:
thanks for the answers, in the meantime I will do as you say hoping that the devs will give the possibility to choose to use the rx even only for pwm

MarkusPfaff, 2021-05-17:
SBUS output for the R9MM is currently not working, right? I tried it with 1.0.0RC2 and the baud rate looked like that of the CRSF protocol. 

Not only PWM and/or SBUS outputs would be interesting but of course telemetry protocol inputs like one for FrSky S.Port for example also. A wing oriented flight controller has some telemetry but not everything one would like to see. Take an ignition engine management unit like the GasSuite as an example.

Still the currently best overall solution for using PWM servos is usage of a flight controller.

Tygrys-1, 2021-05-17:
There is an open sensor project, maybe we can implement this into RX? It seems to be sport compatible.

MarkusPfaff, 2021-05-17:
The OpenXsensor project builds the sensor side of the S.Port. And it works quite well (I've several examples working here). 

S.Port is a UART based protocol using a single line for both data directions. Bus ownership is controlled in a protocol based manner.

Of course we would need to implement the side of the receiver which acts as the S.Port bus master and not the somewhat simpler sensor side. Some logic analyzer work will be needed to find out what the FrSky Rx are doing to interrogate the S.Port for sensors. The algorithm is adaptive for example in that when a sensor was found the S.Port bus master will ask this specific sensor more often from then on. Still not overly complicated and of course UART based.

Tygrys-1, 2021-05-18:
I think that logic analyzer might not be needed. :-) Here is the protocol specification: https://github.com/yaapu/FrskyTelemetryScript/wiki/FrSky-SPort-protocol-specs

raphaelcoeffic, 2021-05-18:
Don’t worry, S.PORT is very well know and dozens of implementations exist. The one from OpenXsensor is really not the best ;-)

MarkusPfaff, 2021-05-18:
True, the protocol is not too difficult even from the master side.
As for the polling algorithm I'd propose not to use the same as FrSky does, because it seems to me they do so only for a lack of RAM or to make programming easier.

My proposal for the physical IDs polling algorithm:

The S.Port master (e.g. a FrSky Rx) polls the physical IDs one after the other initially
to find which ones are present. If only one physical ID is found (i.e. is responded to by a slave, the
receiver will switch to an alternate sensor polling scheme (present
sensor, next ID to search, present sensor, next ID and so on). If
more sensors are found, the poll sequence returns almost to a
normal search pattern. This of course is not a good solution. It
looks like the FrSky receivers have not enougth RAM to keep all
active IDs. A better approach would be to poll all known active
physical IDs then poll one that ist known inactive and use this
pattern until all known inactive pyhsical IDs have been
polled. After that simply start at the first known inactive
physical ID again.

To further optimize the scheme the master would have to remember the number and update rate over all data IDs for 
every physical ID.

CapnBry, 2021-05-19:
I know you all are looking for more functionality and integrating all your sensors without using a flight controller, but I made quick CRSF to PWM converter that allows you to hook any receiver and get 8x PWM outputs with the ability to do channel mapping and failsafe settings (on the RX, not from opentx). Also includes VBAT telemetry.
https://github.com/CapnBry/CRServoF
[image]

JupaCreations, 2021-05-22:
How likely on a scale from 1 to 10 would it be possible to have a HM EP1 or EP2 output on TX and RX pad a PWM signal?
Yes 2 channels............so your asking youself is he joking?
No certainly not. Competition RC-cars use only two channels and guess what telemetry is not allowed.
So a no brainer when it comes to simplicity.
This would bring me a step closer to my project to modify a Jumper T-lite with re-angled sticks with built in "Jey" his 2.4 PCB using the onboard E28 antenna.
With 500Hz and minimal latency it can compete or outperfom high end brands like Sanwa and Futaba? (maybe :-) ).
Maximum range needed.60 meters as all outdoor tracks are not allowed to cross that optical LOS view limit.

Paco

CapnBry, 2021-05-22:
Possible. We just discussed this yesterday for doing a DLG which only has 2x servos. It is possible that anything with pins will be able to use at least some of them for PWM output, but it is too early to say for sure given how many different RX there are and that not all this code has been written yet.

JupaCreations, 2021-05-23:
Thanks  for the answer Captain. We as always wait with patience.

Paco

HVisMyLife, 2021-06-21:
So 8 full resolution channels (1/4 might be good too) are coming in a few weeks? Or you have to have more time? I'm asking, because I want to buy rx and tx from happymodel, but, apart from quads, I'm flying fixed wings too, and I need at least 6 non-aux channels.
Robert

raphaelcoeffic, 2021-06-21:
> So 8 full resolution channels (1/4 might be good too) are coming in a few weeks? Or you have to have more time? I'm asking, because I want to buy rx and tx from happymodel, but, apart from quads, I'm flying fixed wings too, and I need at least 6 non-aux channels.
> Robert

SBUS is in the starting blocks, so get yourself one of these 4 channels SBUS / PPM converter ;-)

CapnBry, 2021-06-21:
> So 8 full resolution channels (1/4 might be good too) are coming in a few weeks?
I'm not sure where you got that impression, Robert, but PWM output is likely coming in a few **months**, and more than 4 full-resolution channels is under consideration but there are no plans at this time for any actual implementation. We're simply not able to fit any more data into our packets / air time.

raphaelcoeffic, 2021-06-21:
> I'm not sure where you got that impression, Robert, but PWM output is likely coming in a few **months**, and more than 4 full-resolution channels is under consideration but there are no plans at [...]

I discussed some other air modes with Sandro, I think this is very well possible. Of course, there will be some hits to either range and/or latency, but well, physics cannot be changed, right?

HVisMyLife, 2021-06-21:
At first thanks for quick answer, but I know how to convert crossfire protocol to pwm, I'm only asking for full resolution channels.

raphaelcoeffic, 2021-06-21:
> At first thanks for quick answer, but I know how to convert crossfire protocol to pwm, I'm only asking for full resolution channels.

This is coming, I need it badly as well.

CapnBry, 2021-06-21:
> I discussed some other air modes with Sandro, I think this is very well possible. Of course, there will be some hits to either range and/or latency, but well, physics cannot be changed, right?

Of course it is possible, Crossfire does 12ch full resolution. Ghost has more full-resolution channels than ELRS as well. We can do that too because it is the same hardware by reducing packet rates / range / sensitivity / etc. That's somewhat outside the scope of this issue though, which is strictly for PWM output directly from the RX.

HVisMyLife, 2021-06-21:
Packet rate is not as important when it comes to fixed wings, channels resolution could be halfed too, to reduce amount of data per second.
And how about something like multiplexing that four full res channels? Is it even possible?

CapnBry, 2021-06-21:
There are a tremendous number of options available to implement different ways to send the channel information which will be explored as we determine what are the best compromises to accept to support more full res channels.

Tdogb, 2021-07-05:
The R9 Slim+ has the capability for pwm outputs, so could pwm just be added for these targets. Would be great not to have to by a different radio system just for my wings

Fpv191, 2021-07-18:
Hi guys, 
Congrats on your 1.0.0 Stable Release !

Just an idea to help with PWM Support.
Is it possible to just reassign the pinouts originally intended on the r9mm ?
What I mean by this,  is to free up and to have the use of the 4ch PWM on the side pins (that you are currently using 2 of 4 for Rx and Tx.)  This would be enough for a wing with no flight controller.

Could you then instead remap the pinout to have ELRS use CRSF protocol over r9mm SBUS pin to be used for TX , And have uninverted Smartport / Or Inv. S.port pin to be used for RX (and telemetry) to connect to a flight controller?  (As originally FrSky r9mm was wired) 

This would also help those to transition to ELRS without rewiring their SBUS and Smartport and for those using F4 FCs to also have telemetry. 

So if you could reassign ELRS CRSF to use SBUS pin as TX , and use Uninverted Smartport / Inv S.Port as RX and for possible passthrough flashing?
 This would leave the 4 PWM pins free for servos for wings with no flight controller. 

Also flashing using the 3 pins for both FrSky bootloader .frk And then the main ELRS firmware using just a servo wire to the transmitter for those wing pilots who don't use or have FCs to complete the second part might be preferred if this is possible. (Just like updating FrSky r9mm originally with .frk from module bay pins)

This servo PWM option could be enabled when SBUS pin for ELRS signal output option is selected in the Configurator GUI so anyone else using the current side pins on r9mm for Rx Tx won't need rewiring.

I hope you follow and understand my explanation.

Thanks for your great work. Cheers.

Tygrys-1, 2021-07-18:
SBUS output pin is hardware inverted, so there might be a problem to output CSRF on it.

Fpv191, 2021-07-18:
Thanks for the reply. I was just throwing around some ideas to see if it was feasible.

My understanding of how it works currently, is when you select SBUS in ELRS Configurator it outputs Epresslrs / CRSF protocol on SBUS pin of r9mm, but I'm also aware it is not sending out SBUS protocol. 

The CRSF signal would be hardware inverted on SBUS as you said, so on a r9mm yes, and for F4s using the SBUS pad on, the FC would uninvert this CRSF signal.
  Most other F3 F7 FCs could invert this signal to work on a normal UART I would think?  
This was just an idea if could reassign the pads to free up the 4 PWM pads to enable servos and with the same Configurator setting to use the SBUS ( for sending ELRS /  CRSF)  and S.Port / uninverted S.Port for telemetry , This was how FrSky was originally wired.

I thought it could be just a remapping of pin assigned in the code for SBUS pin as TX output option and S.Port pin for RX (and telemetry). And for flashing either passthrough or to use Transmitter only for final ELRS firmware file flashing without a FC.

Tygrys-1, 2021-07-18:
You are right, but sending out inverted CRSF is not a standard way of doing things right. AFAIK CRSF requires RX/TX connection, and not on all F4 Controlers the UART dedicated for SBUS exposes it's TX pin. This will make things harder to understand and configure for users and will require extra settings in FC's software (on F7).

Maybe the SBUS pin might be used for PWM?

Fpv191, 2021-07-18:
While I agree you certainly have some good points,

Just thinking with regards to CRSF protocol,  it must work just needing a single wire TX for control, because as it stands now, when you select SBUS option in ELRS Configurator, it will send CRSF / ELRS protocol over the single inverted SBUS pad on a r9mm. 

The Configurator states it has no telemetry when SBUS output mode is selected,
 ( assuming most F4 having TX absent on the dedicated SBUS UART.)  

How possible would it be to implement a smartport or inverted S.Port for the telemetry side of CRSF ? Just as FrSky has always done?  Maybe on a separate UART if it worked?

These options to run like this would only be enabled when the SBUS output mode option is selected in ELRS Config. and would be aimed at F4 users to allow them to use SBUS pads (like it is already able to at the moment ) but to also add the ability to have functional telemetry /  and this frees up 4x PWM side pads for servo outputs for wing guys to also be enabled while in this SBUS mode to cover both these scenarios.

When the SBUS option is unchecked , this reverts to the existing normal mode for setup and wiring using side TX RX pads that would be used for F7 or other normal users. 

Just an idea.

brandonrc, 2021-12-12:
@CapnBry is this something we still want to do? I know you worked on some code that got a few PWM servo outs from RX's but with the matek breakout board what do you think?

FOG-Yamato, 2022-01-09:
Was already added as part of V2 #998

closed 2022-01-09 by FOG-Yamato