ExpressLRS/ExpressLRS #583: Support via_BetaflightPassthrough with Ardupilot

CapnBry, 2021-05-09:
Ardupilot can do passthrough serial as well and if configured manually (holding the boot button and setting the passthrough manually) has worked for a user so 420000 baud passthrough should work.

Needed
* Connect to Ardupilot (baud rate might be anything from 1200 to 1500000) and query `SERIALx_PROTOCOL` (x = 1-8) and look for which has a value indicating it is CRSF (29)
* Set `SERIAL_PASS2` to the `x` discovered to be CRSF
* May need to set `SERIAL_PASSTIMO` to set the timeout for the port to revert back to Ardupilot. 0 is forever, value is in seconds.
* Upload away!

brandonrc, 2021-05-10:
@CapnBry are you saying someone has actually flashed ELRS with Ardu? I attempted this but could not get it to work. Got some weird Errors. I Will try it again.

CapnBry, 2021-05-10:
Yes but they did it all manually by holding down the boot button on the RX, powering, then setting SERIAL_PASSTIMO and SERIAL_PASS2 and using the UARTUpload.py script.

aLDime, 2022-01-18:
> > * SERIALx_PROTOCOL (x = 1-8) and look for which has a value indicating it is CRSF (29)>
`Processing DIY_900_RX_ESP8285_SX127x_via_BetaflightPassthrough (platform: espressif8266@3.2.0; board: esp8285-8285; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 252,49,226,149,186,203
***Building for ESP8266/ESP8285 Platform
AUTO_WIFI_ON_INTERVAL = OFF
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285-8285.html
PLATFORM: Espressif 8266 (3.2.0) > Generic ESP8285 Module
HARDWARE: ESP8266 160MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <SX127xDriver>
|   |-- <logging>
|   |-- <SPI> 1.0
|-- <CONFIG>
|   |-- <EEPROM>
|   |   |-- <logging>
|   |   |-- <EEPROM> 1.0
|   |   |-- <Wire> 1.0
|   |-- <logging>
|   |-- <OTA>
|   |   |-- <CRSF>
|   |   |   |-- <CRC>
|   |   |   |-- <CrsfProtocol>
|   |   |   |   |-- <CRC>
|   |   |   |-- <MSP>
|   |   |   |   |-- <logging>
|   |   |   |-- <TelemetryProtocol>
|   |   |   |-- <FIFO>
|   |   |   |   |-- <logging>
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |-- <POWERMGNT>
|   |   |-- <DAC>
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |   |-- <SX127xDriver>
|   |   |   |-- <logging>
|   |   |   |-- <SPI> 1.0
|-- <CRC>
|-- <CRSF>
|   |-- <CRC>
|   |-- <CrsfProtocol>
|   |   |-- <CRC>
|   |-- <MSP>
|   |   |-- <logging>
|   |-- <TelemetryProtocol>
|   |-- <FIFO>
|   |   |-- <logging>
|   |-- <helpers>
|   |-- <logging>
|-- <BUTTON>
|   |-- <DEVICE>
|   |   |-- <helpers>
[... 172 lines cut ...]
Flash: [====      ]  37.0% (used 378736 bytes from 1023984 bytes)
Configuring upload protocol...
AVAILABLE: custom, espota, esptool
CURRENT: upload_protocol = custom
Uploading .pio\build\DIY_900_RX_ESP8285_SX127x_via_BetaflightPassthrough\firmware.bin
Uploading .pio\build\DIY_900_RX_ESP8285_SX127x_via_BetaflightPassthrough\firmware.bin
======== PASSTHROUGH INIT ========
  Trying to initialize COM14 @ 420000
No CLI available. Already in passthrough mode?, If this fails reboot FC and try again!
======== RESET TO BOOTLOADER ========
  * Using full duplex (CRSF)
Cannot detect RX target, blindly flashing!
Uploading .pio\build\DIY_900_RX_ESP8285_SX127x_via_BetaflightPassthrough\firmware.bin
esptool.py v3.0
Serial port COM14
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Invalid head of packet (0xFD)
*** [upload] Error 2
========================= [FAILED] Took 20.00 seconds =========================

Environment                                          Status    Duration
---------------------------------------------------  --------  ------------
DIY_900_RX_ESP8285_SX127x_via_BetaflightPassthrough  FAILED    00:00:19.998
==================== 1 failed, 0 succeeded in 00:00:19.998 ====================`