ExpressLRS/ExpressLRS #584: Add firmware version on update web server.

brandonrc, 2021-05-10:
Someone on facebook asked how could they verify a flash was succesful with the wifi updater. They are using a radio that does not have a lua script so no way to check hash on that. 

Can we display the hash on the web server so that will allow people to check the flashed hash/version?

kubrick89, 2021-05-12:
Why not just write the version (not the hash) in the static web page on each new release?

greenbigfrog, 2021-05-12:
This could be added nicely as a step to `src/html/compile.sh` from #527 by @pkendall64 
Just add a sed or the like to patch in `git rev-parse --short HEAD`.

pkendall64, 2021-05-12:
Yes, I've got some code that I'm working on for this, but am a bit busy with other things. Once we get to merging features for 1.1 then I'll get that change in there.

pkendall64, 2021-05-12:
Changes have been made to PR#527 which will provide the tag or branch followed by the commit hash as the version.

CapnBry, 2021-07-31:
Resolved for 1.1 thanks to pkendall64 #527

closed 2021-07-31 by CapnBry