ExpressLRS/ExpressLRS #586: Support for Radio master TX16 Multiprotocol Module possible?

evildronepilot, 2021-05-10:
Is it possible to get support for the internal multi-protocol module build into TX16.. radio?
Having the option to select the 2.4 ELRS would be a mind-blowing feature.

evildronepilot, 2021-05-10:
Openned with OpenTX 

https://github.com/opentx/opentx/issues/8495

raphaelcoeffic, 2021-05-10:
@evildronepilot do you mean having ELRS as an internal module? Or do you mean integrating ELRS into MPM?

sweebee, 2021-05-10:
No not possible, the multi-protocol module doesn't have the lora chipset.

evildronepilot, 2021-05-12:
> No not possible, the multi-protocol module doesn't have the lora chipset.

Thanks for looking into this. All thumbs up!

closed 2021-05-12 by evildronepilot

tipouic, 2021-05-14:
> 
> 
> No not possible, the multi-protocol module doesn't have the lora chipset.

Hi the multi-protocol have a free SPI port on the PCB to add another chipset.
Alessandro had started to port his code on it for 868-916Mhz but he stopped because the code changed too much. He had put an RFM95 from memory but now the Jumper T18 radio comes out with a 5in1 pcb (sx1276). 
[https://github.com/AlessandroAU/DIY-Multiprotocol-TX-Module/tree/adds-ExpressLRS-support](https://github.com/AlessandroAU/DIY-Multiprotocol-TX-Module/tree/adds-ExpressLRS-support)