ExpressLRS/ExpressLRS #588: Idea: Support  SmartAudio / IRC Tramp to control a VTX

Pelado-Mat, 2021-05-11:
It would be great to be able to manage the VTX as TBS/Ghost does directly from the rx module and the LUA script. As ExpressLRS is open source, could implement both protocols giving us, the users, more power. I feel that's is way better to manage the VTX from the radio bypasing the FC.

SmartAudio is an open standard (not sure TBS would be very happy about this) and documented here:
https://www.team-blacksheep.com/tbs_smartaudio_rev09.pdf
And implemented in BF
https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c

IRC. I could'nt find any publlic information, but It's all implemented in Betaflight
https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_control.c

Pelado-Mat, 2021-05-12:
Use cases:
- When do you not have a FC
  I have a homebuilt FPV tank without FC
- My original use case:
  I needed to free a UART in a small cheap F4 FC
- My discovered use case:
  While flying with friend and needed to change the video band, It's way easier to change it once on the radio and let the crossfire change the vtx band on all the drones at startup. See "My Vtx" feature in TBS Tracer/Crossfire
- No more VTX tables!
- I can imagine that it would be useful in a race when a band is assigned to you, but I'm no racer.

If you like/need this feature, now you have two incompatible ecosystems, TBS Crossfire/Tracer with smartaudio or IRC Ghost with Tramp.

greenbigfrog, 2021-05-12:
> When do you not have a FC
> I have a homebuilt FPV tank without FC
Well, right now you need some way of getting the PWM or the like signal anyway, so adding a FC is a good way to go here anyway (SBUS output etc is being worked on/being tracked in a other issue)

>  I needed to free a UART in a small cheap F4 FC
Well, this would require adding another pin to HW. I guess maybe for r9*, but another pin usually requires more space, and the smaller the better :wink: 

> "my vtx" feature
this sounds interesting. actually does sound smart/cool. although I wonder if this couldn't be implemented as is somehow via a addition to BF

> no more vtx tables
Huh? Isn't the idea with vtx tables to know which channels are available, what power, etc. How is this not required anymore after?

Overall sounds like at least some people would get a good use out of it.

MSmeets94, 2021-05-12:
> 
> 
> Currently we dont have spare pins to control a VTx directly.
> 
> What advantage is there in controlling a VTx directly with the Rx and not through BF?

I'm leaving a reply to let you guys know that there could be more interest in this. Especially the "my vtx" feature from both immersionRC and TBS sounds great, but because there is no dedicated pin for SA  you would have to do something via BF i guess. 

The advantage is that you can sync all your VTx's using your transmitter. e.g. you select R4 25mW as your channel / power on your tx and as soon as your quad powers up the VTx is changed to R4, 25mW.

endrsmar, 2021-07-19:
Also voicing an interest in such feature. My usecase is that I'm running no-telemetry because of 500Hz mode. I'd like to be able to set up my VTX channel via TX without the need for full MSP. In a race I typically have 3+ quads and in elimination rounds your channel can change quite often. I also feel that VTX tables are somewhat unnecessary complication, the channel frequencies are defined and you as a user should know whether the channel you are trying to set is supported by your VTX. Power levels can be 1, 2, 3, 4, 5 and again leave the responsibility of knowing what power levels correspond to the numbers for their given VTX.

henkwiedig, 2021-07-30:
> Currently we dont have spare pins to control a VTx directly.

This is true for ESP-01F based RX but if i'm not mistaken there are free pins on the STM32 based ones. For the RX_PP Pin8/PA2 could be used for USART2_TX and is currently not connected.

brandonrc, 2021-12-20:
Closing this because we have VTX admin now. 

https://www.expresslrs.org/2.0/quick-start/lua-howto/#vtx-administrator

closed 2021-12-20 by brandonrc