ExpressLRS/ExpressLRS #597: WiFi update on RC5 is broken

FireNWater, 2021-05-14:
Tried to update a DIY 2.4GHz Nano via wifi.  Was able to update to RC5 using wifi, but once RC5 was installed, the wifi feature didn't work.  After 20 seconds of power on, the LED went into fast blink and the Rx showed up on my PC.  While attempting to connect to the Rx, it went back into slow blink and the connection failed.  Nothing ever showed up on the browser at 10.0.0.1.  I was able to re-flash RC5 onto the Rx using BetaFlight passthru.

CapnBry, 2021-05-14:
Working for me with EP2 over wifi on RC5 which I believe is the same hardware? I've flashed 4x in a row via_Wifi, and it is kinda slow but it does work and no reboots like you're seeing.

d3c0deFPV, 2021-05-15:
I've experienced this symptom on a Happymodel EP2 and EP1. Fast/strobe blink after 20 seconds, but upon connecting/authenticating, slow blink and SSID becomes unavailable as if it restarted. My first flash of 1.0.0 RC3 after receiving them caused this. I cleared platformio dependencies, EP2 was reflashed via BF passthrough, EP1 through FTDI after jumping boot.

Upgrading RC3 to RC5 I experienced this again on both the EP1 and EP2. This time, no clearing of platformIO dependencies, just changed build target to BF Passthrough/UART and hit build. I'm now running RC5 on both with wifi that connects. I have not tried reflashing RC5 via wi-fi, but can try this tomorrow.

AlessandroAU, 2021-05-15:
Just tested on a DIY 20x20 board and it worked for me

schugabe, 2021-05-15:
It's also working on my EP2 on RC5.

JupaCreations, 2021-05-15:
I just tried it on a Happy Model EP1 and EP2 which yesterday were updated from RC3 to RC5 with UART succesfully.
With RC5 and Wi-Fi I get the same behaviour as before on RC3.
While attempting to connect to the Rx, it went back into slow blink and the connection failed. Nothing ever showed up on the browser at 10.0.0.1. Also password is never asked as it never gets really connected.
What is different to those who are able to update by Wi-Fi?

Paco

d3c0deFPV, 2021-05-15:
I've flashed the same EP1 about 20 to 30 times this morning trying to reliably reproduce this, but I cannot. In the majority of attempts, wifi updating was broken after update, but I was still able to connect to the TX in all cases.

I also was able to reproduce this several times after flashing over FTDI/UART, so it doesn't seem exclusive to updating over wifi as I previously suspected. Even clearing platformio dependencies each time, I flashed via UART several times and wifi remained broken. I tried from two different Windows systems, too.

In each case I used either HappyModel_EP_2400_RX_via_WIFI or HappyModel_EP_2400_RX_via_UART, respectively. 

I'll continue to experiment, controlling variables as much as possible. Wifi updates are not very essential to me, but I'm happy to help troubleshoot. If there are any devs within the United States that would like to test with my unit specifically, I'm happy to mail it over if you think it may be specific to this one.

yds, 2021-05-15:
been having this same issue with two different EP1 RXen -- pattern is: if I manage to flash via WiFi, then the very next attempt to connect to WiFi causes the EP1 to reboot and go into slow flash then after 20 seconds fast flash WiFi mode and as soon as I attempt to connect to that WiFi the EP1 reboots, rinse repeat.. however if I flash via serial passthrough, then I can flash via WiFi exactly one time which will trigger the reboots every time I try to connect via WiFi.. this is now with two EP1s and on two different computers that this happened.. this was with RC5.. not a complete showstopper cuz I can get back to expected behavior by flashing via passthrough, however it definitely smalls like a bug somewhere there..

the `user_defines.txt` with my random binding phrase, in case that helps to reproduce this.
```make
-DRegulatory_Domain_ISM_2400
-DMY_BINDING_PHRASE="D0wGwDEh6IYm7f2o"
-DHYBRID_SWITCHES_8
-DENABLE_TELEMETRY
-DLOCK_ON_FIRST_CONNECTION
-DUSE_500HZ
-DAUTO_WIFI_ON_INTERVAL="20"
-DFAST_SYNC
```

JupaCreations, 2021-05-15:
For what it is worth.
I have an EP1 and EP2 updated to RC5.
But the Äuto Wifi on interval"is set to 20 seconds.
For the EP2 I could with passtrough (betaflight FC) change the 20 seconds to 60 seconds purely for troubleshooting.
After this I could connect multiple times with the cellphone and the laptop and could succeed update RC5 again with 60 seconds setting..

Then I switched over to the EP1. Wifi update with 20 seconds not possible.
It looks like the 20 seconds is not enough for connecting to the webpage 10.0.0.1
Then I connected for passthorugh (Inav FC) but could not succeed so I could not change the 20 seconds.

But maybe for others to troubleshoot above info to confirm my findings.
I will try some more tests.

Paco

CapnBry, 2021-05-15:
This is the exception when it reboots as I try to connect
```
Begin Webupdater
Stopping Radio
HTTPUpdateServer ready! Open http://elrs_rx in your browser

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x402142e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0c2c0000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc90 end: 3fffffc0 offset: 0190
3ffffe20:  4024ed8c 00000004 3ffffe60 4020bc58
3ffffe30:  3ffeee34 0c2c0000 3ffffe60 4020be17
3ffffe40:  3ffeee34 3ffeee54 3ffeee10 4020be49
3ffffe50:  3ffeee34 3ffeee54 3ffeee10 40203817
3ffffe60:  00000000 2c9f0300 00000000 3fffc278
3ffffe70:  40104758 3fffc200 00000022 3ffeee10
3ffffe80:  402056d0 00000030 00000010 ffffffff
3ffffe90:  40205890 00000001 3ff00014 00000001
3ffffea0:  00000000 00000000 00000000 fffffffe
3ffffeb0:  ffffffff 3fffc6fc 00000000 3ffef05c
3ffffec0:  00000000 00000031 3ffef9d0 01000000
3ffffed0:  3fff0b14 0010001f 80000000 00544547
3ffffee0:  00000000 03000000 3fff0aec 001d001f
3ffffef0:  8a00656e 055cc9a2 4bc6a700 00000000
3fffff00:  00000000 00000000 00000001 00000000
3fffff10:  00000004 00000000 3ffeee10 3ffef9d0
3fffff20:  00000000 4bc6a7f0 56c8b439 3ffef9d0
3fffff30:  3ffeef60 3ffeee34 3ffeee10 40204e93
3fffff40:  3ffeef60 3ffeef0c 401018a9 00000000
3fffff50:  4020dd28 3ffeef93 3ffeef94 4020cdd6
3fffff60:  3ffeef60 3ffeef93 3ffeef94 40204f74
3fffff70:  00000000 3ffeef93 3ffef05c 40205144
3fffff80:  3fffdad0 3ffeef93 3ffeef94 402058f1
3fffff90:  3fffdad0 3ffef838 3ffef9bc 40101655
3fffffa0:  3fffdad0 3ffef838 3ffef9bc 4020cec4
3fffffb0:  feefeffe feefeffe 3ffe8740 40102681
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
I got to this state after flashing a brand new EP2 over wifi five times, then switched to via_BetaflightPassthrough, and now I get this as soon as I try to connect to the wifi. I've gotten it across three power cycles and the dump starts the same every time so it seems pretty reliably broken after working reliably on the exact same hardware just a moment ago.

yds, 2021-05-15:
BTW, in case it helps to narrow this down, I was seeing this happen with my first EP1 with RC2 and RC3 and now RC5.

JupaCreations, 2021-05-15:
Got the EP1 by UART updated with Auto wifi on interval on 60 seconds.
Power cycle the RX and watch the yellow led.
After 60 seconds of start up 1 Hz flashing it goes over into quick flashing (web connection mode).
However the time it is in the quick flashing mode is sometimes 2 minutes when I abort it and power cycle again.
Sometimes it is 30 seconds, sometime 1, sometimes 5, and sometimes 10 seconds.
Irregular intervals where you expect it to be 30 seconds fixed if you only power on the receiver.

So by UART I chnaged the auto wifi on interval to 30 seconds.
Same behaviour as with 60 seconds.

Now there are 3 ways to update
1] UART
2] Passthrough
3] Wi-Fi

As Capnbry wrote what is blocking or interfering on the other settings.
Which mix of usage can cause this?
Now currently I only flashed RC5 with UART to this EP1.

Paco
.

CapnBry, 2021-05-15:
I'm not sure I'm clear on if you're expecting the right behavior, based on your description. When the RX goes into webupdate mode (after the defined 20 second timeout or whatever) it will stay there forever and never go back to slow flashing LED. If you're seeing it go back to slow flashing, it is likely something connected to it and caused it to reboot (maybe you've saved the wifi network so something is autoconnecting). I could be in webupdate mode for as long as I wanted but as soon as I tried to connect to the wifi (and whatever other extra garbage windows does when "Checking network requirements") it would reboot and go back to slow flashing.

I've not been able to reproduce it and I've flashed via_wifi to via_bf, via_bf to via_wifi, via_bf via_bf via_bf to via_wifi, via_wifi via_wifi to via_bf. I did not need to try to flash it via_wifi to get it to crash and reboot. I stayed on via_bf and just clicked the wifi network and it would die.

yds, 2021-05-15:
for me the pattern seemed to be as described earlier.. flashing via WiFi would trigger the power cycle bug whenever connecting to WiFi.. flashing via serial, BF_ passthrough or FTDI would make WiFi work again long enough for me to flash via WiFi which would trigger the bug again.. I don't think I've been able to trigger this bug after flashing via serial.. maybe I didn't give it a good enough chance..

as for power, I've had this bug show up when powered via USB to FTDI on a laptop and a desktop.. I've had it show when powered via FC which is powered via USB (the RX does get power on this FC).. and I've had this bug show when the FC is powered by a fresh 6S LiPo -- when WiFi update works it works with all three power sources, and when it doesn't work, same thing, it's broken regardless of power source..

CapnBry, 2021-05-15:
Agreed. I don't think it is power. Something else is wrong because it works 10x in a row on a power source, then it gets fudged up and it won't work at all on the same power source until you change [something], then it works perfectly again indefinitely.

Benaresq, 2021-05-16:
> I'm not sure I'm clear on if you're expecting the right behavior, based on your description. When the RX goes into webupdate mode (after the defined 20 second timeout or whatever) it will stay there [...]

I'm getting the same behaviour on my EP2 receivers running on RC2. They stay in wifi mode until you try to connect, then the wifi drops out.

The were flashed to RC2 via wifi from new.

Xedos9er, 2021-05-21:
I noticed that my happymodel EX24TX does not show "success" when updating via Wifi after I flashed RC5. Tried to flash RC7 today, it seems to upload the firmware and power cycles right after, but no "success" message in 10.0.0.1 adress like before. 
How can I check if RC7 has been flashed successfully? I managed to flash RC7 via USB using the lower 2 jumpers and ELRS configurator.

timbuktu031, 2021-05-22:
Same for me. I am using an ep1 receiver. I tried wifi connection to upgrade RC7 firmware from RC5 firmware, but failed.
So I upgraded the firmware by UART method. But this also doesn't work. When the battery is plugged in, the LED stays on immediately and does not bind to the remote controller.

yds, 2021-06-03:
Flashed 1.0.0 RC8 with `AUTO_WIFI_ON_INTERVAL=30` via WiFi -- flashed it again via WiFi another 3 or 4 times.. worked every time! for me at least this bug is no longer reproducible..

Benaresq, 2021-06-06:
> Flashed 1.0.0 RC8 with `AUTO_WIFI_ON_INTERVAL=30` via WiFi -- flashed it again via WiFi another 3 or 4 times.. worked every time! for me at least, this bug is no longer reproducible..

Just tried it myself, flashed RC8 using a FTDI adapter with a wifi interval of 20 and had the same wifi problem as before. I changed the interval to 30 and it seems to work perfectly now.

CapnBry, 2021-07-31:
I don't think this is an issue any more, I am going to flag this for closure if no new reports come in.