ExpressLRS/ExpressLRS #598: Telemetry Lost on R9M + R9 Mini 1.0+

AlfredBroda, 2021-05-14:
On 1.0.0-RC5 (and all previous 1.0 versions) I can't get a stable link - after some time (1-5 minutes) the radio announces "telemetry lost" - only a reboot on the radio side brings back the link. Also, RxBat and other sensors are missing (the ones following number 10). We made a video clearly showing what exactly is going on here: https://www.youtube.com/watch?v=9v6rcC5r_hg

Hardware is Taranis X9D Plus (OpenTX 2.3.10 ELRS) + R9M 2018 (with 420 Ohm resistor mod) + R9 Mini.
This setup works perfectly fine on Express LRS 0.1.3-RC2 - the RxBat and remaining sensors visible in the video were detected on that firmware. Settings used were: Packet Rate 100Hz, Telemetry Rate: 1:16, Power 25mW, Band 868MHz (EU).

Fails on Git hash: e25064
Works on Git hash: 2942a6

CapnBry, 2021-05-14:
Nice video showing the issue. I'm pretty sure this is different than the issue we were working in for #553 which was exclusively on the RX, and your TX is locking up. I don't have any answer for that though because it isn't something we've encountered.

The lack of battery telemetry looks like you don't have ENABLE_TELEMETRY on both the RX and TX builds. In 0.13RC3 I think the battery telemetry was squeezed in using another means but now you have to have the enable telemetry defined on both to get anything apart from link statistics in opentx. You might also need to rediscover sensors but I don't think so.

joneseym, 2021-05-14:
I am seeing similar issues on my X9D and RC5, but I'm using 2.4GHz (DIY TX and HM EP1 RX). Yesterday I had no problems with the X9D with RC3 on a long flight but today on RC5 it won't hold a stable telemetry connection even at short range.

An identical TX module is working fine on a Jumper T-Lite (other difference is that the T-Lite is using the stock OpenTX build vs the custom ELRS one on the X9D)

CapnBry, 2021-05-14:
I'm going to change the title on this because opentx saying "Telemetry Lost" doesn't mean a thing about the connection between the TX and RX and it adds to confusion. Yes, opentx does say "Telemetry Lost" when the connection goes down, but it also says "Telemetry Lost" when the connection is up and working just fine. We need to differentiate the two because it sounds like @joneseym is only complaining about unstable telemetry information to opentx and @AlfredBroda's issue is that his link drops entirely and the TX is locked up.

AlfredBroda, 2021-05-17:
I tested both the `9xd-fix` branch and `RC5`; it seems I can get some more telemetry items if I change *TLM Ratio* to 1:8. Just had an uninterrupted link with telemetry on `RC5` for over an hour. The update frequency on the Taranis' screen however is very slow: a stars appears next to a sensor from the extended group in roughly 7 seconds. New sensors have appeared, including GPS. 😁

AlfredBroda, 2021-05-18:
I just got a Jumper T18 and tested on the same settings with the same R9M + R9 Mini - everything works perfectly. Telemetry refresh for the extended items is constant (cycling in sets of 4). I believe this confirms it as a X9D+ issue.

kimman1, 2021-05-18:
But a lot of people are using x9d+ so i think we need to find out why elrs telemetry is not stable in x9d+

CapnBry, 2021-05-19:
Hello folks, we were able to find a developer with an X9D+ and get some test opentx builds out into testers hands. Thanks to bernabap for being the first to report back in!

If you're losing connection while standing in the same room as your ELRS RX, **and** are using a Taranis X9D or X9D+, we'd like you to test one of these modded firmware build. This is a nightly build from today with one small change

* ⚠️ Set your ELRS to 250Hz or lower using the Lua. 500Hz can't be used at 115200 until RC6. If you skip this step your TX module may get into a reboot loop.
* Go to https://heatermeter.com/dl/elrs/ and download either the X9D or X9DP image and flash your radio.
* Once installed, you'll now have the "MAX BAUDS" menu item in the "HARDWARE" page of the tools menu (hold MENU to access and page through to HARDWARE). If you do not see this item in the menu 🛑 because this is required to try the fix. Verify you've got the test firmware by checking the date on the info screen.
* Change MAX BAUDS from 400000 to 115200.
* See if it works! Also try changing the TLM_REPORT_INTERVAL_MS back to the default 320LU which should now work as well. Note that the x9d-fix branch is not needed for this. You can use master or RC5.

Dominic Clifton was inspired by this firmware to rip his X9D+ apart and reverse-engineer the connections, and affect a hardware fix to change [this sloppy signal](https://cdn.discordapp.com/attachments/839849772864503832/844603065311297546/20210519_175058.jpg) into [this clean signal](https://cdn.discordapp.com/attachments/839849772864503832/844608202539925554/20210519_181123.jpg) by replacing a 10K resistor on the inverter with a 1K resistor. We'll have to look into an easier hardware fix, but testing the above firmware will let us know if this is a simple hardware problem like the Q X7 has.

kimman1, 2021-05-19:
I've tested with tbs crsf when using opentx nightly build with X9D+ from TBS and sometime I notice that TBS crossfire lite module switch to Crsf V2 as input instead of crsfshot. I've never tested on QX7 but I think the hardware of X9D+ have problem when handle  400k baudrate

CapnBry, 2021-05-25:
I've created updated X9D/X9D+ builds with a new feature that shows the telemetry status in opentx. Use the "Onebit_Status" build from https://heatermeter.com/dl/elrs/

I'd be interested in hearing what the Sync status reports for X9D/X9D+ to try to figure out where all the packets we're sending are going. See this wiki page about where to look to report this information.
[Troubleshooting your X9D](https://github.com/ExpressLRS/ExpressLRS/wiki/X9D-(Plus)-Troubleshooting#can-it-be-fixed)

jurgelenas, 2021-11-02:
Is this still relevant?

AlfredBroda, 2021-11-02:
Unfortunately I don't have an X9D+ to test this anymore.

closed 2021-12-20 by brandonrc