ExpressLRS/ExpressLRS #606: Use Tx module as Rx to increase telemetry range and stability

MingchenZhang, 2021-05-15:
Hi, I'm new to the expresslrs, I would like to know if using Tx as Rx is a good idea. 

So the motivation is the telemetry range, Most afforable radio link in the market all have lower (<50mW) transmission power in the Rx module, which make the telemetry range much shorter than the Tx. This can cause issue such as limiting the viability of telemetry based antenna tracker. It also makes other application that want to use a telemetry info impractical at longer range. 
If we use the same Tx module in the aircraft, we would be able to match the range of telemetry with RC links. Since it is so cheap to construct/purchase a expresslrs Tx module and it doesn't weight a ton, I think it is quite practical to use Tx module in a mid to large size aircraft. 

What do you think about using Tx as Rx? Can this be easily achieved in the current state (such as just flashing Rx firmware onto a Tx module)?

brandonrc, 2021-05-16:
@MingchenZhang there have been a few people talk about this idea. I think it is totally posible but the issue is how do we support this option for all systems. 

What hardware do you think you would use as a TX and RX? HM 900mhz stuff?

Tygrys-1, 2021-05-17:
You may use SiYi FM 30 system, the receiver does the same transmit power as the transmitter, at least on original firmware.

CapnBry, 2021-05-17:
> You may use SiYi FM 30 system, the receiver does the same transmit power as the transmitter, at least on original firmware.

It still has full output power on ELRS as well, the TX and RX hardware can go the same distance. So for those who want full range 2.4GHz, the SIYI FM30 Mini target already does this and has working diversity as well.

MingchenZhang, 2021-05-18:
@brandonrc I'm waiting for the happymodel 2.4g module to arrive. 
It seems SiYi FM30 is quite interesting, perhaps i will consider to get one in future.

Dropsleutel, 2021-05-18:
@CapNBry, Any update on the SiYi Rx as a Tx ? Did your extra Rx arrive ? it is till on my list to embed it into my T-lite and keep the 4in1.

CapnBry, 2021-05-18:
> @CapnBry, Any update on the SiYi Rx as a Tx ? Did your extra Rx arrive ? it is till on my list to embed it into my T-lite and keep the 4in1.

This is not a forum. Please keep the ticket discussion on topic, and use Discord to ask questions. That said, I've been working on bugs and testing like 8 hours a day/night so I haven't had time apart from implementing it using the pinheader output and not being able to get the output right no matter what combination of PULLUP/PULLDOWN/NOPULL and hardware inverter setting. I think I'll have to fall back to using the TX2/RX2 but I'd like to try longer to get it working. Just not enough time given the RXLOSS issue in 1.0. Please do not respond this message here.

MingchenZhang, 2021-10-01:
Close this issue since we have many option to purchase high power 2.4ghz expresslrs rx, such as SIYI fr mini and matek expresslrs rx.

closed 2021-10-01 by MingchenZhang