ExpressLRS/ExpressLRS #608: Rssi warnings and Telemetry lost/recoverd massage

rashmandash, 2021-05-16:
Hi,

@developers: Super nice Project and i appreciate your work!

I hope someone can help:)
 
Hardware:
- TX: Taranis x9d plus (2018)
- TX-Module: R9m ACCST (2018)
- RX: R9mm
- I did the [Inverter Mod for R9M 2018](https://github.com/ExpressLRS/ExpressLRS/wiki/Inverter-Mod-for-R9M-2018)

I compiled the OpenTX-NB-Companion-file with this Version 
[[companion-windows-2.3.12N500.exe - 16-May-2021 02:05 - 17325937](https://downloads.open-tx.org/2.3/nightlies/companion/windows/)] and flashed it on my TX.

I also used the latest SD-Card version [[sdcard-212x64-2.3V0037.zip - 07-May-2021 01:03 - 128981778](https://downloads.open-tx.org/2.3/nightlies/sdcard/opentx-x9d%2B/)] 

Run the [r9m_elrs_bl.frk](https://github.com/ExpressLRS/ExpressLRS/wiki/Flashing-R9M-TX) file on my TX. 

Used [ExpressLRS Configurator v0.5.0](https://github.com/ExpressLRS/ExpressLRS-Configurator/releases) to flashed my RX with this settings:

[image]

My TX with:

[image]

Also tried the TLM_REPORT_INTERVAL_MS with 100LU.

I use the  ELRS.lua-File from the ExpressLRS Configurator v0.5.0 with:

- 200Hz; 1/32; 100mW and tried all the others. 

I still get the telemetry lost, telemetry recoverd and low RssI massage.

If more informations are needed please let me know

Cheers Rash

deadbytefpv, 2021-05-17:
Have you tried different Telemetry Ratio?
Because 1:32 is kinda slow... 
Try higher ratio (1:8 for example)

rashmandash, 2021-05-17:
> Have you tried different Telemetry Ratio?
> Because 1:32 is kinda slow...
> Try higher ratio (1:8 for example)

Hi,
yes i tried all from 1:2 to 1: ...

deadbytefpv, 2021-05-17:
An error on my part:
TLM_REPORT_INTERVAL_MS = 100LU is only a band-aid solution, and really just hides the underlying issues.
320LU is more than enough, as OpenTX only expects link status once per second, and this setting sends it 3x (320LU is ~3ms) more.. 

The TLM Ratio ONLY prevents the frequent "Sensor Lost" alarm as we aren't sending all Telemetry data all the time. The advanced Telemetry data ("sensors") gets sent in round robin, and if the Ratio is low, any sensors not sent on that period is thought to have been "lost" by OpenTX.. hence the alarm.

I apologize if the recommendation have caused a confusion.

rashmandash, 2021-05-17:
NP m8,

thx for your communication. I think it's a Problem what get tackled in the future.

CapnBry, 2021-05-19:
Hello folks, we were able to find a developer with an X9D+ and get some test opentx builds out into testers hands. Thanks to bernabap for being the first to report back in!

If you're having "Telemetry Lost" constantly while standing in the same room as your ELRS RX, **and** are using a Taranis X9D or X9D+, we'd like you to test one of these modded firmware build. This is a nightly build from today with one small change

* Set your ELRS to 250Hz or lower using the Lua. 500Hz can't be used at 115200 until RC6.
* Go to https://heatermeter.com/dl/elrs/ and download either the X9D or X9DP image and flash your radio.
* Once installed, you'll now have the "MAX BAUDS" menu item in the "HARDWARE" page of the tools menu (hold MENU to access and page through to HARDWARE).
* Change MAX BAUDS from 400000 to 115200
* See if it works! Also try changing the TLM_REPORT_INTERVAL_MS back to the default 320LU which should now work as well. Note that the x9d-fix branch is not needed for this. You can use master or RC5.

rashmandash, 2021-05-20:
Hi,

thx for the fast reply!!! 

I did the compile with the new [[companion-windows-2.3.12N502.exe - 20-May-2021 02:06 - 17330974](https://downloads.open-tx.org/2.3/nightlies/companion/windows/)]
and flashed it.
I also install the SD-Card [[sdcard-212x64-2.3V0037.zip - 07-May-2021 01:03 - 128981778](https://downloads.open-tx.org/2.3/nightlies/sdcard/opentx-x9d%2B/)] content once again.

I flashed both  
[[X9DP_OneBit_20210519.bin | 2021-05-19 19:37 | 424K |](https://heatermeter.com/dl/elrs/)]
and the
[[X9DP_250000_20210519.bin | 2021-05-19 14:39 | 424K |](https://heatermeter.com/dl/elrs/)]

Tried ["Solution 1: Software fix"](https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/#Solution_1_Software_fix)method

Non of them work. :( and  i can't see:
[image]

I will have a look tomorrow again mayby i have an idea over the night. 

Cheers
----------------------------------------------------------------------------------------------------------------------------------------------

[image]

floaledm, 2021-05-25:
Thanks for offering this test!

Was experiencing the same issue and changing TLM_REPORT_INTERVAL_MS = 100LU wasn't really cutting it. I reverted back to TLM_REPORT_INTERVAL_MS = 320LU before this test. Hardware: X9D+ (per 2019), HappyModel ES24TX+EP2 (1.0.0-RC7). TX set to 250Hz, 1:4 TLM, 10mW.

Flashed my  with X9DP_OneBit_Status_20200522.bin, then tried ONEBIT@400k and Normal@115200 and neither lost telemetry after running for about 10 minutes, whereas Normal@400k loses/recovers telemetry repeatedly (on average about once per minute).

CapnBry, 2021-05-25:
> Flashed my X9D+ with X9DP_OneBit_Status_20200522.bin, then tried ONEBIT@400k and Normal@115200 and neither lost telemetry after running for about 10 minutes, whereas Normal@400k loses/recovers telemetry repeatedly (on average about once per minute).

Thanks for the report @floaledm! Since you're running the OneBit_Status can you go into the Model menu page 2, SETUP, where you'd normally set the External Module type to CRSF. There should be two extra lines there with Sync Status. I'm curious what your G/C/F numbers are there in Normal 400k and OneBit 400k (with a model connected). G=Good packets, C=CRC Error, F=Frame Error. The last two should be 0 on a working system, but I am guessing they are not for you since you're experiencing this issue.

floaledm, 2021-05-25:
@CapnBry Thanks for getting back to me so quickly!
Best estimates I can provide by staring at the screen for a couple of minutes.
Normal@400k: G min:05, max:13, avg:8; C min:0, max:3, avg:1; F min:0, max:57, avg:30
OneBit@400k: G min:13, max:14, avg:14; C min:0, max: 0, avg:0; F min:0, max:0, avg:0

floaledm, 2021-05-25:
Normal@115200 gives me results that are in between the other two: sometimes zero Cs and Fs, other times one or two Cs and one or two dozen Fs...

CapnBry, 2021-05-25:
No thank YOU for getting back so fast @floaledm . It has been hard to find people to report their sync stats numbers to help us narrow down where the missing packets problem is exactly. Super interesting to me is that your numbers are almost exactly the same as the only other report I've gotten, which was G8 C3 F59/60. It seems like it may be clear that this model handset just operates worse in one direction than the other. I'm hoping we can maybe do something on the ELRS side to work around it, but I'm not sure there is a way.

Thanks, I am going to share your results with the other devs what we can come up with.

floaledm, 2021-05-25:
Thanks for your time, and don't hesitate to reach out if you'd like some additional testing.

hojen123, 2021-05-28:
Hello.  First of all, I would like to thank all the devs for upgrading the performance of the old R9M, R9mm equipment to such awesome performance.   Was a great feeling to have revived all this old hardware which gave such inconsistent performance on stock firmware.   Would just like to offer my own observations re this "Telemetry lost" problem with the X9d+.   I own THREE of these radios:  the original (~ 2016/2017), a "Mr Steele" special edition, and the stock X9d+ second edition.   I have several revived R9M Tx, both 2018 and 2019, all with fan mods and the 2018 with the additional invertor-resistor mod.   When I first noticed this "telemetry lost" bug (before the current modified 250k and one bit firmware was available), I immediately did some testing with various combinations of R9M Txs, matched with different radios.   Determined that the problem must be radio dependent, since I saw the same relative incidence of errors:   the Mr Steele  radio > stock second edition >>> the original 2017 radio.   The oldest radio paradoxically gave me the best performance, as I seldom (if ever) saw this error.   Was delighted to try the modified radio firmware, and found that for both the problem radios, the error stopped completely by switching from 400k baud to 115200.  Haven't tested enough to say if the one bit option made a difference, since no errors occured @ 115200 on either one bit or regular.  Didn't see this post before, so didn't realize what the "status" firmware offered.   So if you and the devs would like further testing and reporting of sync status codes, I would be happy to help, as soon as I get a chance in the next few days.   Just let me know what combination of radios, modified radio firmware, baud settings you would like reported, for the sync_status codes.

hojen123, 2021-05-28:
OK. Will do. Hope to send you some data this weekend.

hojen123, 2021-05-28:
BTW, brilliant firmware adjustment for this old radio....

hojen123, 2021-05-29:
some prelim data:
Taranis X9D+  "Mr Steele special ed"  R9M 2018 TX with fan and inverter mod, R9mm Rx
10mW, normal, 400k:      G 9,10     C 0   F 13,14,26
10mW, onebit, 400k:       G 10,11   C 0   F 0
10mW, normal, 115200:  G 10        C 0   F 0
10mW, onebit, 115200:   G 10        C 0   F 0

clearly, on my Radio where the error happens most, switching to EITHER one bit mode or 115200 
corrects the error.

hojen123, 2021-05-29:
with my other two taranis X9D+ radios, (one original 2017, the other "special" edition), 
10mW, normal, 400k:   G 10   C 0    F0

I'm not seeing the error on these other radios on stock settings, so didn't bother trying the other settings.
Like I suggested earlier, the error is radio-dependant, perhaps from poor manufacturer qc ?

kingdraki, 2021-05-29:
Taranis X9D+ from 2016 with X9DP_OneBit_Status_20200522.bin. R9M 2018 TX with fan and inverter mod, R9m Rx, 1.0.0 RC7
10mW, 50Hz, normal, 400k: G 10,11 C 0 F 0, sometimes F>10 for a second or so, every several minutes. But no telemetry lost warning during this time with 320LU.
10mW, 50Hz, onebit, 400k: G 10,11 C 0 F 0, could not see the F jumps there.

Run for 2 hours, stayed connected with the rx in another room. -80dbi most of the time.

modellbobby, 2021-07-19:
Many thanks for this fix. I had the exact same problem with old X9Dplus and R9M now on ELRS 1.0.0 release. 

We need this bugfix in the main branch of OpenTX. How can we assure it is in there? The baud Rate and one bit setting fixed it for me.

brandonrc, 2021-12-20:
This issue appears to be resolved.

brandonrc, 2021-12-20:
> 

If there is a change that needs to made in OTX you need to make an issue in that project.

closed 2021-12-20 by brandonrc

venom-code, 2022-09-05:
Taranis qx7 ACCST, using the betafpv ELRS 1w module, qx7 is on latest edgetx fw and the betafpv elrs module flashed with 2.5.1 fw and packet rate set to 500hz 1st test
on the qx7 the Onebit option has been set and in the radio baudrate is set to 400k and underneath that line is the status on radio showing 500hz but with allot of Err value, the elrsv2 lua opens and works but getting allot of telem lost/recovered messages every few seconds even when trying different telem ratio values seems not to resolve the problem
2nd test when lowered the elrs module packets down to 250hz the Err value in radio status still show but the its less than what it was on 500hz, on 250hz the telem lost/recovered happens but not so often...but it still does.
3rd test was lowering the baudrate on the radio back to 115k the Err in the status stop and the telem lost/recovered messages stop, and either if its in normal or onebit the performance limit is the same then and the messages go away.

venom-code, 2022-09-07:
CF mod resolved the issue completely