ExpressLRS/ExpressLRS #632: RSSI Critical reported on every initial connection

pitts-mo, 2021-05-20:
## Current Behavior

RSSI Critical is reported on every initial connection 

## Steps to Reproduce

1. Power up OpenTX (2.3.11 official)  radio (TX16S) with model configured for ExpressLRS (1.0.0-rc* R9M ACCST FEATURE_OPENTX_SYNC not enabled)

2. Power up flight controller and receiver (1.0.0-rc* R9MM)

3. at time of first signal lock OpenTX consistently reports "RSSI Critical"

## Possible Solution (Not obligatory)

I suspect it may be desirable to create a process on RX that evaluates RSSI after initial boot that disposes of RSSI data until the initial sync appears stable.  Or maybe it would be desirable to preload a dummy value until stable?

## Details

I believe I had observed these messages on all 1.0.0-rc builds. However, today I finally realized this may be a bug.

I have tried a new model in OpenTX to make sure it wasn't something in my model settings or detected telemetry  
If others can not confirm this as a bug I can dig around

## Your Environment

- TX hardware: R9M
- RX hardware: R9MM
- Handset model: tx16s
- OpenTX version: 2.3.11 release verion
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0-rc7 (last several rc builds at least)
- Packet Rate: 200hz
- Telemetry Ratio: 1:16
- user_defines: 
- -DRegulatory_Domain_FCC_915
-DMY_BINDING_PHRASE="WELL-I-NEVER"
-DHYBRID_SWITCHES_8
-DENABLE_TELEMETRY
-DLOCK_ON_FIRST_CONNECTION
-DFAST_SYNC

pitts-mo, 2021-05-21:
Experimenting with telemetry ratio appears to make a difference. Telemetry ratios 1:2 and 1:128 do sync without reporting "RSSI Critical". However, telemetry ratios 1:4,1:8,1:16,1:32,1:64 all report RSSI Critical on initial sync.

I am very curios if anyone else can confirm this behavior exists.
-p

deadbytefpv, 2021-05-22:
Linking a wiki FAQ Entry here:
https://github.com/ExpressLRS/ExpressLRS/wiki/FAQ#is-it-normal-to-get-rf-signal-critical-when-plugging-in

pitts-mo, 2021-05-22:
Thank you @deadbytefpv.

Update 2021-05-22 :
This is a known behavior.  
I believe this issue should remain as low priority bug/enhancement.

I did notice a reference that ExpressLRS is averaging 5 RSSI samples.

Does this fix make sense?
-prevent RX send of real RSSI value for telemetry on RX boot.
-on initial sync RX collect X new RSSI samples before sending real RSSI value for telemetry
-if RX reaches X boot time start sending real RSSI value telemetry regardless of sync/sample collection routine (fail safe even if other conditions fail)

Thanks,
-p

pitts-mo, 2021-05-22:
### 

> `_****_`

pitts-mo, 2021-05-22:
Sorry, I was poking around to see if I could find/set labels :-)

CapnBry, 2021-05-25:
I'm against mucking about with the real information to try to fake out some other system. The issue here is that a connection event doesn't just occur when plugging in, it also occurs following failsafe where you're not guaranteed to have a good connection. The link statistics are more relevant at that time and could contain useful information. Holding off on sending the first link telemetry message means that the TX and handset won't report being connected at all, and possibly will disconnect shortly without ever letting the user know they had control for a short time at low LQ.

I could see benefit to maybe forcing the first telemetry packet as being advanced telemetry type, but I think that might trigger TELEMETRY_STREAMING on opentx, but with 0 RSSI because it was never received and trigger a warning. That's the problem with trying to fake the system just to hush up some upstream system, when it is full of issues. We'll end up bending over backward to make it work and then something changes in opentx and we can only have it not be broken for people running one implementation or the other.

I've also considered having LQ start at 100 on connection reset (we've received 100% of the packets so far!) or scale the LQ by packets received (up to the max LQ) but a minimum LQ is required to establish a connection so that would have to be handled by a separate counter and I feel like that's messy.

I'm also going to discount the fact that you keep saying "RSSI" and none of this is based on RSSI, it is all LQ. 😄

michidk, 2021-06-12:
Just to add my two cents, since I found this issue to be kinda annoying:
I am full with you @CapnBry, we shouldn't fake anything just to work around some other systems. Also, the LQ should always report the true link quality. But that is my point: The link quality is not bad, just because we haven't yet received a telemetry packet yet. So I think starting LQ with 100 is a great solution.

PR #763 (2021-08-07): Start LQ at 100% instead of 0%
There have been complaints and questions about the OpenTX "Signal strength critical" warning on link establishment due to the LQ not being very high on initial connect. This changes the LQ calculator class to account for the number of packets that have been received when calculating LQ. Effectively that means it starts at 100% instead of 1% (for N=100) and will not generate the warning, except where applicable.

This is the proper way to do it anyway, since if there has been 1 packet so far and we got all 1 of them, that should be 100 LQ and not 1 LQ.

Fixes #632 [... cut]
closed 2021-08-07 by pkendall64