ExpressLRS/ExpressLRS #642: RFM95PW (30dBm) seems to be a good candidate for the equally powerfull Receiver for Bi-Directional telemetry

VU2ATX, 2021-05-23:
Hi, while searching on the web, I came accross RFM95P module. This module is exactly of the same family RFM95/96/97/98 which is used in most of the Express LRS Receivers but this one is having 30dBm (1000mW) output power incomparison to the 20dBm (100mW) of RFM95.

I have also recently seen on the Express LRS GitHub that full downlink telemetry support is in progress and this module seems to be the solution.

[image]
[RFM95PW.pdf](https://github.com/ExpressLRS/ExpressLRS/files/6527530/RFM95PW.pdf)

brandonrc, 2021-05-27:
This looks interesting!!! 

Looks like the 900mhz is only 500mW at 27dbm but the 433 can reach that 30dbm at 1w!!!

It also looks like we have the sx1276 support for the RFM95 chip. We would still need an MCU to run ELRS on tho. 

So I wonder what pair could we do to make this a small package for fixed-wing or larger quads.

fjodor00, 2021-05-30:
That means the tbs crsf rx/tx will be supported for flashing? 
as I know the rfm95 and sx1276 are the same, with diff names.

CapnBry, 2021-07-31:
The RFM95P is just a combination frontend and lora chip (sx1276) so there's nothing in ExpressLRS needed to support it. This should be moved to the ExpressLRS-Hardware issues list if you're looking pursue making new hardware that uses this instead of some other RF module.

closed 2021-07-31 by CapnBry