ExpressLRS/ExpressLRS #643: HappyModel EP1 RX bricked after firmware update

qdrk, 2021-05-23:
## Current Behavior
Uploaded RC7 to EP1 RX and suddenly no more WIFI available, no passthrough, RX doesn't connect to TX anymore either. I have used the same method and firmware to update 2 other RXs, both of them works. Wonder if there's a way to unbrick the ESP8285 MCU? The following is the log while trying to passthrough flash EP1, have tried to short the firmware upload pin.

```Checking size .pio/build/HappyModel_EP_2400_RX_via_BetaflightPassthrough/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  40.4% (used 33120 bytes from 81920 bytes)
Flash: [=====     ]  46.9% (used 356957 bytes from 761840 bytes)
Configuring upload protocol...
AVAILABLE: custom, espota, esptool
CURRENT: upload_protocol = custom
Uploading .pio/build/HappyModel_EP_2400_RX_via_BetaflightPassthrough/firmware.bin
Uploading .pio/build/HappyModel_EP_2400_RX_via_BetaflightPassthrough/firmware.bin
  ** Searching flight controllers **
      > FC found from '/dev/ttyACM0'

Detected the following serial ports on this system:
  /dev/ttyACM0

======== PASSTHROUGH INIT ========
  Trying to initialize /dev/ttyACM0 @ 420000

Attempting to detect FC UART configuration...
    ** Serial RX config detected: 'serial 2 64 115200 57600 0 115200
serial 3 0 115'
Enabling serial passthrough...
  CMD: 'serialpassthrough 2 420000'
======== PASSTHROUGH DONE ========
======== RESET TO BOOTLOADER ========
Uploading .pio/build/HappyModel_EP_2400_RX_via_BetaflightPassthrough/firmware.bin
esptool.py v3.0
Found 2 serial ports
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to ESP8266: Timed out waiting for packet header
Serial port /dev/ttyACM0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyACM0 failed to connect: Failed to connect to ESP8266: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
*** [upload] Error 2
============================= [FAILED] Took 24.59 seconds =============================

Environment                                      Status    Duration
-----------------------------------------------  --------  ------------
HappyModel_EP_2400_RX_via_BetaflightPassthrough  FAILED    00:00:24.589
======================== 1 failed, 0 succeeded in 00:00:24.589 ========================
The terminal process "platformio 'run', '--target', 'upload', '--environment', 'HappyModel_EP_2400_RX_via_BetaflightPassthrough'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

## Steps to Reproduce

1. Power on the RX

2. Wait for 20s for it to enter wifi mode

3. Go to 10.0.0.1, choose the built ep firmware and upload

4. When it says uploaded, rebooting, power off ep RX

5. Power it on again, no more WIFI available, betaflight pass through doesn't work either

qdrk, 2021-05-23:
After powering on, the yellow light just stays on all the time.

schugabe, 2021-05-23:
How did you flash the rx? Did you use the configurator or pio? What configuration options did you use? 

Can you try to power the rx without the rx/tx connection to the fc?

I had a similar issue after I tried to flash the EP1 via wifi. Since wifi and passthrough did not work I used the target HappyModel_EP_2400_RX_via_UART and a ftdi adapter to flash it.

qdrk, 2021-05-24:
I flashed it through wifi. I used pio to build the wifi target and uploaded through 10.0.0.1. I have done this to 3 other receivers, all works flawlessly. Tried powering it on withour rx/tx connection, the same. Just tried using FTDI to flash through UART, the same, getting no response from ESP8285 as if it's not there. Here's the error log

```
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/HappyModel_EP_2400_RX_via_UART/firmware.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Invalid head of packet (0x38)
*** [upload] Error 2
================================================================ [FAILED] Took 35.13 seconds ================================================================

Environment                     Status    Duration
------------------------------  --------  ------------
HappyModel_EP_2400_RX_via_UART  FAILED    00:00:35.133
=========================================================== 1 failed, 0 succeeded in 00:00:35.133 ===========================================================
The terminal process "platformio 'run', '--target', 'upload', '--environment', 'HappyModel_EP_2400_RX_via_UART'" terminated with exit code: 1.

```

When I flash through WIFI, I thing I did differently is I power it off only 1-2s after seeing it says "upload completed, rebooting" instead of waiting for at least 5s as I did to other receivers. Would that cause the issue?

qdrk, 2021-05-24:
@schugabe Thanks to your suggestion on flashing using UART! It actually works, I just forgot to short the programming pin to ground the first time. After grounding the programming pin, the UART flashing finished successfully. Still not sure what happened to that WIFI update though.

schugabe, 2021-05-24:
Good to know that your rx is working again.

I think we can close this issue since we already have the ticket #597 for the wifi update problems.

tobiasbischoff, 2021-05-25:
I had this once, while wifi flashing from a phone. Bootloader UART flashing recovered the RX fine.

DeusAbsconditus, 2021-06-01:
Had a similar issue with RC2 --> RC3
https://github.com/ExpressLRS/ExpressLRS/issues/574

CapnBry, 2021-06-05:
I think the issue everyone is experiencing here is that when you flash over wifi and it says SUCCESS **the process is not done**. After the firmware is uploaded to the ESP, it still takes time to actually install (unlike other platforms where the firmware is flashed in-placed), so if the user reboots within ~10 seconds of the flash, it is just like unplugging mid-flash for any other device.

I think the build system and that "success" webpage should be updated to indicate that there is a mandatory wait required before pulling power.

AlessandroAU, 2021-09-14:
I'm assuming this is fixed, reopen if not.

closed 2021-09-14 by AlessandroAU