ExpressLRS/ExpressLRS #645: Crossfire receiver hardware compatibility

padcom, 2021-05-24:
I have a bunch of crossfire receivers and I recently bought the Jumper T18 with 5in1 transmitter. It would be absolutely awesome to use the Crossfire receivers with the 300mW transmitter in the T18.

It's not only for the range but most of all for the ease of use. The transmitter comes with a connector for the 868/915 antenna which can be fitted with a proper dipole antenna which should drastically improve the range (as if the 30km wouldn't be enough :D)

But seriously, not having to install the Crossfire transmitter in the JR bay and still being able to use the crossfire receivers with ELRS would be a huge advantage and give the hardware a whole new purpose. Sure, those receivers aren't the cheapest ones but if one already has them buying the nanos from HappyModel just isn't justified.

padcom, 2021-05-25:
And if need be I can donate some hardware (receivers) for the purpose of implementing this.

brandonrc, 2021-05-27:
Working on pic32 support for Tracer. Then I will add support for CRSF.

brandonrc, 2021-05-27:
Changed the title to pic32 hardware compatibility, we can always tweak this back to PIC32 900mhz hardware if we want to separate the issue but for now, I think we can just include this in one large PR.

howels, 2021-07-15:
What happened to the PR? I know @brandonrc was very busy but the PR was closed with no comments.

brandonrc, 2021-07-15:
Maybe it auto closed because of no activity. 

:/ 

Going to flash some hardware this weekend :) so let's see how that goes.

AlessandroAU, 2021-07-15:
Strange, wasn't me!

howels, 2021-07-31:
Any updates on this? Don't want to nag but I'm excited about the prospect.

padcom, 2021-10-14:
@brandonrc Do you think it makes sense to invest time into making ELRS work on Crossfire hardware in the light of what you said about it needing pickit3? Maybe it'd be just best to relay on ELRS-specific hardware?

brandonrc, 2023-03-19:
Going to close this. 

There is no plans from any developers to add or support pic32. 

Main reason, we rely heavily on platformio which has libraries that are supported outside of ELRS. Pic32 lacks the support compared to ESP/STM and will be unmaintainable. 

Second reason, we want to focus our time on hardware that is made for ELRS. Lots have changed in 2 years and the need to use non-elrs hardware is fading fast.

closed 2023-03-19 by brandonrc