ExpressLRS/ExpressLRS #648: Make BINDING_PHRASE editable via Lua Script

tobiasbischoff, 2021-05-25:
How cool would that be? Want to fly your friends quad? Just enter his Binding_Phrase into your Radio. Seriously that would make my life easier. I often share quads and often you can't just use someones Radio because its using different gimbal configuration, switch settings etc..

flosean, 2021-05-27:
Sounds like a good idea

DungAdumbs, 2021-07-01:
would be a great time to add receiver numbers/model matching to avoid using wrong model profile.  agree, bindphrase makes helping others difficult in the field.

paveltar, 2021-07-02:
Don't forget that all the RX/TX's would have to be flashed with the same settings (those that are required to be the same). So maybe before the implementation of this feature, feature of settings synchronization will be needed.

course-up, 2021-12-20:
Isn't this feature request very similar to the already existing [Setting Binding Phrase](https://www.expresslrs.org/2.0/hardware/spi-receivers/#setting-binding-phrase) commands for SPI receivers:

- set expresslrs_uid
- set expresslrs_rate_index

It would be great to extend the SPI feature to all ELRS receivers and allow the use of the above two commands.
Of course the settings have to be set correctly, but with SPI receivers this is obviously not a big problem. The feature would help me a lot and I think I'm not completely alone.