ExpressLRS/ExpressLRS #650: Add support for bi-directional passthrough telemetry

floaledm, 2021-05-26:
First of all, as someone new to ExpressLRS, I congratulate the team on the impressive capabilities (along with quality documentation) which represent a technological leap in the RC world, especially now that a variety of off-the-shelf hardware is available.

In addition to standard CRSF telemetry packets, there would be multiple uses for supporting custom CRSF telemetry packets that are designed to pass through between the RC controller and the flight controller.

For example, ArduPilot uses passthrough to communicate with OpenTX via FrSky and TBS radio systems. OpenTX makes all CRSF packets it receives with unrecognized IDs available to lua scripts and can also be output to the RC controller's external UART (telemetry mirror). With this setup, detailed flight controller info (that goes far beyond what standard CRSF telemetry can provide) can be displayed on the RC controller using OpenTX scripts. The bi-directional capability also allows users to for example change ArduPilot parameters from within OpenTX scripts. TBS has a frame type reserved specifically for this use: 0x80

I've anecdotally heard a few R9 users waiting for this functionality before moving to ELRS. Thoughts?

CapnBry, 2021-05-26:
We already have bidirectional passthrough telemetry: MSP. It is how the Betaflight Lua script works by tunneling MSP data in CRSF packets.

I think what you're really asking for is for Yaapu scripts support, which uses extended CRSF telemetry IDs that we discard. The telemetry sender we have is very different from other systems that transmit all the telemetry that they get. We have a small buffer that holds the latest one of every telemetry type and rotates through sending them. Yaapu expects that all the data it sends actually gets sent on air, and the packets can be much larger than standard CRSF types, and it sends very fast.

We'll look at the _possibility_ of supporting that in 1.x,, but the downstream bandwidth is even more limited than on systems that use a 1:2 telemetry ratio due to how small our packets are so it may not be practically usable.

floaledm, 2021-05-26:
Thanks for the clarifications, and especially your mention of MSP. Yaapu script support would be terrific, although ultimately maybe it's some kind of MSP implementation of Yaapu that is needed? I understand MSP, at least as used in ArduPilot, as a protocol that handles telemetry (https://ardupilot.org/copter/docs/common-msp-overview.html). 

The telemetry passthrough capability already works out of the box on Crossfire and FrSky radios (e.g., X and R), even though FrSky X radios practically have (based on tests I conducted) no more than 1kbps of payload available for downstream. I don't know what the bandwidth of ExpressLRS is, especially in comparison with CrossFire, but I'd be interested in finding out.

The focus of ExpressLRS may be on latency which I'm sure fits the needs of FPV racers very well. I assume most ArduPilot users don't do FPV, which makes having information displayed on the RC controller that more essential. I also assume most ArduPilot users would rather have more information displayed on their RC controller rather than extremely responsive controls.

Anyway, thanks for looking at that possibility as, in my experience and based on the feedback I've received, this capability is of interest to many ArduPilot users, especially those who want to skip using a GCS altogether.

jimsynz, 2021-06-17:
Just putting my vote in for ExpressLRS to work with Yaapu.  I'm using ExpressLRS flashed R9 gear with my TX16S and Ardupilot for an autonomous rover project.  I'm not so concerned with the "Express" part of ExpressLRS, more the "LRS" part 😄.

schugabe, 2021-09-10:
@floaledm Can you link me a script that would use this 0x80 frame type? And more specific what action in the script does currently not work? I added a first implementation to support this frame type but I'm are struggling to find a test case to test if it's working.

floaledm, 2021-09-10:
@schugabe Thanks for implementing this. Here's an open-source script you can use: https://github.com/yaapu/FrskyTelemetryScript

The documentation for passthrough can be found here: https://ardupilot.org/copter/docs/common-frsky-passthrough.html

For now, none of the passthrough packets from ArduPilot get forwarded by ELRS to OpenTX

schugabe, 2021-09-10:
> @schugabe Thanks for implementing this. Here's an open-source script you can use: https://github.com/yaapu/FrskyTelemetryScript
>
> The documentation for passthrough can be found here: [...]

But isn't that script just using frsky telemetry? I found https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668 but tbh I never used it and don't have any arudpilot planes/quads so I don't really know what to look for.

floaledm, 2021-09-11:
My bad. I thought Yaapu had a CSRF implementation. How about this one: https://github.com/opentx/opentx/blob/2.3/radio/sdcard/taranis-x9/SCRIPTS/TOOLS/CROSSFIRE/crossfire.lua

In the ArduPilot code, this might be helpful: https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_RCTelemetry

yaapu, 2021-09-13:
@schugabe @floaledm hi, my script supports crossfire, you need to enable CRSF support in the config menu.
this is the thread discussing how I implemented Ardupilot custom CRSF telemetry (frame 0x80)
https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668

alphacharlie, 2021-09-28:
Jut a bump to request what the status of this is?

(i.e. - Have any of the above patches been merged upstream with Ardupilot/YAAPU Telemetry/ELRS or are custom versions of the firmware/scripts still required?)

schugabe, 2021-09-28:
> 
> 
> Jut a bump to request what the status of this is?
> 
> (i.e. - Have any of the above patches been merged upstream with Ardupilot/YAAPU Telemetry/ELRS or are custom versions of the firmware/scripts still required?)

The PR is still in draft state and I plan to get it merged before the next major ELRS release.

alphacharlie, 2021-09-28:
That is awesome news!  (I'm still going to try out the PR, was just wondering if the instructions had changed. ;-)

Thank you SO MUCH for all of your hard work on this project! I am really impressed with ELRS performance.

RMFPV, 2021-10-01:
Just my opinion - for Ardupilot users who mostly use fixedwings or big quads it would be more important to have Yappu support even if it leads to some sort of bigger rc link delay.
I've seen many people discussed that in different chats and channels.

Thank you for making this possible!

schugabe, 2021-10-01:
> Just my opinion - for Ardupilot users who mostly use fixedwings or big quads it would be more important to have Yappu support even if it leads to some sort of bigger rc link delay. I've seen many people discussed that in different chats and channels.
> 
> Thank you for making this possible!

https://github.com/ExpressLRS/ExpressLRS/pull/880 work on this feature is already in progress

0crap, 2021-10-09:
Will this have any downsides for people who use ExpressLRS for the initial reason it was made?
High packet rates (500hz) and lowest latency on the market as of today.

avsaase, 2021-11-05:
I think this can be closed.

closed 2022-01-24 by maybenikhil