ExpressLRS/ExpressLRS #654: Dynamic set telemetry to 1:2 when disarmed

ae5au, 2021-05-27:
Has any consideration been given to dynamically setting telemetry to 1:2 when not armed to allow the Betaflight (or other) LUA scripts to be more responsive?  When arming, it would then change back to the normal telemetry setting automatically.  I couldn't think of many downsides to this approach unless there is a technical limitation that I'm missing.  Feel free to close if not feasible.

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

alexdibi, 2021-05-27:
it could also be useful for those who race, the possibility of setting betaflight from lua and once armed, the telemetry off in order to have the lowest possible latency

brandonrc, 2021-05-28:
@ae5au so one thing I could see is a fail-safe situation.. where you lose signal and it goes to 1:2? That will reduce the changes you will get that connection back and really hurt your range. 

I guess we could have it to where it will not go into 1:2 after a connection failed.. that way you can get a connection back. 

The biggest I see tho honestly is your TX has no idea what arming state you are in, arm or disarm.

ae5au, 2021-05-28:
@brandonrc Good points.  I agree that I wouldn't want it to go back to 1:2 after a failsafe, either.

Why not use the same logic for arm as the NO_SYNC_ON_ARM flag?  Does it work as described in the wiki?:

"AUX1 is the channel ExpressLRS uses to detect "ARMED", and this feature assumes that a low value of the arm switch is disarmed, and a high value is armed. OpenTX can invert your switch if you prefer it to be mechanically inverted."

Sorry I haven't reviewed the codebase itself much yet.

JyeSmith, 2021-05-28:
> it could also be useful for those who race, the possibility of setting betaflight from lua and once armed, the telemetry off in order to have the lowest possible latency

Gotta be careful blasting tlm on the block before arming.  This may cause a lot of issue with LQ and potentially arming.

kf4mot, 2021-06-02:
Instead of using arming as a trigger. Could we detect that lua is running and set telem ratio accordingly? Or maybe the lua script itself should set rate on enter/exit if possible?

CapnBry, 2021-08-28:
Fixed via #786

closed 2021-08-28 by CapnBry