ExpressLRS/ExpressLRS #659: ES24TX stops working when changing models, works after reboot

MightyWobbert, 2021-05-29:
Hello Guys!
First, thank you for the awesome project, it´s super cool to see, how fast developement is going and what features ELRS already has. Now to the issue i found:

## Current Behavior

When changing a model in the Radio to another (both models using ELRS), the model you changed to doesn´t connect. The Lua Script isn´t connecting to the module as well (shows connecting, but doesn´t connect and shows no data).

## Steps to Reproduce

1. Be in a model on your Radio running ELRS

2. Change to another Model in your Radio running ELRS as well

3. Power on your quad/dixed wing associated to the newly chosen model on your radio

4. The Transmitter and Receiver will not connect

## Possible Solution (Not obligatory)

Reboot the Radio. Then the Lua Script will work again and the receiver and transmitter will connect normaly.

## Your Environment

Hardware used:
Radiomaster TX16s with Happymodel ES24TX and Happymodel EP1 and EP2 receivers
Software used:
OpenTX 2.3.12 Nighttly as suggested, ELRS 1.0 RC7 radio hardware
Settings used:
500 Hz Packet Rate, 100 mW TX power, enabled Telemetry, Telemetry Rate 1:64, Lock on first connection enabled, no sync on arm disabled

I hope the information is sufficiant to help you to look in to this bug or give me hints for a solution. The problem isn´t realy bothering me, because it can easily be fixed by rebooting the Radio, but it´s a bug nonetheless.
Greets, Robert

sweebee, 2021-05-29:
This is a bug in opentx. See similar issue #628

MightyWobbert, 2021-05-29:
Ok, will be waíting for EdgeTX then, thanks for the answer!

closed 2021-05-29 by MightyWobbert