ExpressLRS/ExpressLRS #660: Frozen throttling quad with no telemetry on the radio

terrysahaidak, 2021-05-29:
## Current Behavior

I've just switched to ELRS on my 3" 3s drone and had few packs on it and it was great
But then suddenly really weird stuff happened.

I'm running DIY TX Slimmer version for TLite with 3.3v step-down/step-up regulator and Happymodel EP2 (SMD antenna) RX on the latest available firmware (RC.7) on 100mw

Today my friend was flying the quad and at some point, it just froze, on DVR I could see that the throttle value just froze and it didn't react to sticks, just smashed into the ground and kept throttling, there was no telemetry on the radio (it shows like 4 lines for telemetry RSSI on t-lite) with the quad screaming upside down next to us, no switches worked, no sticks, after 10-15s I tried to just turn off the radio and it stopped
On DVR I saw that as soon as I turned off the radio it's got RXLOSS and eventually run FS with stage "drop".

This was super scary and I really glad it didn't hurt anybody. I've never had this happened with any of the links (I've used XM+, DSMX, various multiprotocol stuff).

So basically the friend was flying it, at some point it stopped reacting to the sticks, he tried to disarm and nothing happened, it's smashed into the ground then just went crazy with throttle 21, was no telemetry on the radio, even when I turned off the radio, it didn't tell me that there is RX still connected but just turned off and the quad got to FS and stopped.

DVR can be found [here](https://drive.google.com/file/d/1kwEbMvaHM6-tMljuD-nBE2pTWliRLZmk/view?usp=sharing).

CLI dump is [here](https://gist.github.com/terrysahaidak/35b7740856d3e38cf88aa451c397da83).

## Steps to Reproduce

Not sure how to reproduce it yet.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

The only thing I could think about is that 3.3v regulator. T-lite is running from a single 18650. That's why we have to use step-up/step-down regulator. And after I looked at the radio – it showed that the voltage was 3.4. I think my regulator just switched from step-down to step-up thing and could cause temporal power lost for the TX.

## Your Environment

- TX hardware: DIY TX_SX1280_Slim_remix_t-lite but using [this](https://www.pololu.com/product/2842) regulator
- RX hardware: Happymodel EP2 (SMD Antenna)
- Handset model: T-Lite 
- OpenTX version (including nightly number) opentx-tlite factory 84c86f71 (2.3v0035)
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0-RC7
- Packet Rate: 250
- Telemetry Ratio: 1:64
- user_defines: no.

0crap, 2021-05-30:
The regulator you mention is not a step-down/step-up regulator.
It's only a step-down regulator.
If you look at the "Dropout Voltage vs Output Current" graph on the [product page](https://www.pololu.com/product/2842), you see that the input voltage you need to supply is somewhere around 3.4V/3.5V, depending on the current draw of your module.

The 18650 in your handset can go as low as 2.8V under load, not something I would be comfortable with.
Waiting for a disaster to happen, as you experienced.

terrysahaidak, 2021-05-30:
Hi @0crap I'm not sure if I used the right link for it, but it outputs constant 3.3v if I pass 2v or 4,2v.

I'm pretty sure I have other power supply problems as well in my tlite radio:
- I need to use better 18650 cell
- I calibrated the voltmeter in my radio
- also I probably need to power the external module from the battery directly, as suggested for the crossfire, because the power supply cycle in tlite outputs 3.5v to the module even though battery has 4.1v at that moment

But still it doesn't fix the problem. I was able to reproduce it for the second time - quadcopter just froze and was doing bthe last thing it received from the tx, I heard "telemetry lost" from my radio just before that, I immediately turn of the radio and got RXLOSS.

chronowarper, 2021-05-31:
This sounds eerily similar to two of my experiences (in hindsight). I had gotten into ELRS around 1.0 RC1/RC2 with whatever ELRS OpenTX fork at that time (it was from ELRS, not OpenTX's github)

On one of maiden 7" quad flights, the quad suddenly went high throttle after I "lost telemetry" and started lifting into the air at a 20degree angle, whilst ignoring my throttle and disarm commands. I turned off my radio to force it to failsafe. I toasted like $150 worth of parts in the crash. I initially thought this was some issue with GPS rescue, but I was only about 100ft from the quad; now I suspect it's unlikely the issue is GPS rescue.

The 2nd time I was flying my 3" japalura (which I've flown at least 70 times up to this point). However this Japalura runs Frsky. On the 3rd flight, the quad flipped out in a roll and went into the ground. But the throttle stayed on. The quad completely ignored my zero-throttle and wouldn't disarm... it only shut off when I shut off the radio to failsafe.

Because the 2nd flight was with Frsky, I ruled out ELRS. This could be an issue with either OpenTX fork or some really random issue with my radio (which I've used for 3 years...). Since then I've moved back to OpenTx 2.3.11 but also touched up the solder connections in my quad, so I can't say for sure which was the problem.

AlessandroAU, 2021-05-31:
> Hi @0crap I'm not sure if I used the right link for it, but it outputs constant 3.3v if I pass 2v or 4,2v.
>
> I'm pretty sure I have other power supply problems as well in my tlite radio:
>
> * I need to [...]

Can you go over the steps to reproduce it?

terrysahaidak, 2021-05-31:
2 days ago I tried to reproduce it on a bench with same radio. The battery level was at point it doesn't let me enter the Lua script. 

But this time I've got telemetry lost and fail safe after that, seems like the module just shut down at that point. I was trying to record blackbox.

I will try to find a power source I can connect to my radio so I can change the voltage/amperage and simulate this behavior.

Because I'm still not sure is this the right "steps to reproduce".

0crap, 2021-05-31:
Would be nice if you somehow can connect a DMM to your ExpressLRS TX module.
Just to verify you don't have a brownout.

Tweetfpv, 2021-05-31:
I had this same uncontrollable situation as well with a ep2 on rc7.

terrysahaidak, 2021-05-31:
I just flown 3 packs on the same setup but with better fully charged 18650 cell and no issues at all. 

I hope I will be able to simulate power supply issues on a bench today/tomorrow

terrysahaidak, 2021-06-02:
After testing for few days on a fully charged 18650 each time on my t-lite with the same setup I had issues – no problems at all

So yeah 100% for my case it's power issues.

0crap, 2021-06-02:
Thx for the update!

schugabe, 2022-01-21:
I think we can close this issue since we did not have any more reports on this?

closed 2022-01-22 by AlessandroAU