ExpressLRS/ExpressLRS #674: EP2 happymodel no bind mode

noman1971pl, 2021-06-05:
## Current Behavior

EP2 happymodel no bind mode

## Steps to Reproduce
I have three EP2s.
In two pieces, I changed the frimeware to RC1.
There is no bind mode after software change.
The third EP2 of dyeing easily starts bind mode.
I tried to change EP2 software to RC5 or RC7 and it doesn't change anything. I still can't start bind mode and can't bind TX with these two RX

## Your Environment

- TX hardware: happymodel ES24TX
- RX hardware: happymodel EP2
- Handset model:
- OpenTX version (including nightly number):2.3.5
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0-RC1
- Packet Rate:
- Telemetry Ratio:
- user_defines:

noman1971pl, 2021-06-05:
Correction.
After passing the binding procedure (3x power on / off), the yellow LED blinks at 1Hz (not: two blinks, pause).
Nevertheless, I put the TX in binding mode and my RX and TX were bound :)

I still don't know why the RX is not blinking properly (two blinks pause) in binding mode.

pitts-mo, 2021-06-05:
Hello @noman1971pl ,
I think we will need more of the detail that was requested by the issue template. Specifically, can you add the user_defines you use when building the new firmware for your RX/TX?

Taking a guess at the root of your problem... 
TX and RX firmware should be built to match and building from the ExpressLRS Configurator to allow binding support does require you to manually set all your user_defines items and you must NOT include the BINDING_PHRASE in your user_defines.txt.... 

Hope this helps,
-p

related thought:
I have been holding out on using traditional binding until model number support is added (issue #544) and I do find the BINDING_PHRASE method simplifies things and offers some other benefits of interest.

noman1971pl, 2021-06-05:
I don't use the file: user_defines.txt, I don't know what it's for.
I am using Expresslrs Configurator when changing frimeware for RX SP2.
I think that regardless of the uploaded software, whether RC1 or RC8 or any other, the RX receiver should always enter into bind mode.
I may not use TX but RX should always go into bind mode.
The Expresslrs Wiki describes how to start bind mode in RX, and it doesn't work.

CapnBry, 2021-06-05:
If the yellow LED on the RX was blinking at 1Hz you probably weren't actually binding.

The reason the user_defines was requested is because that disables binding altogether if you flash with a bind phrase. The configurator just builds the user_defines for you, so we need everything you've set in the "Device options" in the configurator. A screenshot is ok.

noman1971pl, 2021-06-05:
I know what it means to blink yellow at 1HZ.
As I wrote, despite such blinking, the RX entered the binding mode and I was able to bind with the TX.

Here is my user_defines.txt:
-DRegulatory_Domain_ISM_2400
-DMY_BINDING_PHRASE = "expresslrs"
-DAUTO_WIFI_ON_INTERVAL = "20"
-DFAST_SYNC.

Sorry, I don't know how to put a picture here, here's the link:
https://zapodaj.net/289b0248870cb.jpg.html

I have all settings in Expresslrs Configurator for RX are disabled, except for Auto_WiFi
Why after changing the firmware, no matter what firmware number I upload, the RX is not blinking that it is in bind mode?

CapnBry, 2021-06-05:
You have a bind phrase set, so that completely disables the ability for the RX to enter bind mode. When it was blinking at 1Hz it was waiting for a connection, and not waiting for bind.

RC8 (back some releases before, for sure RC7) has a bug that causes the RX to lock up exiting bind so that could cause confusion, but if you used `-DMY_BINDING_PHRASE = "expresslrs"` to flash the RX then your RX was already bound the instant it was powered. You have to disable the binding phrase on the RX if you want to do traditional binding, however the TX will just use its bind phrase to bind so you're effectively doing the same thing but with an extra step. There's really no reason to use traditional binding given you have to flash the TX and RX anyway.

closed 2021-06-05 by CapnBry

noman1971pl, 2021-06-05:
So we have two options:
1.none "-DMY_BINDING_PHRASE = ..." in user_defines.txt.
In the Expresslrs Configurator, the BINDING_PHRASE option must be turned off.
The RX can then be put into bind mode, blinks twice and pause.
Have to select BIND in Elrs.lua in TX

2. is "-DMY_BINDING_PHRASE = .." in user_defines.
In the Expresslrs Configurator, the BINDING_PHRASE option must be enabled.
Then RX blinks 1 Hz, RX and TX bind automatically after power on ??

Do I understand correctly ??

CapnBry, 2021-06-05:
That's exactly correct. I actually just updated the wiki with this information because I can understand the confusion.