ExpressLRS/ExpressLRS #690: [1.0.0 RC8] In 500Hz mode BF lua script does not work

0crap, 2021-06-16:
## Current Behavior

When in 500Hz packet rate mode, even with Telemetry ratio on 1:2, the BF lua script will not work.
All other packet rate modes work with a decent Telemetry throughput setting. 

## Steps to Reproduce

Select 500Hz packet rate mode, at any TLM ratio setting and start BF lua script.

## Possible Solution (Not obligatory)

Select 250Hz packet rate or lower with a decent TLM ratio will work.

## Details

DIY 2.4TX and EP1 or EP2 tested with same result on RC8.

## Your Environment

Taranis QX7S with TBS crossfire mod and crsfshot capable OpenTX version.
DIY 2.4TX
EP1 and EP2 RX tested.
user_defines and BF set to use Telemetry.

hsuq510, 2021-06-17:
I can confirm same issue with RC8 2.4ghz. 500HZ never work with betaflight lua at any telemetry ratio. Have to move to 250hz to make it work. Using Opentx 2.3.12 nighly, and betaflight 2.4.9 and lua script 1.5.0

ps:  sensor telemetry works fine at 500hz

schugabe, 2021-07-04:
@0crap Please test it again with RC9

0crap, 2021-07-05:
> @0crap Please test it again with RC9

Done and the BF script works great again at 500Hz! Thx!

I did notice one weird thing with RC9 while testing which might be related due to this change?
In my user_defines, for programming the TX, I of course always uncomment -DENABLE_TELEMETRY.
Most of my RX's I also flash with the exact same settings.

However for one small quad, I flash the EP2 with #-DENABLE_TELEMETRY commented. (Only use the basic link quality telemetry for that one.)
This quad will now not bind. ¯\_(ツ)_/¯
If I uncomment -DENABLE_TELEMETRY it immediately binds.

This scenario worked fine using RC8.
Related? Or should I close this issue and make a new one?

0crap, 2021-07-07:
I'll close this one because this is solved.

closed 2021-07-07 by 0crap