ExpressLRS/ExpressLRS #694: PPM output support for RX

FelixvdDonk, 2021-06-18:
The drone I'm targeting only supports PPM input. Is this an option that could be added or enabled in settings?

Tygrys-1, 2021-06-22:
It is kinda not efficient. The PPM frame is more than 16ms long, so you will not benefit from ELRS rates of more than 50Hz.

FelixvdDonk, 2021-06-22:
I understand, but I'm not looking for the benefits of high rates. I care about the long-range benefits. It's merely a problem of compatibility I'm trying to solve here

DungAdumbs, 2021-07-01:
plenty of hardward sbus>ppm>pwm solutions out there.

FelixvdDonk, 2021-07-01:
> 
> 
> plenty of hardware sbus>ppm>pwm solutions out there.

ExpressLRS doesnt output sbus. How is that a solution?

DungAdumbs, 2021-07-01:
Very true, my oversight.  Have not yet seen a crsf hardware solution.

microraptor, 2021-07-12:
SBUS is in development: https://github.com/ExpressLRS/ExpressLRS/issues/581#issuecomment-865076299
As mentioned there is plenty of hardware for SBUS > PWM and some can also output PPM.

goatzillax, 2021-09-06:
Using our existing hardware without extra PPM hardware widgets would be really nice.