ExpressLRS/ExpressLRS #699: Cannot flash EP via Betaflight

nablabla, 2021-06-19:
## Current Behavior
Flash error
*** [upload] Error 2 ========================= [FAILED] Took 18.85 seconds ========================= Environment Status Duration ----------------------------------------------- -------- ------------ HappyModel_EP_2400_RX_via_BetaflightPassthrough FAILED 00:00:18.850 ==================== 1 failed, 0 succeeded in 00:00:18.850 ====================

## Steps to Reproduce
I did as they say here: https://github.com/ExpressLRS/ExpressLRS/wiki/BF-Passthrough.
I have an EP2 (HappyModel)
in ß-flight configurator I set up uart6 to be serialRX, and chose CRSF as serial receiver provider
Connected RX<->TX6, TX<->RX6, VCC<->5V, GND<->GND, connected battery to VBAT on FC.
Connected FC via USB to PC. 
Use ExpressLRS Configurator to build & Flash target "HappyModel_EP_2400_RX_via_BetaflightPassthrough"
Selectec COM3 (which is correct)
(ß-flight configurator is not running)

## Details
My PC is not wifi capable... i have to do it this way. or shall i try to use a usart converter?

## Your Environment
Windows10
FC: mamba F405 mini mk2
- TX hardware: EP02 (happymodel)
- RX hardware: irrelevant
- Handset model: irrelevant
- OpenTX version (including nightly number) irrelevant
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0-RC8
- Packet Rate: donno bro
- Telemetry Ratio: 1:1!
- user_defines: used standart mode in the configurator

pitts-mo, 2021-06-19:
Some initial items I can think of checking:
verify RX actually gets 5v?
Can you try another serial port?
Flip TX/RX pads?

verified EP working elsewhere?
verified FC working with another RX?

Let us know what your find.
-p

other thoughts:
If the boot loader is broken you man need to re-flash utilizing the UART method, an FTDI adapter, and shorting the boot pins on the EP.

I did find the betaflight target for Mamba F405 mk2 should be FURYF4 which is utilizing PC6 PC7 for rx6 so I do not think you are hitting issue #655.

references:
https://www.diatone.us/products/mamba-f405-flight-controller-mk2
https://github.com/betaflight/betaflight/blob/master/src/main/target/FURYF4/target.h
https://www.st.com/en/microcontrollers-microprocessors/stm32f405-415.html#documentation

deadbytefpv, 2021-06-19:
Was the boot pads bridged?
It was required for first time flashing from factory.

[image]
(pads under red marks)

would've been solved via discord..

nablabla, 2021-06-19:
I realized that my phone is a Wifi device, that could do the job. I was able to flash it via the wifi page of th esp.

boot loader NOT bridged

verify RX actually gets 5v?  => true
Can you try another serial port? => i could, would it help tracing the issue?
Flip TX/RX pads? => it works now, so I assume it is all correct

verified EP working elsewhere? => working on this quad now
verified FC working with another RX? => yes and working now

other thoughts:
If the boot loader is broken you man need to re-flash utilizing the UART method, an FTDI adapter, and shorting the boot pins on the EP. => the bootloader appeared not to be broken

I did find the betaflight target for Mamba F405 mk2 should be FURYF4 => it says FURYF4OSD

pitts-mo, 2021-06-19:
Glad you were able to get this sorted out @nablabla

@deadbytefpv nailed it with first time flash, boot pads, and use of discord.

This is another instance where I have briefly thought about indexing or maybe master pages for each major hardware in the Wiki... Easier said than done. 

Good to close this issue?
-p

closed 2021-07-05 by AlessandroAU