ExpressLRS/ExpressLRS #700: ELRS LUA: Script Syntax Error ?:0 Attempt to compare number with nil

cmanley, 2021-06-20:
## Current Behavior

I was calibrating my RSSI in Betaflight and had the ELRS LUA script (version 98310a) open in my X9D to set the update rate to 250Hz. I went for a walk to check the RSSI and link loss behavior and when I came back, I saw this on my X9D (OpenTX 2.3.12) screen:
```
Script Syntax Error 
?:0 
Attempt to compare number with nil
```
## Steps to Reproduce

Not sure. I thought that a link loss may have triggered it but that was not the case. I haven't been able to reproduce it. 
Unfortunately, the line number in the error is meaningless.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Happymodel JR module
- RX hardware: Happymodel EP1
- Handset model: X9D
- OpenTX version (including nightly number) 2.3.12 release
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0 RC8
- Packet Rate: 250Hz @ 10mw
- Telemetry Ratio:
- user_defines:

codigodiabolico, 2021-06-22:
I had the same issue going to 25hz packet rate, is related to telemetry interval.. a workaround: delete TLMinterval from menu and change packet rate, this restores the default telemetry interval.

https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lua/ELRS.lua#L143 «- delete TLMinterval

CapnBry, 2021-07-31:
This seems to be caused by an invalid configuration on the TX and the Lua not being defensive enough about the data to work around it. The Lua is going to be replaced entirely in 1.1 by #620 so this issue should be closed when that is merged.

whmeade50, 2021-08-07:
I have been battling the same syntax error, on a brand new Happy Model Expresslrs, for two days and finally fixed it.  I moved the jumpers on the back to the far left side (They were in the middle) and built and flashed the ExpressLRS TX using the HappyModel_ES24TX_2400_TX_via_UART and plugged in a USB cable from my computer to the back of the TX.  Then I flashed the receiver via Wi-Fi. Everything is working perfectly now.  Just thought this might help someone else out.

CapnBry, 2021-08-28:
I don't think this is an issue on 1.0.1 since there haven't been any other reports of it, I'm closing this issue but if others are experiencing it, be sure you've updated your firmware.

closed 2021-08-28 by CapnBry

sercona, 2022-11-08:
let me add this just in case someone finds this in a search.

my environ: just bought a new radiomaster tx16s and was trying to copy over data from my older jumper t16.  I converted most of my quads over to elrs 2.4ghz (3.0.0) and I have quite a few tx jr plugin modules for my full size radios and the zorro small format, as well.  overall, I love elrs and am actively weeding out old frsky rx's one by one.

on my jumper and zorro, both at edgetx 2.7.1 iirc, the lua scripts for the elrs modules worked great.  

then, I started adding some lua 'packs' like the fm2m, the inav widget, the betaflight config widget and the yaapu widget.

when I added all those scripts to both my tx16s and t16 (leaving my zorro alone, for now), I found that I could not run the lua config for elrs no matter what I tried.  just would show errors or hang but never really run.  mind you, I could USE the internal elrs (on tx16s) and external on t16; my custom fw worked fine with my binding phrase and I could see telemetry and fly my elrs 2.4ghz quads.

its JUST lua that would not work and not let me change things.

something told me to remove the 'extra' lua packs.  that fixed it for me!

my lesson: beware of 3rd party lua stuff going on.  I have not debugged this, but I suspect some interaction going on and for now, I removed all of those since none are truly needed TO FLY.  the lua for elrs is semi-needed; but the others are fluff and so they are now removed from my radios.

in case anyone gets weird errors on trying to run the lua config on internal or external elrs, think about your plugins and removing them.