ExpressLRS/ExpressLRS #701: Bluetooth Joystick

michidk, 2021-06-21:
While playing an FPV simulator, the USB cable can be very annoying. There are commercial solutions for this problem, like the [FrSky XSR-SIM Usb Dongle](https://oscarliang.com/frsky-xsr-sim-dongle/). Also, it is possible to build such a dongle using a [Betaflight FC and a receiver](https://oscarliang.com/betaflight-fc-fpv-simulator/). I always wondered, why most remotes don't just ship with a Bluetooth module, so this can be used to connect to the PC. ELRS can fix this issue (and already has).

There is this commit by Alessandro that implements a proof-of-concept: https://github.com/ExpressLRS/ExpressLRS/commit/8ed72e0d95ebd7d7a786eca13fae7edd10c5a5a6
Also, the configurator project already implemented the required flag: https://github.com/ExpressLRS/ExpressLRS-Configurator/blob/831f18118cea2b6e55adf7af6176e4abd275739f/src/api/src/library/FirmwareBuilder/Enum/UserDefineKey.ts#L33

So I wonder two things:
1. Is it planned for this to be included in v1?
2. What is missing here to merge this feature?

Right now the WiFi update feature is disabled and used to toggle the Bluetooth joystick. One would have to adjust the LUA script to add a button and a new CRSF telemetry packet like 0xFD to enable this feature. Not sure if @AlessandroAU is already working on this? Otherwise, I could implement it.

pkendall64, 2021-06-21:
I too have made some changes which adds a new button in the LUA script. But was waiting for the 1.1 work to start before creating a PR as I know the LUA config model is changing. The code also needs to be rebased onto the latest tip.
https://github.com/pkendall64/ExpressLRS/tree/ble-joystick-changes

michidk, 2021-06-21:
@pkendall64 Great to hear! This seems pretty much complete then (besides the rebase and adjusting the LUA for 1.1).
The only thing I would add (but not necessarily with the first release of this feature), would be to pass the other channels as gamepad axis (does this even work? if not we have to map them to buttons), too. I like to use my arm switch in Velocidrone to reset my drone for example.

maybenikhil, 2021-09-02:
Done for ESP32 Modules #815

closed 2021-09-02 by AlessandroAU

sfjuocekr, 2022-01-06:
@michidk if you have a Teensy board you can make your own little receiver as well, I've got a project on my page that does that.

I still have to implement CRSF, but it works for any SBUS receiver!

L1cardo, 2022-12-05:
@michidk There is a project called CRSFJoystick, and I made a pr recently, you may want to have a look.
https://github.com/mikeneiderhauser/CRSFJoystick/pull/8

[image]
[image]
[image]
[image]