ExpressLRS/ExpressLRS #702: Add firmware version information in LUA script screen

MicroMontis, 2021-06-25:
It would be great to have ELRS TX firmware version  (eg. "FW Ver. 1.0 RC8") on the LUA script screen. This would solve a lot of confusion as I can see people scratching their heads and wondering what exact version they have on their TX.
There is a plenty space for additional info on the screen ;)
[image]

Related to the FR: https://github.com/ExpressLRS/ExpressLRS/issues/584

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

zhoumigo, 2021-06-26:
yes  nice !!!

NiklasVoigt, 2021-07-07:
it is more accurate to specify the git hash because it is unique. this way you can identify which branch etc. it is.

98310a == RC8

[image]

MicroMontis, 2021-07-21:
A common user (that does not use github in everyday life) would rather see "V1.0.0 RC8" instead of "98310a9" ;)

howels, 2021-07-27:
I'd rather have the hash, it's less ambiguous.

MicroMontis, 2021-07-28:
Why can't we have both? "So the wolves could be fed and the sheep kept safe.”

maybenikhil, 2021-09-02:
#813 does this, although not perfectly, but can fetch tags + branches that aren't master

jurgelenas, 2021-11-02:
@CapnBry since it was implemented in #704, can we close this now?

CapnBry, 2021-11-02:
I suppose we can. 2.0 has the TX version number in the lua, and both the RX and TX show their versions and target on the webui. 1.x has the version number of the TX in the lua but there's still no way to see the RX version number, and we're not going to fix it for that branch since it is already done in master.

closed 2021-11-02 by CapnBry