ExpressLRS/ExpressLRS #703: 100Hz does not work

qubic, 2021-06-26:
## Current Behavior

100Hz does not work
## Steps to Reproduce

1. set MY_BINDING_PHRASE like "hello"

2. set Packet Rate to 100Hz

3.

4.

## Possible Solution (Not obligatory)

## Details
100Hz does not work since abd63d9, and 200Hz and 50Hz does not work at that commit.

## Your Environment

- TX hardware:HappyModel ES915TX
- RX hardware:FrSky R9MM
- Handset model:FrSky X9D+
- OpenTX version (including nightly number): 2.3.12 (fdb5cc37)
- ExpressLRS version (TX & RX MUST MATCH): 29d834c
- Packet Rate: 100Hz
- Telemetry Ratio: 1:32
- user_defines:
```
-DMY_BINDING_PHRASE="hello"
-DRegulatory_Domain_FCC_915
-DDISABLE_STARTUP_BEEP
```

deadbytefpv, 2021-06-27:
I don't think reporting a "bug" on a commit that's not even a Release Candidate matters... there's a reason it's not a Release Candidate..

qubic, 2021-06-27:
> I don't think reporting a "bug" on a commit that's not even a Release Candidate matters... there's a reason it's not a Release Candidate..

I am reporting the bug on master and 1.0.0-RC8

deadbytefpv, 2021-06-27:
What is not working with it? i.e. How can we determine what exactly is not working? How can we verify "it" is not working?

Been flying on 200Hz, Wez range testing on 900Mhz were on 50Hz.. granted we're not on X9D(+)..

qubic, 2021-06-27:
The tx can't bind rx at 100Hz, they can bind at 200Hz and 50Hz.

deadbytefpv, 2021-06-27:
So I'll set my tx module (r9M 2019 ACCESS; ExpressLRS RC8) to 10mw 100Hz 1:16TLM... 
Radio is TX16s SE on EdgeTX RC3

Turn my Quad on (with an r9mm Receiver on RC8) and it connects.. almost instantly...

FCC_915
Lock on First Connection
Binding Phrase

qubic, 2021-06-27:
It doesn't work with r9m lite (in x-lite) and r9mm at 100Hz either.

mhon44, 2021-06-29:
> It doesn't work with r9m lite (in x-lite) and r9mm at 100Hz either.

I do use X-Lite with R9M Lite, RC8, no problem with connection on any of packet rates (I am flying mostly 50 and 100Hz, 1:16TLM). RX's R9Mini and R9MM

CapnBry, 2021-07-31:
I can't reproduce this on 1.0.0 with ES900TX + R9 Mini
```
-DMY_BINDING_PHRASE="hello"
-DRegulatory_Domain_FCC_915
-DDISABLE_STARTUP_BEEP
```
100Hz 1:32 10mW, but I tried other rates as well. They all connect with at the most 20s of delay, and 100Hz connects in 2 LED blinks if the TX is running before the RX is powered.

qubic, 2021-08-06:
After i flash r9mm via STLINK, the issue gone.

closed 2021-08-06 by qubic