ExpressLRS/ExpressLRS #704: Show Version in LUA GUI (not same as #702)

ltwin8, 2021-06-27:
I have noticed a problem/bug:

flashing via wifi from an iPhone (using safari on iOS 14.6) I can flash the Receiver and Module via WiFi, it Shows „success, rebooting“, however I have trouble believe in this because the version of the *TX* in the LUA Script does not change. On the receivers I do not know.

however flashing from windows laptop via WiFi works. Also flashing the TX via USB works on Windows perfectly and can be confirmed via LUA script.

the files are the *.bin files compiled by the configurator.

Hardware: one ES24TX and eight EP2 RXs, QX7 as tx

my feature request would be following: 
When devices are upgradable via WIFI please show the version on the hosted Website, as the String should be available? The GitHub Tag would be enough, as we only need to compare and not know exactly what RC we have.

Second request: in the LUA script we get a readout on the version of the TX FW, is it possible to pull the firmware-version of the RX on power up and just bitwise compare both strings? Followed by maybe a message like this:
FW OK (for versions matching, version can be found on top of the screen)
UPDATE (when Receiver has different FW, maybe alternate the current RX version in  the textbook)
ERROR (for the time the TX was not able to pull the version of the RX, this would be the default state when no RX is connected or is running FW that does not support FW control)

additional: maybe add a byte at the end of the FW string where the bits represent settings like Sync, 500Hz mode, switchmode etc)

The First request should be easy as the website used to upgrade already shows version (v1.0.0 for example) please just extend that with -98310A9 for RC8

ltwin8, 2021-06-27:
Please add lable „bug“ in regards of flashing via iOS, as there is no way of knowing in the moment without having an exact readout

MicroMontis, 2021-06-27:
Somehow related to https://github.com/ExpressLRS/ExpressLRS/issues/584
Agreed, we need version numbers in LUA GUI and Wifi Update Webpage. And it's better to have RC number than some hash string. Average user does not know where to find or what to do with that string.

CapnBry, 2021-08-28:
The version number is now available in the web update screen #584, which just leaves the Lua to have a better display #702 so I am closing this as a duplicate.

closed 2021-08-28 by CapnBry