ExpressLRS/ExpressLRS #708: SX1280 high gain mode

Sekilsgs2, 2021-07-02:
When i'm testing modules with sx1280 i'm see that this line does not work properly - 

`hal.WriteRegister(0x0891, (hal.ReadRegister(0x0891) | 0xC0));`

i'm comment this line and rssi not decreasing.

Next i'm change this to 
`hal.WriteRegister(0x0891, 0xC0); `

and now i'm have more better rssi - anybody test this moment?

JBKingdon, 2021-07-02:
Woah, that sounds interesting. Can you check what the result of `hal.ReadRegister(0x0891) | 0xC0` is?

Sekilsgs2, 2021-07-02:
> Woah, that sounds interesting. Can you check what the result of `hal.ReadRegister(0x0891) | 0xC0` is?

This interesting - but i'm not check :) 
But this need confirmations from other users - maybe i'm have special chip :)
And i'm get more than 3db improvement

JBKingdon, 2021-07-02:
This is the section from the datasheet:
`High sensitivity mode is enabled by setting bits 7:6 at address 0x891 to 0x3. Once enabled the noise figure of the receiver is improved by up to 3 dB for 700 μA of additional current consumption.`
So it looks like we're doing the right thing, but we should definitely try and test and see what happens in practice.

Sekilsgs2, 2021-07-02:
> This is the section from the datasheet:

Yes i'm read this too - but my test's have other results...

JBKingdon, 2021-07-02:
Awesome, thanks for opening the issue. I tested and thought I saw a difference at the time, but 3dB is easy to swamp with small changes in position. I remember the other guys saying it didn't seem to do much.

Sekilsgs2, 2021-07-02:
> Awesome, thanks for opening the issue. I tested and thought I saw a difference at the time, but 3dB is easy to swamp with small changes in position. I remember the other guys saying it didn't seem to do much.

I'm tested with fixed transmitter and receiver - only changing the code - for me - the result was immediately visible
But again need test's

JBKingdon, 2021-07-02:
Some quick debug shows the register was 00100101 at boot up and 11100101 after we set the top bits. I can't find any documentation for what the low bits do, but it does support the possibility that clearing them changes the behaviour.

JBKingdon, 2021-07-02:
My reading of the datasheet is that high sensitivity mode enables additional steps of gain in the LNA. With AGC running (which is our default setup) this is likely only to have a visible effect at the lowest signal strengths when the AGC is running at max available gain. I setup a test with the sensitivity mode toggled by aux4, min tx power and got the rssi down to about -100dBm and wasn't able to see any benefit from our existing use of high sensitivity mode. Next I'll test with clearing the low bits.

JBKingdon, 2021-07-02:
Well, there's something dramatic going on when the low bits are toggled. I could be persuaded that the rssi does get a roughly 3dB boost, but the link also becomes unstable and drops at around -50dBm. I think we need to find out more about what these lower bits do before we can make any changes, but it does seem that the current implementation isn't doing anything useful.

Sekilsgs2, 2021-07-02:
> Well, there's something dramatic going on when the low bits are cleared. I could be persuaded that the rssi does get a roughly 3dB boost, but the link also becomes unstable and drops at around -50dBm. I think we need to find out more about what these lower bits do before we can make any changes, but it does seem that the current implementation isn't doing anything useful.

Oh - yes - i'm have this too (link drop with -50dbm) when flying on E28-2G4T27SX and ported to this modules elrs - but then I didn't know what it had to do with it.. Strange - have signal boost, but -50db signal lost - very strange.. and strange that we dont have signal boost with cuurrent code..

JBKingdon, 2021-07-02:
Ah, I need to test again while monitoring SNR as well as RSSI to see if the original code is in fact doing anything useful. I forgot that RSSI at low signal levels should be correct by SNR.

CapnBry, 2021-11-04:
I'm going to flag this as closing, as I have also just tested this and the signal strength seems better if you clear the bottom bits but the connection also drops sooner. It doesn't matter that the number reported is better if it also raises the sensitivity limit by the same amount. I can also confirm that the bits are being set correctly in normal operation.

closed 2021-11-08 by Sekilsgs2