ExpressLRS/ExpressLRS #715: Dynamic RF update rate

SunjunKim, 2021-07-05:
I'm new to this community, transferred from the TBS ecosystem.

Overall the system is satisfactory for almost the entire aspects, however, one regret is a dynamic rate as like in a Crossfire module. When encountering a low LQ in a long-range flight, Crossfire switches to lower rates to regain the link with better sensitivity.

We may consider two options:
1) Change the link speed via an AUX channel (the pilot manually selects them during a flight).
2) Automatic speed change regarding LQ/RSSI dBm.

I'm not sure this is a simple implementation or requires fundamental changes in the binding process, but I hope it would be certainly beneficial for long-range pilots.

SunjunKim, 2021-07-05:
I quickly checked the TX code, and I believe ```config.SetRate``` function could be utilized without messing up the existing TX/RX binding.  I wonder about the risk of changing rates in mid-flight, for example freaking out the FC with false stick positions and AUX switches.

russphelan, 2021-07-11:
To be fair, this is exactly why I don't like crossfire. I want a lock at a consistent update rate. I'm not going anywhere near the edge of my range as a freestyle/racing miniquad pilot, so the failsafe/sensitivity issue isn't on my radar. It would be nice to have as an option, as long as it's possible to turn it off.

CapnBry, 2021-07-12:
This is a pretty bad idea as far as the current codebase goes. Changing rates requires that both the RX and TX agree to make the switch and presumably this rate change would take place at a distance where packets are already being lost. Currently, the TX sends a handful of SYNC packets with the new rate and then switches over to the new rate, assuming the RX will get at least one and also switch.

However, if the RX does not switch rate, you're going to failsafe and recover in a long time (if LOCK_ON_FIRST_CONNECTION= false) or never (if LOCK_ON_FIRST_CONNECTION=true). There's no solvable way to guarantee the RX got the rate change, even if the RX sends a telemetry packet that says "OK I got it" because that packet could also be lost on the way back. 

Our rate changes also change the encoding parameters, so packets won't even be detected if they're at the wrong rate. It isn't just a case of them coming at a different cadence that the RX will figure out on its own. I'm not sure how Crossfire does it, because it is sort of magic.

JyeSmith, 2021-07-12:
Just select the correct RF mode required for the flight. Done.

russphelan, 2021-07-12:
> This is a pretty bad idea as far as the current codebase goes. Changing rates requires that both the RX and TX agree to make the switch and presumably this rate change would take place at a distance [...]

I do know that crossfire makes the switch way, way, way before the edge of your range, which is what makes it so annoying. Maybe that's how they get around the packet loss issue. If you're in dynamic mode, crossfire will take from 150hz to 50hz as quickly as 10 or 20 meters away flying freestyle. It will even switch you within a race track in an open field.

Another thing: this forced you to use dynamic cutoff frequency detection on your betaflight rc smoothing, and whenever the switch occurs, your smoothing will be confused for a period. A small issue, but it's nice to set the cutoff right at your locked packet rate and get a smoooooth trace.

SunjunKim, 2021-07-13:
Thanks for all your responses. 
I've started thinking dynamic RF is not for ELRS.

SunjunKim, 2021-07-13:
> as long as it's possible to turn it off.
Doesn't TBS Crossfire provide an option to turn it off?

0crap, 2021-07-13:
> Thanks for all your responses.
> I've started thinking dynamic RF is not for ELRS.

Dynamic Packet rate, no, not so much.
For TBS crossfire it's a necessity, otherwise their 900 Mhz stuff just drops from the sky.
For ExpressLRS this is not an issue. (Because we exclusively use LoRa modulation.)

On the other hand, dynamic power output would be a great feature to have!

> > as long as it's possible to turn it off.
> 
> Doesn't TBS Crossfire provide an option to turn it off?

Sure, it does.
If you do, you will failsafe after 100 meters or so. (CRSF 900 Mhz.)

SunjunKim, 2021-07-13:
> On the other hand, dynamic power output would be a great feature to have!

I'm working on a dynamic power feature (see #305) and hopefully it could be merged to v1.1.

sweebee, 2021-07-13:
Dynamic power is a great feature. Dynamic packet rate would be a bad thing since you could feel it. Dynamic power has no disadvantages. 

Is especially great if you use both ceramic and the normal rx. I have more than enough range on 10mw with normal antennas, but I need 100mw on the ceramic rx.

DroneQuad, 2021-07-16:
i think dynamic power output would be a great idea, when packet rate still manual, cause can make a trouble on mid flight i guest

SunjunKim, 2021-07-16:
So let's close this issue, as we (kinda) get a consensus that Dynamic RF rate is not needed for ELRS.

closed 2021-07-16 by SunjunKim