ExpressLRS/ExpressLRS #719: [1.0.0 RC9] Binding issue with #-DENABLE_TELEMETRY setting

0crap, 2021-07-07:
## Current Behavior

When I program my TX with advanced telemetry enabled, it will not bind with an RX that has this setting disabled.

## Steps to Reproduce

Flash my DIY 2.4GHz TX with the option -DENABLE_TELEMETRY.
So advanced telemetry is enabled.

Flash an EP2 with the option #-DENABLE_TELEMETRY.
So advanced telemetry is disabled for this specific RX.

Now the TX and RX will not bind.

This possibility was working with RC8. 
If I enable advanced telemetry on this RX (flash again with -DENABLE_TELEMETRY) it binds perfectly and quickly.

## Possible Solution (Not obligatory)

None.

## Your Environment

- TX hardware: DIY JR-bay 2.4 GHz
- RX hardware: EP2
- Handset model: Taranis QX7S
- OpenTX version (including nightly number): latest stable
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0 RC9
- Packet Rate: 250Hz
- Telemetry Ratio: 1:64
- user_defines: [here](https://pastebin.com/qaGuE2XA)

deadbytefpv, 2021-07-07:
I hope you know that firmware options on TX and RX MUST be the same for a bind to happen...

0crap, 2021-07-07:
Certainly.

But this option has been an exception.
It worked in RC8.
If this changed just let me know, I'll close this issue.

0crap, 2021-07-08:
@deadbytefpv Not sure what went wrong, but I flashed the EP2 a second time and all is as expected.
So TX is using advanced telem. and RX is not using it, set in user_defines before compile.

closed 2021-07-08 by 0crap