ExpressLRS/ExpressLRS #720: EP2 update to 1.0.0-RC9 - now its just slowly flashing.

sasplayers, 2021-07-07:
## Current Behavior
I just updated both the Happymodel ExpressLRS 2.4Ghz TX to the newest 1.0.0-RC9 version that went fine or so I think. But once I updated the EP2 to the new version, it didn't bind and now just flashes slowly - won't even go into WiFi mode after 20 seconds passes.

As explained above, once flashing the EP2 was complete it just flashed slowly, so I've no idea

## Steps to Reproduce
just update to 1.0.0-RC9

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

It means that I can't use that drone, whiling to try anything to get it going again, I'll not be updating the others until this one is fixed.

Happymodel ELRS2.4- ready made.

## Your Environment

- TX hardware: HappyModel ELRS 2.4Ghz
- RX hardware: HappyModel EP2 2.4Ghz
- Handset model: Radiomaster TX16s Hall sensors
- OpenTX version (including nightly number) Open TX 2.3.13
- ExpressLRS version (TX & RX MUST MATCH): 1.0.0-RC9 on both
- Packet Rate: default
- Telemetry Ratio: default
- user_defines: default

Dravonic, 2021-07-07:
I have the same problem, flashed my Happymodel 2.4ghz TX and EP RX to RC9 but now it won't bind. Pretty sure I'm not doing anything stupid as both RX and TX have the same binding phrase and the same features selected where applicable. Have been updating every RC release since RC6.

I can still enter WiFi mode after 20 seconds though, might try flashing back to RC8 to see it it helps.

sasplayers, 2021-07-07:
Dravonic your lucky having WiFi, looks like I'm going to have to make a USB TTL converter with an Ardunio Uno just to flash the old firmware back on it.. P.S. don't forget to flash RC8 on your TX too.

> I have the same problem, flashed my Happymodel 2.4ghz TX and EP RX to RC9 but now it won't bind. Pretty sure I'm not doing anything stupid as both RX and TX have the same binding phrase and the same [...]

And Good luck..

Dravonic, 2021-07-07:
Went back to RC8 and everything is working again.

sasplayers, 2021-07-08:
well I've reinstalled ELRS configurator v.0.5.5 and did a flash via Betaflight passthrough with RC9, taking note to bridge the boot loader pins on the EP2 then plug in the flight controller via USB, start the configurator (build/flash) button, then wait in the log window for the configurator to open the COM port and then plug the battery into the flight controller.

in that order.

not sure if the backward configurator version to v0.5.5 did the trick most likely not. Probably just connecting the battery at the right time.

this thread can be closed.. although I'd leave it posted or pinned as I can imagine a few people will get this problem on updating to the EP2 to version RC9 may also effect EP1.

0crap, 2021-07-09:
> this thread can be closed.. although I'd leave it posted or pinned as I can imagine a few people will get this problem on updating to the EP2 to version RC9 may also effect EP1.

You can always close your own issue by clicking on the "Close and comment" icon.
Thx for the housekeeping.

closed 2021-07-09 by sasplayers

Stratoglide, 2021-07-12:
I just managed to have this issue with three brand new happy model EP1 receivers. All slow flashing yellow light and wont ever enter wifi mode. I tried the above method to update via betaflight passthrough but haven't had any success there either. 

usb to serial adapter is on order so hopefully that will work

Receiver 1 was previously was working on rc8  with no issues number 2 and 3 where fresh outta the package. Number 1 and 2 where updated via android while powered over usb, number 3 was powered via battery

Even stranger I just updated another EP1 receiver on another build yesterday and it had no issues, using the same firmware and same phone/browser to update it.

Seems to be a bit of a bigger issue here. Initially I thought it might be due to the usb power but i bricked my final reciever just to double check.

SUPERBAFFOFPV, 2022-05-12:
Hi all. I think to have the same problem to my receiver, I just updated both the Happymodel ExpressLRS 2.4Ghz pro TX to the newest 2.5 elrs version that went fine. But once I updated the EP2 (which was binded to an other old tx in version 1) to the new 2.5 version, it didn't bind and now just flashes slowly - won't even go into WiFi mode after 1 minute  passes.

The only thing that maybe I noticed is that when i built the firmware for the tx, the auto wifi interval was automatically set to 60 and i left that value, but when I built the rx firmware I set it to 20... maybe that was the problem? now the rx just flashes once slowly, won't go into WiFi mode even after 2 minutes, and even if I turn on and off the power quickly for 3 times...

Can you help me?