ExpressLRS/ExpressLRS #730: Betafpv RX Wifi Target

Guili123, 2021-07-19:
In the ELRS configurator, the BETafpv Wifi targets are missing.

I found in the BETAfpv description that the 2.4G and 900/868 RX is also providing the ability to update via wifi?
It would be great to get this target also.

michidk, 2021-07-19:
This is an issue for https://github.com/ExpressLRS/ExpressLRS-Configurator
Also, this was fixed about three hours ago. There is already a new release for the configurator that fixes this: https://github.com/ExpressLRS/ExpressLRS-Configurator/releases/tag/v1.0.1

Both target definitions are here: https://github.com/ExpressLRS/ExpressLRS/blob/master/src/targets/betafpv_900.ini and https://github.com/ExpressLRS/ExpressLRS/blob/master/src/targets/betafpv_2400.ini

jurgelenas, 2021-07-19:
> In the ELRS configurator, the BETafpv Wifi targets are missing.
> 
> I found in the BETAfpv description that the 2.4G and 900/868 RX is also providing the ability to update via wifi?
> It would be great to get this target also.

_via_WIFI targets are not required for WIFI and they actually do not work correctly. You have to build binary file for your TX/RX using configurator and flash it through flashing web page when connected to your tx/rx wifi hotspot.

Guili123, 2021-07-19:
They don`t work correctly?
Never had issues with my wifi bin files.

jurgelenas, 2021-07-19:
> They don`t work correctly?
> Never had issues with my wifi bin files.

You can upload binary files from _via_UART and _via_Betaflight_passthrough targets - they are all the same.

jurgelenas, 2021-07-19:
> They don`t work correctly?
> Never had issues with my wifi bin files.

via_WIFI target is meant to be used bydirectly flashing device via wifi from configurator, but at the moment it is difficult to orchestrate that

Guili123, 2021-07-19:
Ah I see. Never flashed this anyway directly from the configurator.
But thanks, this is a very useful information!

JyeSmith, 2021-07-20:
Tagging @LinKR00001 to have a look at this later.

CapnBry, 2021-07-31:
BetaFPV via_WIFI targets are now available in 1.0.1 release thanks to wvarty #736

closed 2021-07-31 by CapnBry