ExpressLRS/ExpressLRS #738: RSSI dBm value dependent on the packet rate

ondrascz, 2021-07-23:
The RSSI dBm readout is consistently higher while using the higher packet rate (50Hz vs 200Hz). At the same output power and the same distance and position of the RX and TX the RSSI dBm is always higher on higher packet rate. I'd expect that the RSSI dBm readout would be the same and independent on the packet rate. Phenomenon discovered by flights to about 2.5km, retested on bench in as controllable environment as possible using three different receivers.

Test results in the attached table:
[image]
 

- TX hardware: R9M lite (ELRS 1.0.0)
- RX hardware: Multiple R9MM and R9Mini (ELRS 1.0.0)
- Handset model: FrSky Taranis X-Lite
- OpenTX version OpenTX 2.3.13
- ExpressLRS version 1.0.0
- Packet Rate: 50Hz and 200Hz
- Telemetry Ratio: 1:8
- user_defines:

AlessandroAU, 2021-07-27:
I've checked power output against different packet rates and it did not change. Still investigating...

CapnBry, 2021-07-31:
I would say you're crazy because it shouldn't make any difference, but I tested on a HappyModel ES900TX and I see about a 5dB drop in reported RSSI switching from 200Hz to 50Hz (50Hz has lower RSSI). This is not the case on any Team2.4 rates-- all the rates 50Hz to 500Hz report RSSI in the same range in my testing.

I tested the power output at the different rates and the peak is the same, so power isn't the problem. I guess the question is if we're doing something wrong at 50Hz that causes worse reception, or if the SX127X is just reporting different RSSI at that rate.

0crap, 2021-08-01:
I tested this with my 2.4GHz DIY TX module (DIY_2400_TX_ESP32_SX1280_E28 target) and a HappyModel EP2 RX.
All flashed on v1.0.1
Tested at 10mW and 25mW at all possible packet rates.
The RSSI dBm readings I got are always exactly the same.

So I agree with CapnBry, #Team2.4 seems not affected regarding this issue.

CapnBry, 2021-11-04:
I'm closing this because there's been no further evidence to support that the RSSI depends on Packet Rate, and there's nothing we could do about that if there was.

closed 2021-11-04 by CapnBry